dorssel/usbipd-win #154: USB mass storage connecting and visble as device but not accessible

akuropka, 2021-12-08:
MS Windows 11 x64 [Version 10.0.22000.348]
WSL v2 Ubuntu-CommPrev Linux 5.10.74.3-microsoft-standard-WSL2+
**usbipd-win 1.3.0**

Attaching a USB mass storage seems to be successful, however it is not accessible within the WSL VM.

> $ lsusb
> NAME MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
> sda    8:0    0 344,9M  1 disk
> sdb    8:16   0     4G  0 disk [SWAP]
> sdc    8:32   0   256G  0 disk /
> sdd    8:48   1  29,7G  0 disk

> $ sudo fdisk /dev/sdd
> Welcome to fdisk (util-linux 2.37.2).
> Changes will remain in memory only, until you decide to write them.
> Be careful before using the write command.
> 
> fdisk: cannot open /dev/sdd: Input/output error

@dorssel traced it down to this error:
> USBSUP_XFER_UNDERRUN -> EREMOTEIO -> -121
> ...
> USBSUP_XFER_STALL -> EPIPE -> -32

see: #153 
Trace: [log1.txt](https://github.com/dorssel/usbipd-win/files/7677879/log1.txt)

akuropka, 2021-12-09:
The USB mass storage device is a USB 2.0 card reader from Hama (https://www.hama.com/00091092/hama-8in1-usb-20-card-reader-sd-microsd-transparent?srcLang=0&targetLang=1). The microSD-slot works while the standard sized SD-slot does not work (tested with different memory cards). However, I can access those cards with a different reader without any issues.

Maybe this is a too special case to deal with.

`dmesg` output for reference:

> [ 1039.002282] vhci_hcd vhci_hcd.0: pdev(0) rhport(0) sockfd(3)
> [ 1039.003069] vhci_hcd vhci_hcd.0: devid(65541) speed(3) speed_str(high-speed)
> [ 1039.003747] vhci_hcd vhci_hcd.0: Device attached
> [ 1039.345610] usb 1-1: new high-speed USB device number 6 using vhci_hcd
> [ 1039.495846] usb 1-1: SetAddress Request (6) to port 0
> [ 1039.530190] usb 1-1: New USB device found, idVendor=058f, idProduct=6459, bcdDevice= 1.00
> [ 1039.530777] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [ 1039.531307] usb 1-1: Product: Flash Card Reader/Writer
> [ 1039.531622] usb 1-1: Manufacturer: Generic
> [ 1039.531895] usb 1-1: SerialNumber: 058F64596479
> [ 1039.533353] usb-storage 1-1:1.0: USB Mass Storage device detected
> [ 1039.534300] scsi host1: usb-storage 1-1:1.0
> [ 1040.583945] scsi 1:0:0:0: Direct-Access     Generic- Multiple Reader  1.07 PQ: 0 ANSI: 4
> [ 1040.588220] sd 1:0:0:0: Attached scsi generic sg3 type 0
> [ 1041.447056] scsi 1:0:0:1: Direct-Access     Generic- MicroSD/M2       1.08 PQ: 0 ANSI: 4 CCS
> [ 1041.451890] sd 1:0:0:1: Attached scsi generic sg4 type 0
> [ 1041.457481] sd 1:0:0:0: [sdd] 31116288 512-byte logical blocks: (15.9 GB/14.8 GiB)
> [ 1041.468916] sd 1:0:0:0: [sdd] Write Protect is off
> [ 1041.469291] sd 1:0:0:0: [sdd] Mode Sense: 23 00 00 00
> [ 1041.474687] sd 1:0:0:0: [sdd] Write cache: disabled, read cache: disabled, doesn't support DPO or FUA
> [ 1041.476386] sd 1:0:0:1: [sde] Attached SCSI removable disk
> [ 1041.488238] sd 1:0:0:0: [sdd] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x00 driverbyte=0x08 cmd_age=0s
> [ 1041.489163] sd 1:0:0:0: [sdd] tag#0 Sense Key : 0x2 [current]
> [ 1041.489612] sd 1:0:0:0: [sdd] tag#0 <<vendor>>ASC=0xff <<vendor>>ASCQ=0xff
> [ 1041.489996] sd 1:0:0:0: [sdd] tag#0 CDB: opcode=0x28 28 00 00 00 00 00 00 00 08 00
> [ 1041.490436] blk_update_request: I/O error, dev sdd, sector 0 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
> [ 1041.491020] Buffer I/O error on dev sdd, logical block 0, async page read
> [ 1042.305982] sd 1:0:0:0: [sdd] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x00 driverbyte=0x08 cmd_age=0s
> [ 1042.306898] sd 1:0:0:0: [sdd] tag#0 Sense Key : 0x6 [current]
> [ 1042.307423] sd 1:0:0:0: [sdd] tag#0 ASC=0x28 ASCQ=0x0
> [ 1042.307979] sd 1:0:0:0: [sdd] tag#0 CDB: opcode=0x28 28 00 00 00 00 00 00 00 08 00
> [ 1042.308479] blk_update_request: I/O error, dev sdd, sector 0 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
> [ 1042.309077] Buffer I/O error on dev sdd, logical block 0, async page read
> [ 1042.309481]  sdd: unable to read partition table
> [ 1042.714243] sd 1:0:0:0: [sdd] Attached SCSI removable disk

dorssel, 2021-12-09:
@akuropka 
Does this configuration work on a native Linux machine?

akuropka, 2021-12-09:
I am able to mount it on a RPi4 (running Arch).
If I bind the device there with usbip I am able to attach and access it properly within the WSL on my Windows machine.

dorssel, 2021-12-09:
@akuropka 

> If I bind the device there with usbip I am able to attach and access it properly within the WSL on my Windows machine.

Excellent feedback! My guess is that there are still some USB message(s) that need to be trapped and handled differently. This is already the case for: SET_CONFIGURATION, STALL recovery, SET_ALTERNATE_INTERFACE, and REMOTE_WAKE. These are "special" and require more handling than just proxying directly to the device. Looks like I missed one...
However, I need some debugging information that is currently not generated. Will think about on how to obtain that...

akuropka, 2021-12-09:
Thanks for your efforts; anything I can do to help please let me know anytime.

akuropka, 2021-12-10:
Here you go...

[win.zip](https://github.com/dorssel/usbipd-win/files/7690421/win.zip)
[rpi.zip](https://github.com/dorssel/usbipd-win/files/7690422/rpi.zip)

dorssel, 2021-12-10:
@akuropka 
Thanks for the captures.
Very clear results:
- Both show no obvious errors on the USB side of things.
- Device setup is identical on both (getting DESCRIPTORS, SET CONFIGURATION, etc.)
- Both show a short period after device init where the device shows "not ready" while it is detecting the card: `[Status: Check Condition (0x02)]`. This is normal.
- After the device is ready, both show a read of the first page (8 sectors of 512 bytes); this contains the partition data

Then the differences:
- On RPi: the first read succeeds immediately, after which a few more pages are read; these contain the file system info
- On Win: the first read fails (not a USB failure, but a SCSI layer failure) with another "not ready". The read is then repeated, which succeeds, with the exact (byte-for-byte) data as RPi did. But then: no further read requests.

Also, the commands are forwarded faster on Win (localhost) than on RPi (remote). For example, the time between "prevent medium removal" and the read of the first page:
- On RPi: 44 ms
- On Win: 0.9 ms (failure "not ready") and again after 64 ms (successful)

So it looks like we're "too fast": the first read fails with "device not ready". The read *is* repeated, and then succeeds, but apparently Linux already decided on failure.

Could you try to repeat the mount a few times? It just looks like bad luck with timing. Or maybe a flaky/dodgy/cheepo reader...

In any case, all the USB stuff works fine; it's just the SCSI sublayer that shows errors.

akuropka, 2021-12-11:
@dorssel, very interesting elaboration, thank you for your efforts.
Based on this I tested further:
- waiting does not have any effect, it does not become ready by it's own.
- mounting does not have any effect as the system does not access the device but throwing an error as /dev/sdd1 does not exist yet.
- calling `fdisk /dev/sdd` fails twice (but the system tries to access the device), however always succeeds exactly on the third occasion.
Strange behavior...

dorssel, 2021-12-12:
@akuropka 
After `fdisk` succeeds (3rd time), can you reread the partition table with:

```
sudo partprobe
```

Does that create the partition block devices?

akuropka, 2021-12-13:
If using `partprobe` it works instantly afterwards;

> andi@ANDI:\~$ lsblk
> NAME MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
> sda    8:0    0 344,9M  1 disk
> sdb    8:16   0     4G  0 disk [SWAP]
> sdc    8:32   0   256G  0 disk /
> sdd    8:48   1  29,7G  0 disk
> sde    8:64   1     0B  0 disk
> andi@ANDI:~$ sudo partprobe
> Warning: Unable to open /dev/sda read-write (Read-only file system).  /dev/sda has been opened read-only.
> andi@ANDI:\~$ lsblk
> NAME   MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
> sda      8:0    0 344,9M  1 disk
> sdb      8:16   0     4G  0 disk [SWAP]
> sdc      8:32   0   256G  0 disk /
> sdd      8:48   1  29,7G  0 disk
> └─sdd1   8:49   1  29,7G  0 part
> sde      8:64   1     0B  0 disk

The way through `fdisk` updates the devices as well (no need for partprobe):

> andi@ANDI:\~$ lsblk
> NAME MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
> sda    8:0    0 344,9M  1 disk
> sdb    8:16   0     4G  0 disk [SWAP]
> sdc    8:32   0   256G  0 disk /
> sdd    8:48   1  29,7G  0 disk
> sde    8:64   1     0B  0 disk
> andi@ANDI:\~$ sudo mount /dev/sdd1 /media
> mount: /media: special device /dev/sdd1 does not exist.
> andi@ANDI:\~$ sudo fdisk /dev/sdd
> 
> Welcome to fdisk (util-linux 2.37.2).
> Changes will remain in memory only, until you decide to write them.
> Be careful before using the write command.
> 
> fdisk: cannot open /dev/sdd: Input/output error
> andi@ANDI:\~$ sudo fdisk /dev/sdd
> 
> Welcome to fdisk (util-linux 2.37.2).
> Changes will remain in memory only, until you decide to write them.
> Be careful before using the write command.
> 
> fdisk: cannot open /dev/sdd: Input/output error
> andi@ANDI:\~$ sudo fdisk /dev/sdd
> 
> Welcome to fdisk (util-linux 2.37.2).
> Changes will remain in memory only, until you decide to write them.
> Be careful before using the write command.
> 
> 
> Command (m for help): p
> Disk /dev/sdd: 29,72 GiB, 31914983424 bytes, 62333952 sectors
> Disk model: Multiple Reader
> Units: sectors of 1 * 512 = 512 bytes
> Sector size (logical/physical): 512 bytes / 512 bytes
> I/O size (minimum/optimal): 512 bytes / 512 bytes
> Disklabel type: gpt
> Disk identifier: 1D906131-83B6-4307-BC38-C93C41C3F772
> 
> Device     Start      End  Sectors  Size Type
> /dev/sdd1   2048 62332927 62330880 29,7G Microsoft basic data
> 
> Command (m for help): q
> 
> andi@ANDI:\~$ sudo mount /dev/sdd1 /media
> andi@ANDI:\~$ sudo umount /dev/sdd1

closed 2021-12-14 by akuropka