dorssel/usbipd-win #155: Unplugging USB mass storage in WSL leads to crash

akuropka, 2021-12-08:
MS Windows 11 x64 [Version 10.0.22000.348]
WSL v2 Ubuntu-CommPrev Linux 5.10.74.3-microsoft-standard-WSL2+
**usbipd-win 1.3.0**

Unplugging a USB mass storage after unmounting it in WSL2 (but not detaching with usbipd) leads to crash:
`Fatal error while logging another fatal error.`
Windows doesn't realize that this device was unplugged. 
Plugging in the device again does not cause any reaction on Windows. After restarting usbipd it is still attached there.

see: #153 
Trace: [log2.txt](https://github.com/dorssel/usbipd-win/files/7677870/log2.txt)

akuropka, 2021-12-08:
👍
But detaching means attaching to Windows and that I wanted to avoid.

closed 2021-12-10 by dorssel

dorssel, 2022-01-17:
@akuropka 
I've just finished a PR that may solve your problem. See #164, the installer is at https://github.com/dorssel/usbipd-win/actions/runs/1705631433. Could you give that a go?
This version supports "force-bind" with "usbipd bind --busid --force". This will permanently install the stub driver, as an alternative to messing with device enable/disable in device manager.

akuropka, 2022-01-18:
Hej @dorssel,
this are great news. However, it doesn't work as expected.
Thew bind force command works without error but the driver seems not to be installed properly ("object name not found."),.

[image]
[image]

[usbipd.txt](https://github.com/dorssel/usbipd-win/files/7890189/usbipd.txt)
(While binding there was no trace given.)

dorssel, 2022-01-18:
@akuropka 
Can you find in the Windows EventLog (probably under System log) what the driver error is?
Is VBoxUsbMon running (it should be!), you can check with:

> sc query VBoxUsbMon

akuropka, 2022-01-18:
[image]
[image]

dorssel, 2022-01-18:
@akuropka 
That makes sense. VBoxUsb driver requires VBoxUsbMon driver to be running. When VBoxUsbMon driver is installed, it *should be* automatically started. However, I've seen situations where uninstalling it and then immediately reinstalling it does not do that. Did you uninstall `usbipd-win` and then immediately installed the PR-version, without reboot?

Can you please first try (as admin):
```
sc start VBoxUsbMon
```

and the replug the device?
If that does not work, does a reboot help?

I think this is an artifact from uninstall/reinstall wihtout reboot.

dorssel, 2022-01-18:
@akuropka 

> Btw generally it should be possible to do this all without restarting if no device is in use by usbipd.

Agreed, but Windows... The driver installation is entirely handled by Windows PnP manager. And it often just requires a reboot for no apparent reason. But in any case, the VBoxUsbMon driver should always start, both after install and after reboot. I wonder why it wasn't.

dorssel, 2022-01-18:
@akuropka 
On my machine, I can query/stop/start VBoxUsbMon at will:
```
PS C:\Users\frans.vandorsselaer> sc query VBoxUsbMon

SERVICE_NAME: VBoxUsbMon
        TYPE               : 1  KERNEL_DRIVER
        STATE              : 4  RUNNING
                                (STOPPABLE, NOT_PAUSABLE, IGNORES_SHUTDOWN)
        WIN32_EXIT_CODE    : 0  (0x0)
        SERVICE_EXIT_CODE  : 0  (0x0)
        CHECKPOINT         : 0x0
        WAIT_HINT          : 0x0
PS C:\Users\frans.vandorsselaer> sc stop VBoxUsbMon

SERVICE_NAME: VBoxUsbMon
        TYPE               : 1  KERNEL_DRIVER
        STATE              : 1  STOPPED
        WIN32_EXIT_CODE    : 0  (0x0)
        SERVICE_EXIT_CODE  : 0  (0x0)
        CHECKPOINT         : 0x0
        WAIT_HINT          : 0x0
PS C:\Users\frans.vandorsselaer> sc start VBoxUsbMon

SERVICE_NAME: VBoxUsbMon
        TYPE               : 1  KERNEL_DRIVER
        STATE              : 4  RUNNING
                                (STOPPABLE, NOT_PAUSABLE, IGNORES_SHUTDOWN)
        WIN32_EXIT_CODE    : 0  (0x0)
        SERVICE_EXIT_CODE  : 0  (0x0)
        CHECKPOINT         : 0x0
        WAIT_HINT          : 0x0
        PID                : 0
        FLAGS              :
PS C:\Users\frans.vandorsselaer>
```

akuropka, 2022-01-18:
Yay! I am getting closer... everything seems to work except that I'm not getting the drive mounted. It just hangs indefinitely...
[usbipd.txt](https://github.com/dorssel/usbipd-win/files/7890647/usbipd.txt)
Regarding that I will elaborate further tomorrow.

akuropka, 2022-01-19:
I'm sorry for all the inconvenience. The mount issues seem to be kernel related... tested thoroughly with an older kernel as well as other Linux machines and everything seems to work as expected.

One thing I noted is that if I do a cleanup ([Device Cleanup Cmd](https://www.uwe-sieber.de/misc_tools.html)) the binding still remains but the forced flag is removed which is kind of inconsistent. But this may be a too special use case.

Second, `unbind --all` removed forced bindings as well and makes the device visible in Windows again. Considering this is intended to be a long term setting this may be unexpected.

And I am wondering if `forced` is really the perfect term (while if basically fits) or if something like `persistent` would be better.

dorssel, 2022-01-19:
@akuropka 
> One thing I noted is that if I do a cleanup (Device Cleanup Cmd) the binding still remains but the forced flag is removed which is kind of inconsistent. But this may be a too special use case.

`--force` overrides the driver *at that time*. If you then (later) run other tools that change or update drivers... 'Forced' is nothing more than setting the driver immediately that would otherwise be set only temporarily when attaching. If you remove or change the driver, then you fall back to the normal "Shared" state.

> Second, unbind --all removed forced bindings as well and makes the device visible in Windows again. Considering this is intended to be a long term setting this may be unexpected.

Not at all, `unbind --all` means "stop `usbipd-win` from messing with my devices". Which is exactly what it does. It undoes *everything* that `usbipd-win` changed. It is also used by the uninstaller. It is supposed to leave no traces.

> Second, unbind --all removed forced bindings as well and makes the device visible in Windows again. Considering this is intended to be a long term setting this may be unexpected.

Yes, I agree here. The terminology isn't quite right. Linux `usbipd` uses "bind", but it is temporarily: unplug or reboot and all is forgotten. `usbipd-win` does things persistently, it remembers devices across unplug/replug as well as across reboots. Much like sharing a folder, or a printer. That's why we decided calling it "sharing", in line with Windows terminology. So "persistent" isn't quite right either, as all binds are persistent. What `--force` does is set the driver immediately (and indeed persistently, provided you do not use other tools to change the driver yet again). It's maybe not a great term, but it is the best I could think of.

All in all: thanks for all you great reports! I think the PR is stable now. Will merge.

PR #153 (2021-12-10): Refactor attach
- Use restart instead of disable/enable
- Fix unplug detect
- Fix cleanup on startup for non-present devices

Fixes #152 
Fixed #155 
Resolves #91