dorssel/usbipd-win #156: Allow sharing of disabled devices

dorssel, 2021-12-08:
Use case as described in https://github.com/dorssel/usbipd-win/issues/155#issuecomment-989135302.

You may want to use a device *exclusively* for remote sharing, and never use it on the host. Current behavior is that when a device is not attached, it is available to the host. Attaching it to a client means "disconnecting" it from the host, and that may fail if the host runs a driver/service/software that holds the device in busy mode. What if you don't want to use it on the host, ever?

Normally when you don't want to use a device, you disable it in device manager. Currently, `usbipd-win` will not enumerate such devices as available for sharing. This implies that all shared devices must be enabled on the host, causing the host to have to deal with "disconnect" on attach, and "reconnect" on detach. 

### Feature Request

Allow `usbipd-win` to share devices that are disabled on the host, and that stay disabled on the host when handed back after detach.

Thanks @akuropka for this idea.

dorssel, 2021-12-10:
@akuropka
You can now disable the device in the Device Manager and still attach it from a client. When you detach it, it will still be disabled on the host. Saves you the trouble of unplugging/replugging.
Build artifact at https://github.com/dorssel/usbipd-win/actions/runs/1565351426

akuropka, 2021-12-11:
Actually this is very neat 😎 and works like a charm based on preliminary testing.

PR #160 (2021-12-11): Allow sharing of devices that are disabled on the host
Resolves #156
closed 2021-12-11 by dorssel