dorssel/usbipd-win #159: Jetson Flash through usbipd not working

tdallapiazza, 2021-12-10:
Hi,
First thank you for the excellent job and opening possibilities to develop for embedded on WSL on Docker, these are really great perspectives.
By the way I just tried to flash a jetson nano devkit through the usbipd from wsl and unfortunately it is not working.

Basically, after wsl attach, the devise is well listed with lsusb but the flash script get stuck at some point.

So this is what I did:

1. Install usbipd-win (latest release at the time of writing)
2. start a wsl vm
3. in an **elevated** windows shell I attached the usb device with 'usbipd wsl attach --busid=<BUSID>'
4. in WSL I can see the device when issuing 'lsusb'

I just post it here to notify my trial and stay ready to deliver some logs or whatever will be riquired to track down what happens.

Cheers,
       Thomas

dorssel, 2021-12-10:
You probably don't have the right driver enabled in the WSL kernel. See https://github.com/dorssel/usbipd-win/wiki/WSL-support.

tdallapiazza, 2021-12-13:
I have original WSL Kernel is 5.10.60.1-microsoft-standard-WSL2. I did checkout the WSL2 kernel and checked tha config.
Appart the "Debug messages for USB/IP" all the other components you mentionned in the wiki are already built-in. I agree that maybe something is missing but how to figure out what?

I'm now compiling the kernel with USB Mass Storage support just to see if it helps...
Cheers!

tdallapiazza, 2021-12-13:
My kernel is now 5.10.74.3-microsoft-standard-WSL2+ compiled with required config from your wiki but there's no change...
How do I get som debug info from the communication when I try to flash (like wireshark but for USB device?)
Cheers!

dorssel, 2021-12-20:
@tdallapiazza 
Since the device appears on the WSL side in `lsusb`, I suspect the problems are on the WSL side. Usually it is a missing driver, but it can also be `udev` related. Can you post the output of `lsusb`. It may also help to see the `syslog` (on WSL, this may require that you first install `rsyslogd`).

Just to confirm that there is nothing wrong on the Windows side: see https://github.com/dorssel/usbipd-win/wiki/Troubleshooting on how to run `usbipd-win` with debug logging.

dorssel, 2022-02-11:
Closing due to lack of response.

closed 2022-02-11 by dorssel

tdallapiazza, 2022-08-15:
Hi,
Sorry for not responding for a while!
By the way I managed to make it work but I needed to also add support for Mass Storage USB devices in the WSL kernel as the flashing script will finally burn the emmc memory on the device.
So I wanted to take the opportunity to thank you for the excellent job! Now there is no limitation of developing embedded systems directly from WSL. Brilliant!