dorssel/usbipd-win #161: Windows 11

HackXIt, 2021-12-12:


HackXIt, 2021-12-12:
Just noticed that #118 is also related to USB 3.1, so this issue probably is a duplicate.

dorssel, 2021-12-12:
Could you post the output of:

```
reg query "HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}" /v UpperFilters
```

HackXIt, 2021-12-12:
@dorssel 
```
❯ reg query "HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}" /v UpperFilters

HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}
    UpperFilters    REG_MULTI_SZ    USBPcap
```

dorssel, 2021-12-12:
@HackXIt 
I've noticed that some UpperFilters interfere with VBoxUsbMon grabbing the devices. USBPcap (older versions) used to be incompatible with VirtualBox as well. Could you (temporarily) remove USBPcap, or rename the UpperFilters registry key? Then reboot, and try `usbipd-win` again.

HackXIt, 2021-12-12:
@dorssel
Just did as instructed and now it works, no issue.

So, simple solution would be uninstalling of USBPcap ?
Personally I have no need at the moment, just so happened that I blindly clicked through Wireshark installation and installed that driver too.

dorssel, 2021-12-12:
@HackXIt 
I don't know how recent your version of USBPcap was. Maybe the latest version is OK, but certainly older versions were incompatible with the VirtualBox drivers. Let me know if you get it to work once you upgrade USBPcap. If the latest version of USBPcap also does not work, then yes, all you can do is uninstall USBPcap.

In any case, glad you have it working now. Please let me know about the USBPcap upgrade, if you are willing to try. If not, I think you can close this issue.

HackXIt, 2021-12-12:
@dorssel 
I just checked my version: `1.5.4.0`
This appears to be the most recent version, at least from what I can tell on https://github.com/desowin/usbpcap

However, since I'm not using that bit of software anyways, I'll just uninstall it and call it a day. Thank you for the help.

closed 2021-12-12 by HackXIt