dorssel/usbipd-win #163: Busy device while trying to attach it

akuropka, 2021-12-14:
**usbipd-win 1.3.1 on Windows 10 x64 19044.1387**

I am connecting a Samsung Galaxy S20+ to the Windows PC and bind it which goes without error but the typical windows notification sound does not happen as well. Trying to attach it to a remote Ubuntu system fails.

Server side console:

> Microsoft Windows [Version 10.0.19044.1387]
> (c) Microsoft Corporation. Alle Rechte vorbehalten.
> 
> C:\WINDOWS\system32>usbipd list
> Present:
> BUSID  DEVICE                                                        STATE
> 4-1    Synaptics FP Sensors (WBF) (PID=003d)                         Not shared
> 4-3    HP HD Webcam [Fixed]                                          Not shared
> 4-5    HP hs2350 HSPA+ Mobile Broadband Module USB Device            Not shared
> 4-6    Andreas's S20+, SAMSUNG Android ADB Interface                 Not shared
> 5-6    Broadcom 20702 Bluetooth 4.0 Adapter                          Not shared
> 
> Persisted:
> GUID                                    BUSID  DEVICE
> 
> C:\WINDOWS\system32>usbipd bind --busid 4-6
> 
> C:\WINDOWS\system32>usbipd list
> Present:
> BUSID  DEVICE                                                        STATE
> 4-1    Synaptics FP Sensors (WBF) (PID=003d)                         Not shared
> 4-3    HP HD Webcam [Fixed]                                          Not shared
> 4-5    HP hs2350 HSPA+ Mobile Broadband Module USB Device            Not shared
> 4-6    Andreas's S20+, SAMSUNG Android ADB Interface                 Shared
> 5-6    Broadcom 20702 Bluetooth 4.0 Adapter                          Not shared
> 
> Persisted:
> GUID                                    BUSID  DEVICE
> 
> C:\WINDOWS\system32>

Client side console:

> andi@ANDI:~$ sudo usbip list --remote=LING
> [sudo] password for andi:
> usbip: error: Protocol spec without prior Class and Subclass spec at line 23179
> Exportable USB devices
> \======================
>  - LING
>         4-6: Samsung Electronics Co., Ltd : Galaxy A5 (MTP) (04e8:6860)
>            : USB\VID_04E8&PID_6860\R58N41ZS9BP
>            : (Defined at Interface level) (00/00/00)
>            :  0 - Imaging / Still Image Capture / Picture Transfer Protocol (PIMA 15470) (06/01/01)
>            :  1 - Vendor Specific Class / unknown subclass / unknown protocol (ff/42/01)
> 
> andi@ANDI:~$ sudo usbip attach --remote=LING --busid=4-6
> usbip: error: Attach Request for 4-6 failed - Device busy (exported)
> 
> andi@ANDI:~$ sudo usbip attach --remote=LING --busid=4-6
> usbip: error: Attach Request for 4-6 failed - Device busy (exported)
> 
> andi@ANDI:~$

Manually ejecting the device before the attach command does the trick.

_I am wondering if this is intended like this?_

dorssel, 2021-12-14:
Yes, it is intended like that. Some process or service on the host was using the device exclusively. In earlier versions of `usbipd-win`  the device was simply "yanked" from the host, which could cause data loss. See #91.

Work is currently ongoing to make a `bind --force` flag that claims the device for remote sharing permanently, i.e. without the host getting control between attaches.

What is interesting is:

> usbip: error: Protocol spec without prior Class and Subclass spec at line 23179

It looks like a mistake in the Linux hwdata database, but maybe it is related to `usbipd-win`. Can you try to find out more about this error?

akuropka, 2021-12-14:
Yeah, I was reporting that one. But I was wondering here because in #91 I don't need to eject the device first manually. But it may be different with other types (mass storage vs MTP).

The usbip error is due to a misformat in `usb.ids`:

> \# Syntax:
> \# vendor  vendor_name
> \#	device  device_name				<-- single tab
> \#		interface  interface_name		<-- two tabs
> 23170  8087  Intel Corp.
> 23179                  **8087** 07da  Centrino Advanced-N 6235

Happens with Ubuntu's v0.354-1 hwdata. Didn't bother too much about that...

closed 2021-12-14 by akuropka