dorssel/usbipd-win #166: No Serial Port

szafrir, 2021-12-18:
The attached usb has no serial port associated with it.  Can't be used with MCU such as ESP32 unless a new kernel is built as described in here [https://askubuntu.com/questions/1373910/ch340-serial-device-doesnt-appear-in-dev-wsl/](url)

Could be nice if it was fixed without a custom Kernel....

dorssel, 2021-12-18:
@szafrir 
That is not something `usbipd-win` can fix. Maybe you could report this to https://github.com/microsoft/WSL2-Linux-Kernel.

szafrir, 2021-12-18:
> @szafrir That is not something `usbipd-win` can fix.

@dorssel  Completely understood, yet maybe your friends at MS will listen to you as a fix may advance both of your nice projects...

dorssel, 2021-12-18:
@benmcmorran 
"They are calling us friends now..."

szafrir, 2021-12-18:
@dorssel btw - same thing as here for #167 - USB Flash drives seem to require a custom kernel.
If the MS guys decide to write about you, the least they can do is get their kernel to support the USB devices...

dorssel, 2021-12-18:
@szafrir 
Thanks for reporting.

These are known issues with the WSL kernel. I am sure they have been reported to the WSL project already. I suggest you add your +1 there. I'm sorry, but reporting it here will probably not get noticed by Microsoft. And I am not affiliated with Microsoft.

I would love to help you with any `usbipd-win` problems you may experience, if you have any. If not, is it OK to close this issue?

szafrir, 2021-12-19:
@dorssel - It works perfectly with the custom kernel (with the USB on), except for the problem with v1.2 that I reported in #167 .

Thank you.

closed 2021-12-19 by szafrir