dorssel/usbipd-win #167: USB Drive cannot be attached

szafrir, 2021-12-18:
When trying to attach a USB drive the following error is triggered: "usbipd: error: There is no compatible device with busid [id]"

dorssel, 2021-12-18:
@szafrir 
Can you please run with debug logging, see https://github.com/dorssel/usbipd-win/wiki/Troubleshooting.

We've had some attach problems in the past, plus there is a regression 1.2.0 -> 1.3.0. You can try either reverting back to 1.2.0, or try the latest master build, which can be found here: https://github.com/dorssel/usbipd-win/actions/runs/1566987611.

Please report if it makes any difference.

szafrir, 2021-12-18:
@dorssel - Thank you.  1.2.0 Works.

here is the logging for 1.3.0:
dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started
dbug: UsbIpServer.Server[1000]
      new connection from 172.17.91.18
dbug: UsbIpServer.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
fail: UsbIpServer.ConnectedClient[3]
      An exception occurred while communicating with the client:
      System.IO.FileNotFoundException: Unable to find the specified file.
         at UsbIpServer.ConfigurationManager.GetVBoxDevice(BusId busId) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConfigurationManager.cs:line 295
         at UsbIpServer.VBoxUsbMon.ClaimDeviceOnce(ExportedDevice device) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 141
         at UsbIpServer.VBoxUsbMon.ClaimDevice(ExportedDevice device) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 151
         at UsbIpServer.VBoxUsbMon.ClaimDevice(ExportedDevice device) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 157
         at UsbIpServer.ConnectedClient.HandleRequestImportAsync(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConnectedClient.cs:line 139
dbug: UsbIpServer.Server[1000]
      connection closed

szafrir, 2021-12-18:
@dorssel  one last bug... the service has to be restarted after every detach of the usb drive.  I don't know if this bug was fixed in 1.3.0.
Otherwise everything works now.  Thank you.

dorssel, 2021-12-20:
@szafrir 

> the service has to be restarted after every detach of the usb drive

Do I understand correctly that the `usbipd-win` service crashes? That is certainly a bug. It may have to do with the way we detect unplug events. That has changed since 1.3.0. And then again, in the master tree.

Can you please try the latest master build to see if it fixed now? It can be found here: https://github.com/dorssel/usbipd-win/actions/runs/1566987611.

21494, 2021-12-29:
same here with last version:

```
usbip: error: Attach Request for 2-7 failed - Request Failed
usbipd: error: Failed to attach device with ID '2-7'.
```

lOGS:

```
fail: UsbIpServer.ConnectedClient[3]
      An exception occurred while communicating with the client:
      System.IO.FileNotFoundException: Unable to find the specified file.
         at UsbIpServer.ConfigurationManager.GetVBoxDevice(BusId busId) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConfigurationManager.cs:line 295
         at UsbIpServer.VBoxUsbMon.ClaimDeviceOnce(ExportedDevice device) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 141
         at UsbIpServer.VBoxUsbMon.ClaimDevice(ExportedDevice device) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 151
         at UsbIpServer.VBoxUsbMon.ClaimDevice(ExportedDevice device) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 157
         at UsbIpServer.ConnectedClient.HandleRequestImportAsync(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConnectedClient.cs:line 148
```

dorssel, 2021-12-29:
@21494 
Could you please post the output of:

```
reg query "HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}" /v UpperFilters
```

21494, 2021-12-29:
` UpperFilters    REG_MULTI_SZ    USBPcap`
I would like to keep USBPcap
Thank you

dorssel, 2021-12-29:
@21494 
Then you'll have to wait a few weeks. I'm working on a change that will work around this problem, but filters like usbpcap are incompatible with the current version of usbipd-win.

szafrir, 2021-12-30:
@dorssel - sorry for the long silence.  I tried the master build.  Same problem as in 1.3.0 - It can't attach the device at all.  Here is the log:

dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started
dbug: UsbIpServer.Server[1000]
      new connection from 172.17.88.170
dbug: UsbIpServer.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
fail: UsbIpServer.ConnectedClient[3]
      An exception occurred while communicating with the client:
      System.IO.FileNotFoundException: Unable to find the specified file.
         at UsbIpServer.ConfigurationManager.GetVBoxDevice(BusId busId) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConfigurationManager.cs:line 295
         at UsbIpServer.VBoxUsbMon.ClaimDeviceOnce(ExportedDevice device) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 141
         at UsbIpServer.VBoxUsbMon.ClaimDevice(ExportedDevice device) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 151
         at UsbIpServer.VBoxUsbMon.ClaimDevice(ExportedDevice device) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 157
         at UsbIpServer.ConnectedClient.HandleRequestImportAsync(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConnectedClient.cs:line 148
dbug: UsbIpServer.Server[1000]
      connection closed

dorssel, 2021-12-30:
@szafrir 
OK, just to make sure:
- 1.2.0 works, but you have to restart the service after detach
- 1.3.0 gives a "FileNotFound" exception during attach
  After the "FileNotFound" error, the device is disabled and needs to be re-plugged before it works again.
- current master build also gives a "FileNotFound" exception during attach
  Is the replugging necessary here as well (I suspect it is not, this was changed between 1.3.0 and current master).
 
Is that a correct summary?

Could you please post the output of:
```
reg query "HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}" /v UpperFilters
```

szafrir, 2021-12-30:
@dorssel Correct - except that reinserting the USB drive is still required under 1.3.1 :   see below
[image]

szafrir, 2021-12-30:
@dorssel  the registry key doesn't seem to be there in v1.3.1

dorssel, 2021-12-30:
@szafrir 
Interesting. Can you find the 7-4 device in Windows Device Manager? Select it (single click) and then "View by connection". Then do the (failed) attach on 1.3.1.
- What hub(s) is the device connected to?
- What is the USB version of the device and of the hub? (2/3/3.1)
- What is the state (error code in device manager properties) after the attach failed?

szafrir, 2021-12-30:
@dorssel 
first the HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000} exists. when I search it without the /v UpperFilters.

Here is the device:
 
[image]
The device is usb v3.1
After the failed attach the device is doesn't show in the device manager at all.
I don't know if it means anything but I have this warning event on the device:
[image]

dorssel, 2021-12-30:
@szafrir 
The UpperFilters not existing (or being empty) is a good thing. The fact that it does not exist at all just means that you never had a single driver (not even in the past) that installed a filter.

That's a lot of hubs in sequence. Is that some kind of docking device? Could you try the same device in a port "closer" to the root controller? Does this machine have a root that is not 3.1 that you could try, maybe a USB 2 port? Because I suspect it is some kind of incompatibility with a 3.1 hub.

And as a second test. What happens if you "disable device" in device manager? Because that is how 1.3.0 works: disable device, change driver, re-enable device. In the current master it does something different: pause device, change driver, unpause device. That works with devices that are disabled to begin with as well. If you are able to disable the device manually, can you try attach from the current master again?

szafrir, 2021-12-30:
@dorssel 
On 1.3.1
First test - Same results, on the root as well.
Second test - attachment worked and drivers replaced, both on the root and through the nested hubs.

This is the root:
[image]

szafrir, 2021-12-30:
@dorssel 
Another phenomenon - the device is removed after detach.  i.e. It has to be physically removed and reinserted before another attach.  The service, though, doesn't have to be restarted.

dorssel, 2021-12-30:
@szafrir 
Thanks for all the testing. So it appears that both "disable-change-enable" (1.3.0) and "pause-change-unpause" (1.3.1) do not work programmatically, but "manual disable" + "pause-change-unpause" does work. And then the reverse makes the device disappear...

And then there is: 1.2.0 worked (!). All the changes in 1.3.x are to circumvent the problems other people had with the 1.2.0 strategy: USB Reset Port. You are the first to report that 1.2.0 works, but 1.3.1 (master) does not...

I think it may have to do with "reboot required". That's what I am currently suppressing and ignoring altogether. I think this device just doesn't like to be dynamically configured.

I am working on a generic workaround for all of this: permanently changing the driver. In that case the device is never available to Windows, it is always assigned to the sharing driver. Quite drastic, but works always, even if reboot is required upon binding. If you are interested, see #164.

I have one more question: can you figure out if this is device or machine related? In other words, does this happen for all devices on this computer, or only for that Mass Storage Device?

szafrir, 2021-12-30:
@dorssel 
Here is something you may find interesting:
Since the whole thing is a bit strange... I tried a Samsung USB Drive.  It worked with no issues at all.  Including the detach.
SanDisk works only when disabled and the detach does half the job (it removes your driver but doesn't install back the original)

I hope it helps...

dorssel, 2021-12-30:
@szafrir 
OK, so it is device specific. I think we'll leave it here; not much more I can do about it now. Once the '--force' option is released you may give that a try. But I am afraid this SanDisk device just doesn't play nice...
Thanks for the testing and accurate reports!

szafrir, 2021-12-30:
@dorssel 
I tried another SanDisk - same result.
Thank you for investing in this nice software!  I really like it because I used to forward COM to tcp-rfc2217 to flash MCUs (which I program as a hobby) and now you made it much simpler.

mi-hol, 2022-01-04:
@szafrir with the learning that only SanDisk devices are affected maybe you could change the issue title to make this obvious?

dorssel, 2022-01-17:
@szafrir 
I've just finished a PR that may solve your problem with the SanDisk. See #164, the installer is at https://github.com/dorssel/usbipd-win/actions/runs/1705631433. Could you give that a go?
You will have to "force-bind" the drive with "usbipd bind --busid BUSID --force".
Also, once bound (forced or not), wsl attach/detach no longer require admin rights...

dorssel, 2022-01-23:
@szafrir 
Could you please try PR #224, the installer is at https://github.com/dorssel/usbipd-win/actions/runs/1734739747.

dorssel, 2022-02-11:
@szafrir 
Could you please test the SanDisk with latest release (version 2.1.0)?

szafrir, 2022-02-19:
@dorssel  sorry for the long silence...

No luck with 2.1 : 
[image]

dorssel, 2022-02-19:
@szafrir 
This is with `bind --force`?

szafrir, 2022-02-19:
@dorssel when I just do the bind --force it makes the device disappear.  I'm not sure I'm doing things correctly - It has been a while...
[image]

dorssel, 2022-02-19:
@szafrir 
For this "strange" device, we expect that. `bind --force` will immediately set the driver. That also resets the port, which causes the device to disappear (probably it just crashes internally). But a reboot should fix that. Then, after reboot, the VBoxUsb driver should be loaded, and the attach should work (in theory).

szafrir, 2022-02-19:
@dorssel - I can give a reboot a try.  Obviously it is not a solution but it can test your theory.

szafrir, 2022-02-19:
@dorssel 
Negative.  Same thing.  Surprisingly, when after boot the device was not visible.  I had to remove and reinsert it to see it with "usbipd wsl list"

szafrir, 2022-02-19:
@dorssel  BTW - It didn't get to the stage when it asks for the wsl password

szafrir, 2022-02-19:
@dorssel I don't know if it may help - but here is the error event (Port reset Failed):

Device settings for USB\VID_0000&PID_0001\b&3aae6739&0&2 were not migrated from previous OS installation due to partial or ambiguous device match.

Last Device Instance Id: USB\VID_03F0&PID_8140\070A99A3E09B9455
Class Guid: {36fc9e60-c465-11cf-8056-444553540000}
Location Path: 
Migration Rank: 0xF000FFFFFFFFF132
Present: false
Status: 0xC0000719

dorssel, 2022-02-19:
Thanks for trying. Too bad it didn't work.

szafrir, 2022-02-19:
Thank you.  Maybe with the next version :)

dorssel, 2022-03-26:
@szafrir 
Version 2.2.0 has been released. Could you give that SanDisk another try? Don't think it has been solved, but you never know...

szafrir, 2022-03-30:
@dorssel  I'm travelling.  I'll try in a couple of weeks...

dorssel, 2022-05-13:
@szafrir 
Version 2.3.0 has been released. Since 2.2.0 there is new port reset behavior upon attach/detach. Could you give that a try?

Nehcknarf, 2022-05-23:
> @szafrir Version 2.3.0 has been released. Since 2.2.0 there is new port reset behavior upon attach/detach. Could you give that a try?

Hi, I also have a SanDisk USB flash drive, it‘s still not work with version 2.3.0, failed to attach device, and popup "USB Device not recognized" in Windows. rerun "usbipd wsl list" it says unknown USB device (port reset failed).
Env: Win11 with WSL2 preview + Ubuntu 22.04
```
PS C:\Users\Nehcknarf> usbipd --version
2.3.0+42.Branch.master.Sha.3d9f5c5acc4e133ab8147684ad1463cbaec43240
PS C:\Users\Nehcknarf> usbipd wsl list
BUSID  VID:PID    DEVICE                                                        STATE
1-8    046d:c52b  Logitech USB Input Device, USB 输入设备                       Not attached
1-14   0cf3:e009  Qualcomm QCA9377 Bluetooth                                    Not attached
1-17   0781:5591  USB 大容量存储设备                                            Not attached

PS C:\Users\Nehcknarf> usbipd wsl attach --busid=1-17
usbipd: info: Using default distribution 'Ubuntu-22.04'.
usbip: error: Attach Request for 1-17 failed - Device in error state

usbipd: error: Failed to attach device with busid '1-17'.
PS C:\Users\Nehcknarf> usbipd wsl list
BUSID  VID:PID    DEVICE                                                        STATE
1-8    046d:c52b  Logitech USB Input Device, USB 输入设备                       Not attached
1-14   0cf3:e009  Qualcomm QCA9377 Bluetooth                                    Not attached
1-17   0000:0001  未知 USB 设备(端口重置失败)                                   Not attached
```

dorssel, 2022-05-26:
@Nehcknarf 
Thanks for these details. Interesting to see that the device ID changes from 0781:5591 to 0000:0001. Obviously, this device has some weird initialization mode, maybe for firmware updates, or else maybe because of the password function it has. Can you try `usbview.exe` on the device when it is in this state? See https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/usbview.

dorssel, 2022-05-27:
@Nehcknarf 
Thanks, this confirms that the device locks up completely after a port reset (returns 0x00 for everything). Not much we can do about that. The port reset is required to simulate an unplug/replug to let the VBoxUsb driver take over. You could try using this device in a USB2 port instead of a USB3 port ... but no guarantees ...

I'll add the SanDisk Ultra Flair (0781:5591) and SanDisk Ultra Fit (0781:5583) as incompatible to the list ... sorry ...

dorssel, 2022-05-27:
Added as incompatible in https://github.com/dorssel/usbipd-win/wiki/Tested-Devices.
Not solvable ... Closing ...

closed 2022-05-27 by dorssel

Nehcknarf, 2022-05-27:
> @Nehcknarf Thanks, this confirms that the device locks up completely after a port reset (returns 0x00 for everything). Not much we can do about that. The port reset is required to simulate an [...]

Thanks for the clue, it works at USB2.0 port!
`PS C:\Users\Nehcknarf> usbipd wsl list
BUSID  VID:PID    DEVICE                                                        STATE
1-7    0781:5591  USB 大容量存储设备                                            Not attached
1-8    046d:c52b  Logitech USB Input Device, USB 输入设备                       Not attached
1-14   0cf3:e009  Qualcomm QCA9377 Bluetooth                                    Not attached

PS C:\Users\Nehcknarf> usbipd wsl attach --busid=1-7
usbipd: info: Using default distribution 'Ubuntu-22.04'.
PS C:\Users\Nehcknarf> usbipd wsl list
BUSID  VID:PID    DEVICE                                                        STATE
1-7    0781:5591  USB 大容量存储设备                                            Attached - Ubuntu-22.04
1-8    046d:c52b  Logitech USB Input Device, USB 输入设备                       Not attached
1-14   0cf3:e009  Qualcomm QCA9377 Bluetooth                                    Not attached`
`(base) nehcknarf@DESKTOP-TFD98TM:~$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 0781:5591 SanDisk Corp. Ultra Flair
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`

dorssel, 2022-05-27:
@Nehcknarf 
Wow, didn't expect that, to be honest ... 
I'll add that info to the device list.