dorssel/usbipd-win #169: Unable to connect USB device - Device Busy (exported)

burpitt, 2021-12-21:
`        2-1: JMicron Technology Corp. / JMicron USA Technology Corp. : JM20329 SATA Bridge (152d:2329)
           : USB\VID_152D&PID_2329\1ADFFFFFFFFF
           : (Defined at Interface level) (00/00/00)
           :  0 - Mass Storage / SCSI / Bulk-Only (08/06/50)`

[image]

This is a USB connected device that allows for the mounting of SATA HDD devices, has a speaker (!), and numerous female USB ports. The SATA connector is above as shown. The SATA is on Bus 2-1 and the rest on 2-3, 2-4, 2-5 etc. 

Once I have it listing on **lsusb** I do not seem to be able to locate it anywhere else. I have tried all the usual commands (**fdisk -l** for e.g.) to list the devices and/or mount but not had any luck. Am I missing something obvious or is this more a case of device/driver support?

The aforementioned USB device works as it should in Windows.

Many thanks,

dorssel, 2021-12-21:
The issue title is

> Unable to connect USB device - Device Busy (exported)

This means the device is in use by the host (or already attached by another or even the same client), by some program/service that opened it exclusively. One possible solution is to disable the device in device manager and use the current master build (which allows sharing of devices that are disabled on the host). It can be found here: https://github.com/dorssel/usbipd-win/actions/runs/1566987611.

Then you report that the device is listed in `lsusb` (is this another device? or did you already fix the title issue?)

Are you using WSL? Then it is possibly missing drivers. See https://github.com/dorssel/usbipd-win/wiki/WSL-support.

burpitt, 2021-12-21:
I am assuming it is a driver issue yes that is why I put the info in the code at the top. Are you suggesting I try building the kernal method at the end of your second link? I have not gone that far yet I just wanted to check I am not missing anything obvious. Presumably a listed USB storage device on the busid *should* show on fdisk?

Meanwhile I will give [that](https://github.com/dorssel/usbipd-win/wiki/WSL-support#building-your-own-usbip-enabled-wsl-2-kernel) a go. The device works in Windows just I cannot get further that lsusb listing on WSL....

Thanks.

dorssel, 2021-12-21:
@burpitt 
Yes, WSL currently only has drivers for USB-to-serial (a few, not even all of them). Audio and SATA certainly not. Those work fine (or so they should) with `usbipd-win`, but you either need a full distro kernel (via Hyper-V for example) or compile your own WSL kernel.

Other people have requested more drivers in WSL, see for example https://github.com/microsoft/WSL/issues/7770.  This is not something `usbipd-win` can solve.

burpitt, 2021-12-21:
OK that answers a lot! Thank you for the explanation. I will see if I can solve it with 'usbipb-win' - if not it is time to get building tomorrow.

Thanks again.

closed 2021-12-22 by burpitt