dorssel/usbipd-win #172: WSL setup linux-tools

XOverMindX, 2021-12-26:
issue installing package during WSL setup process

root@XJ-9:/mnt/c/WINDOWS/system32# sudo apt install linux-tools-5.4.0-77-generic hwdata
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package linux-tools-5.4.0-77-generic
root@XJ-9:/mnt/c/WINDOWS/system32#

ran from inside Kali as well 

┌──(overmind㉿XJ-9)-[~]
└─$ sudo apt install linux-tools-5.4.0-77-generic hwdata
[sudo] password for overmind: 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package linux-tools-5.4.0-77-generic
E: Couldn't find any package by glob 'linux-tools-5.4.0-77-generic'

XOverMindX, 2021-12-26:
nvm had to enable windows product updates and got newest version of WSL kernel everything works as advertised now.

closed 2021-12-26 by XOverMindX

Chris2011, 2021-12-29:
Did you solved it in WSL 2 or 1? I still got the issue, I updated WSL kernel to the newest one. I have Ubuntu 21.10 and everything is up to date.

dorssel, 2021-12-29:
@Chris2011 
`usbipd-win` only works for WSL 2, a stated in the documentation. And Ubuntu 21.10 has its own topic: #67. And replying to a closed issue is not the fastest way to get help.