dorssel/usbipd-win #173: libusbip: error: when attach device to WSL2

sudashannon, 2021-12-27:
When trying to attach an allwinner SOC named v3s in its fel mode, the error happens, after it, I have tried to attach some other devices, but also failed as the same reason.

PS C:\Users\Shannon> usbipd wsl attach --busid 4-14
[sudo] password for sdshannon:
libusbip: error: udev_device_new_from_subsystem_sysname failed
usbip: error: open vhci_driver
usbipd: error: Failed to attach device with ID '4-14'.

PS C:\Users\Shannon> usbipd server Logging:LogLevel:Default=Trace
dbug: Microsoft.Extensions.Hosting.Internal.Host[1]
      Hosting starting
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\Users\Shannon
dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started
dbug: UsbIpServer.Server[1000]
      new connection from 172.21.202.60
dbug: UsbIpServer.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
info: UsbIpServer.ConnectedClient[1]
      Client 172.21.202.60 claimed device at 4-14 (USB\VID_1F3A&PID_EFE8\5&330350CB&0&14).
dbug: UsbIpServer.ConnectedClient[1000]
      Unbind while attached
info: UsbIpServer.ConnectedClient[2]
      Client 172.21.202.60 released device at 4-14 (USB\VID_1F3A&PID_EFE8\5&330350CB&0&14).
dbug: UsbIpServer.Server[1000]
      connection closed

dorssel, 2021-12-27:
> usbip: error: open vhci_driver

Have you followed https://github.com/dorssel/usbipd-win/wiki/WSL-support? You need a WSL kernel with USBIP support, yours is too old.

sudashannon, 2021-12-30:
> > usbip: error: open vhci_driver
> 
> Have you followed https://github.com/dorssel/usbipd-win/wiki/WSL-support? You need a WSL kernel with USBIP support, yours is too old.

Thank u, I updated my wsl's kernel and it works properly now.

closed 2021-12-30 by sudashannon