dorssel/usbipd-win #174: USBIP Device Host Service stopped running(may cause automatic disconnection if not used for a period of time)

SunJun8, 2021-12-30:
OS: Windows 11 Pro 22000.376

Version: 1.3.0

Scenes: Share FT2232 to wsl2

Q：USBIP Device Host Service stopped running(may cause automatic disconnection if not used for a period of time)

dorssel, 2021-12-30:
@SunJun8 
Thank you for the report. Can you find an error/crash report in the Windows Eventlog for this? It could give me a pointer to why it crashed. Was the computer "on" the whole time, or was there any standby/hibernation involved?

SunJun8, 2021-12-31:
@dorssel 
Sorry I don’t know where the crash log is saved, and the computer does not enter hibernation or standby

SunJun8, 2021-12-31:
[image]
[image]
[image]

SunJun8, 2021-12-31:
Unplug the usb interface without detach, it will probably cause the USBIP Device host to stop service

dorssel, 2021-12-31:
@SunJun8 
Yes, I understand. It probably has to do with unplugging. Can you please try with the latest master build? The code to deal with unplug detection ha changed, so this bug may have been fixed already. You can find it here: https://github.com/dorssel/usbipd-win/actions/runs/1613336236.

And I think there should be something logged on the Windows side, in Windows Event Viewer (my best guess: 事件查看器).

SunJun8, 2022-01-04:
Thanks, 1.3.1 work well

closed 2022-01-04 by SunJun8