dorssel/usbipd-win #176: usbip: error: Attach Request for <busid> failed - Request Failed

hsiboy, 2021-12-30:
Windows Pro 11 - Insider Preview
```PS C:\Program Files\PowerShell\7> [System.Environment]::OSVersion.Version

Major  Minor  Build  Revision
-----  -----  -----  --------
10     0      22523  0
```

WSL2
```
Description:    Ubuntu 18.04.6 LTS
Release:        18.04
Codename:       bionic
```

List USB devices:

```
PS C:\Program Files\PowerShell\7> usbipd wsl list
BUSID  DEVICE                                                        STATE
3-3    USB Mass Storage Device                                       Not attached
3-4    Iomega USB Zip 100                                            Not attached
3-9    Intel(R) Wireless Bluetooth(R)                                Not attached
3-11   Integrated Webcam                                             Not attached
4-2    USB Input Device                                              Not attached
PS C:\Program Files\PowerShell\7>
```

attach the Zip Drive

```
PS C:\Program Files\PowerShell\7> usbipd wsl attach --busid 3-4
[sudo] password for pi:
usbip: error: Attach Request for 3-4 failed - Request Failed

usbipd: error: Failed to attach device with ID '3-4'.
```

Check the server log:

```
PS C:\Program Files\PowerShell\7> usbipd server
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\Program Files\PowerShell\7
fail: UsbIpServer.ConnectedClient[3]
      An exception occurred while communicating with the client:
      System.IO.FileNotFoundException: Unable to find the specified file.
         at UsbIpServer.ConfigurationManager.GetVBoxDevice(BusId busId) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConfigurationManager.cs:line 295
         at UsbIpServer.VBoxUsbMon.ClaimDeviceOnce(ExportedDevice device) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 141
         at UsbIpServer.VBoxUsbMon.ClaimDevice(ExportedDevice device) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 151
         at UsbIpServer.VBoxUsbMon.ClaimDevice(ExportedDevice device) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 157
         at UsbIpServer.ConnectedClient.HandleRequestImportAsync(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConnectedClient.cs:line 139
```

Drive D: ?? lol wut? VBox, is VirtualBox no? very confused about what i'm seeing in the logs.

Please advise on how best to troubleshoot.

Cheers.

hsiboy, 2021-12-30:
i've just read #118 

For brevity...

```
PS C:\Program Files\PowerShell\7> reg query "HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}" /v UpperFilters

HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}
    UpperFilters    REG_MULTI_SZ    UsbDk
```

dorssel, 2021-12-30:
@hsiboy 
Yes, UsbDk is known to be incompatible with the current version of `usbipd-win`. For now, you will have to remove UsbDk.
There is work in progress to work around this, but you will have to wait a few weeks before that is ready.

hsiboy, 2021-12-31:
Hey @dorssel I don't know how UsbDk got installed, but given the nature of what I do (tinker) I'm not surprised it is installed.

I don't know what will break if I remove it.

Might be worth adding some upper filter detection to the installer and alerting the user to incompatibles.

dorssel, 2021-12-31:
@hsiboy 
The detection is already in the works (I was already thinking the same thing).

UsbDk is for driver development. Unless you're actively using it, it can be safely uninstalled.

hsiboy, 2021-12-31:
Cool.

Ok, let's close this issue then 👍

PR #184 (2022-01-05): Add warning for incompatible UpperFilters
Fixes #176
closed 2022-01-05 by dorssel