dorssel/usbipd-win #178: usbipd-win 1.3.0: Windows 11 (OS build 22000.376), WSL 2 (Ubuntu 20.04 LTS-Linux kernel 5.10.60.1-microsoft-standard-WSL2) error connecting external USB drive

key4josh, 2021-12-31:
The following is the command sequence.

```
PS C:\Windows\system32> usbipd wsl list
BUSID  DEVICE                                                        STATE
1-3    Integrated Camera                                             Not attached
1-6    USB Attached SCSI (UAS) Mass Storage Device                   Not attached
2-3    Synaptics UWP WBDI                                            Not attached
2-4    Realtek Bluetooth Adapter                                     Not attached
PS C:\Windows\system32> usbipd wsl attach --busid=1-6
[sudo] password for abc:
usbip: error: tcp connect
usbipd: error: Failed to attach device with ID '1-6'.
```

dorssel, 2021-12-31:
@key4josh 
Are you running a third party firewall?

key4josh, 2022-01-02:
Yes, it turns out I was.
- The USB device got attached once I turned off the Firewall. However, couldn't figure what next I was supposed to do to access the content of this external USB drive.
 - I used the following instead that did mount the USB in linux as I intended.
 `PS C:\Windows\system32> GET-CimInstance -query "Select * from Win32_DiskDrive"

DeviceID           Caption                          Partitions Size         Model
--------           -------                          ---------- ----         -----
\\.\PHYSICALDRIVE0 WDC PC SN530 SDBPMPZ-512G-1101   3          512105932800 WDC PC SN530 SDBPMPZ-512G-1101
\\.\PHYSICALDRIVE1 Samsung PSSD T7 SCSI Disk Device 1          500105249280 Samsung PSSD T7 SCSI Disk Device

PS C:\Windows\system32> wsl --mount \\.\PHYSICALDRIVE1 --partition 1`
this made the external USB available in ubuntu.

closed 2022-01-02 by key4josh