dorssel/usbipd-win #180: USB scanner disconnects after scanning completes

tiborbaksa, 2022-01-01:
I have a Canon CanoScan LiDE 50 flatbed scanner that I'm trying to use from WSL2 with SANE.

1. I attach the scanner with `usbipd wsl attach --busid 1-1` and I can see the Linux system recognizes the device:

    ```
    $ dmesg
    ...
    [ 1370.392022] vhci_hcd vhci_hcd.0: pdev(0) rhport(0) sockfd(3)
    [ 1370.392078] vhci_hcd vhci_hcd.0: devid(65537) speed(3) speed_str(high-speed)
    [ 1370.392176] vhci_hcd vhci_hcd.0: Device attached
    [ 1370.752489] usb 1-1: new high-speed USB device number 2 using vhci_hcd
    [ 1370.921854] usb 1-1: SetAddress Request (2) to port 0
    [ 1370.973832] usb 1-1: New USB device found, idVendor=04a9, idProduct=2213, bcdDevice= 3.05
    [ 1370.973875] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
    [ 1370.973878] usb 1-1: Product: CanoScan
    [ 1370.973880] usb 1-1: Manufacturer: Canon
    
    $ sudo scanimage -L
    device `genesys:libusb:001:002' is a Canon LiDE 35/40/50 flatbed scanner
    ```

2. I successfully scan a single image, but just after that the device disconnects:

    ```
    $ sudo scanimage --format png > image.png
    $ sudo scanimage -L
    
    No scanners were identified. If you were expecting something different,
    check that the scanner is plugged in, turned on and detected by the
    sane-find-scanner tool (if appropriate). Please read the documentation
    which came with this software (README, FAQ, manpages).
    $ dmesg
    ...
    [ 1775.212280] usb usb1-port1: Cannot enable. Maybe the USB cable is bad?
    [ 1776.172058] usb usb1-port1: Cannot enable. Maybe the USB cable is bad?
    [ 1777.131838] usb usb1-port1: Cannot enable. Maybe the USB cable is bad?
    [ 1778.091866] usb usb1-port1: Cannot enable. Maybe the USB cable is bad?
    [ 1778.091949] usb 1-1: USB disconnect, device number 2
    [ 1779.051874] usb usb1-port1: Cannot enable. Maybe the USB cable is bad?
    [ 1780.011859] usb usb1-port1: Cannot enable. Maybe the USB cable is bad?
    [ 1780.011869] usb usb1-port1: attempt power cycle
    [ 1781.011843] usb usb1-port1: Cannot enable. Maybe the USB cable is bad?
    [ 1781.972067] usb usb1-port1: Cannot enable. Maybe the USB cable is bad?
    [ 1781.974005] usb usb1-port1: unable to enumerate USB device
    ```

3. At this point `usbipd` shows that the device is still attached:

    ```
    > usbipd wsl list
    BUSID  DEVICE                                                        STATE
    1-1    CanoScan                                                      Attached - Ubuntu
    ...
    ```

4. In order to use the scanner again I have to detach the device with `usbipd wsl detach --busid 1-1`:

    ```
    $ dmesg
    ...
    [ 1929.430151] vhci_hcd: connection closed
    [ 1929.430335] vhci_hcd: stop threads
    [ 1929.430338] vhci_hcd: release socket
    [ 1929.430342] vhci_hcd: disconnect device
    ```

5. I can attach the scanner again with `usbipd wsl attach --busid 1-1`. It assigns a new device number:

    ```
    $ dmesg
    ...
    [ 1935.952450] vhci_hcd vhci_hcd.0: pdev(0) rhport(0) sockfd(3)
    [ 1935.952454] vhci_hcd vhci_hcd.0: devid(65537) speed(3) speed_str(high-speed)
    [ 1935.952517] vhci_hcd vhci_hcd.0: Device attached
    [ 1936.301871] usb 1-1: new high-speed USB device number 7 using vhci_hcd
    [ 1936.451850] usb 1-1: SetAddress Request (7) to port 0
    [ 1936.534863] usb 1-1: New USB device found, idVendor=04a9, idProduct=2213, bcdDevice= 3.05
    [ 1936.534866] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
    [ 1936.534868] usb 1-1: Product: CanoScan
    [ 1936.534870] usb 1-1: Manufacturer: Canon
    
    $ sudo scanimage -L
    device `genesys:libusb:001:007' is a Canon LiDE 35/40/50 flatbed scanner
    ```

System information:
* Windows: Windows 10 Pro 21H2 build 19044.1415
* Linux: Ubuntu 20.04.3 LTS

dorssel, 2022-01-01:
@tiborbaksa 
Interesting. Possibly the device has some special "reset sequence" after each scan. Also, in the past we've had problems recovering from STALL situations. This could be another example of that.

Can you run `usbipd-win` with debug output as explained here: https://github.com/dorssel/usbipd-win/wiki/Troubleshooting?
You probably want to redirect to a file as the first (successful) scan will produce a lot of output. Furthermore, are there any other details in syslog before:

> [ 1775.212280] usb usb1-port1: Cannot enable. Maybe the USB cable is bad?

Usually the first error reported has a little more detail. You may have to install/start rsyslogd to get syslog working in WSL.

tiborbaksa, 2022-01-02:
I ran `usbipd` with Trace logging, see [usbipd.log](https://github.com/dorssel/usbipd-win/files/7799314/usbipd.log). I attached the scanner, scanned an image, then detached the scanner.

I also ran `rsyslogd` while doing this, see [syslog](https://github.com/dorssel/usbipd-win/files/7799341/syslog.log). Unfortunately I don't see more details that seems relevant.

dorssel, 2022-01-02:
@tiborbaksa 
It all looks good:
- Setup of USB device (endpoint 0) is fine. REMOTE_WAKE is masked out, just as it should.
- Setup of scan (endpoint 2) is fine.
- Read of scan (endpoint 1) is fine.
- There are no STALLs or errors of any kind.

But ... there is one oddity at the end of the log: even though both endpoint 1 and 2 complete all their requests, Linux is still sending a pipe reset (CLEAR_FEATURE) for both pipes. Normally, I only see that when a pipe has in fact stalled. But your pipes did not stall and still they get a reset. It could be that I am handling that situation wrong, I will have to dive into the USB documentation to see what is the required response when a pipe gets reset when it is *not* stalled.

dorssel, 2022-01-17:
@tiborbaksa 
I've just finished a PR that may solve your problem. See #164, the installer is at https://github.com/dorssel/usbipd-win/actions/runs/1705631433. Could you give that a go?
You will have to "force-bind" the scanner with "usbipd bind --busid <BUSID> --force".

SubstringTV, 2022-01-17:
Thanks @dorssel, this PR release helped me get an Android 10 device to attach which was disconnecting and reconnecting back to the host each time I tried to attach using version 1.3.0 while an Android 8.1 device was attaching fine.

dorssel, 2022-01-23:
@tiborbaksa 
Could you please try PR #224, the installer is at https://github.com/dorssel/usbipd-win/actions/runs/1734739747.
This could solve the problem (no guarantees...).

tiborbaksa, 2022-01-29:
@dorssel
Sorry for the late reply, I was out of town and couldn't test the proposed fixes earlier. Unfortunately neither versions seem to fix the issue. I tried both with and without the force bind option. Here are some trace logs from the [first](https://github.com/dorssel/usbipd-win/files/7964382/usbipd1.log) and the [second](https://github.com/dorssel/usbipd-win/files/7964383/usbipd2.log) versions with force bind.

dorssel, 2022-01-29:
@tiborbaksa 
Too bad. Thanks for testing!

dorssel, 2022-03-16:
@tiborbaksa 
An issue that may be related to yours was just fixed in the master build. Could you give https://github.com/dorssel/usbipd-win/actions/runs/1991685584 a try?

dorssel, 2022-03-26:
@tiborbaksa 
That last fix is now in the new 2.2.0 release. Could you give that a try?

tiborbaksa, 2022-04-18:
@dorssel
Sorry for the late reply again. I tested with 2.2.0 release and it fixes the issue. I can use the scanner repeatedly regardless of using the force bind option.

FYI I can see the following message appear after each scanning:

```
[  133.067440] usb 1-1: reset high-speed USB device number 2 using vhci_hcd
[  133.217422] usb 1-1: SetAddress Request (2) to port 0
```

closed 2022-04-18 by tiborbaksa