dorssel/usbipd-win #182: usbipd: error: Access denied

yorchAdalbert, 2022-01-02:
I'm trying to attach an Arduino Uno to my wsl. I did all the steps written on the Microsoft docs and after I type the command to attach the devide I got this error: usbipd: error: Access denied.

dorssel, 2022-01-02:
@yorchAdalbert 
You need administrator rights. Run the command from an elevated terminal.

c0d3r43v3r, 2022-01-10:
Hi There,
My situation is similar but different.
I have installed Ubuntu focal on wsl2 from a normal prompt.
I have configured vscode for remote development on that distro.
When i do <wsl -l> in a normal prompt i can see my distro.
When i do <wsl -l> in an admin prompt i can not see any distros.
So obviously when i do <usbipd wsl attach --busid XX> i am doing it from a normal prompt and i get access denied.
I can not do it in an admin prompt because it doesn't see the distro.

Any help will be appreciated..

dorssel, 2022-01-10:
@c0d3r43v3r 
You need an elevated command prompt *for the same user*, not for a different admin user. See https://github.com/dorssel/usbipd-win/discussions/105#discussioncomment-1673052. If your user is not part of the Administrator group, you cannot use `usbipd wsl` (but you can run `usbip` yourself from within WSL as an alternative).

c0d3r43v3r, 2022-01-10:
Thanks for your quick answer.
As i see it i have a windows logged-in user and an Ubuntu user in the WSL distro.
When i open a cmd as an administrator on Windows it can not see the distro,  so i should add what to what ?
I am confused, it seems like a windows WSL problem to begin with, i can see my distro in a regular cmd but not in an admin cmd.
If i could see my distro in an admin cmd then all of my problems would be solved.
Do you have an idea ?

dorssel, 2022-01-10:
@c0d3r43v3r 
There are 2 ways of "running as an administrator".
1) Your user is a member of the Administrator group. By default, this user runs with restricted rights. But with *elevated privileges* you obtain the required admin rights. You are still the same user, however. This is what nowadays most people mean when they say "run as an admin": elevated privileges.
2) Your user is only a member of the Standard user group. In this case "running as an administrator" means running as a totally different user that is part of the Administrator group. In this case, you (temporarily) become a totally different user.

In case 1: you are still the same user, but now with more access rights. You will be able to see your WSL instances.
In case 2: the other (admin) user has no access to the WSL instances of the original standard user.

So, now answer this question first: is your user a Standard user, or part of the Administrators group?

PS: None if this has to do with the Linux user, this is all about the Windows side of things.

c0d3r43v3r, 2022-01-11:
Thank you for the explanation, it is clear now i asked out IT to add my user to the Admin group.

c0d3r43v3r, 2022-01-12:
Hi there,
I managed to solve my access denied problem, the IT added my windows user to the admin group.
I opened a normal cmd and did <wsl> 
I opened another  cmd with admin privilege's on windows.
Did  <usbipd wsl list> saw that the Camera that i want to use has bus ID 2-17
Did <usbipd wsl attach --busid 2-17>  got no errors or verbose
Went to the normal wsl cmd and did lsusb and didn't see the device
Went to the admin cmd and did <usbipd wsl list> and saw the device as <Not attached>

Can you help me troubleshoot this ?
Thanks

dorssel, 2022-01-12:
@c0d3r43v3r 
Please run the server with verbose debugging, as explained here https://github.com/dorssel/usbipd-win/wiki/Troubleshooting.

c0d3r43v3r, 2022-01-12:
Hi, 
The output of <uspipd wsl attach --busid 2-17> is:

------>    Start of console
C:\windows\system32>usbipd server Logging:LogLevel:Default=Trace
dbug: Microsoft.Extensions.Hosting.Internal.Host[1]
      Hosting starting
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\windows\system32
dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started
dbug: UsbIpServer.Server[1000]
      new connection from 172.26.47.23
dbug: UsbIpServer.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
info: UsbIpServer.ConnectedClient[1]
      Client 172.26.47.23 claimed device at 2-17 (USB\VID_1409&PID_3240\4102987712).
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=25, flags=512, length=8, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 256 0 8
trce: UsbIpServer.AttachedClient[1001]
      Scheduled seqnum=25, pending count = 1
trce: UsbIpServer.AttachedClient[1001]
      actual: 8, requested: 8
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=26, flags=0, length=0, ep=0
trce: UsbIpServer.AttachedClient[1001]
      0 49 40 0 0
trce: UsbIpServer.AttachedClient[1001]
      Scheduled seqnum=26, pending count = 1
trce: UsbIpServer.AttachedClient[1001]
      actual: 0, requested: 0
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=27, flags=512, length=18, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 256 0 18
trce: UsbIpServer.AttachedClient[1001]
      Scheduled seqnum=27, pending count = 1
trce: UsbIpServer.AttachedClient[1001]
      actual: 18, requested: 18
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=28, flags=512, length=5, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 3840 0 5
trce: UsbIpServer.AttachedClient[1001]
      Scheduled seqnum=28, pending count = 1
trce: UsbIpServer.AttachedClient[1001]
      actual: 5, requested: 5
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=29, flags=512, length=22, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 3840 0 22
trce: UsbIpServer.AttachedClient[1001]
      Scheduled seqnum=29, pending count = 1
trce: UsbIpServer.AttachedClient[1001]
      actual: 22, requested: 22
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=30, flags=512, length=9, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 512 0 9
trce: UsbIpServer.AttachedClient[1001]
      Scheduled seqnum=30, pending count = 1
trce: UsbIpServer.AttachedClient[1001]
      actual: 9, requested: 9
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=31, flags=512, length=57, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 512 0 57
trce: UsbIpServer.AttachedClient[1001]
      Scheduled seqnum=31, pending count = 1
trce: UsbIpServer.AttachedClient[1001]
      actual: 57, requested: 57
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=32, flags=512, length=255, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 768 0 255
trce: UsbIpServer.AttachedClient[1001]
      Scheduled seqnum=32, pending count = 1
trce: UsbIpServer.AttachedClient[1001]
      actual: 4, requested: 255
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=33, flags=512, length=255, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 770 1033 255
trce: UsbIpServer.AttachedClient[1001]
      Scheduled seqnum=33, pending count = 1
trce: UsbIpServer.AttachedClient[1001]
      actual: 30, requested: 255
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=34, flags=512, length=255, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 769 1033 255
trce: UsbIpServer.AttachedClient[1001]
      Scheduled seqnum=34, pending count = 1
trce: UsbIpServer.AttachedClient[1001]
      actual: 40, requested: 255
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=35, flags=512, length=255, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 771 1033 255
trce: UsbIpServer.AttachedClient[1001]
      Scheduled seqnum=35, pending count = 1
trce: UsbIpServer.AttachedClient[1001]
      actual: 26, requested: 255
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=36, flags=0, length=0, ep=0
dbug: UsbIpServer.AttachedClient[1000]
      Trapped SET_CONFIGURATION: 1
trce: UsbIpServer.AttachedClient[1001]
      actual: 0, requested: 0
dbug: UsbIpServer.ConnectedClient[1000]
      Device removal detected
dbug: UsbIpServer.ConnectedClient[1000]
      Unbind while attached
info: UsbIpServer.ConnectedClient[2]
      Client 172.26.47.23 released device at 2-17 (USB\VID_1409&PID_3240\4102987712).
dbug: UsbIpServer.Server[1000]
      connection closed
<-------- End of console

Can you see what is preventing the Camera from being attached ?

dorssel, 2022-01-12:
@c0d3r43v3r 
The attach initially is fine, no error. Linux starts querying the device for details (as is normal, again no errors), and at the end enables the default device configuration:

> Trapped SET_CONFIGURATION: 1

So far so good. But then immediately:

> Device removal detected

This looks to me that the device resets itself (as if unplug/replug is happening). Normally, I see this only with programmable devices that first have some sort of "bootloader" mode, and then, after flashing, reboot and present themselves as a different device. Unfortunately, this is not something that `usbipd-win` can handle or fix. If the device indeed reboots when configured, then all we can do is conclude that it is incompatible with `usbipd-win`. There is no way to "re-attach" the device, as from a Windows point of view, this is now a "new" device, which starts again unconfigured. It would simply enter a config-reboot-config-reboot etc. loop.

c0d3r43v3r, 2022-01-13:
@dorssel 
Thanks for you reply, yes I noticed the disconnection also.
As I am developing Linux on WSL2 and I need to work with a USB camera i have no option but to switch to developing against a remote Linux machine and connect the camera there.
Thanks for you effort in this project, I hope it helped many, i am sure that we are not far from full integration of USB devices support in WSL...

dorssel, 2022-01-17:
@dorssel 
I've just finished a PR that may solve your problem. See #164, the installer is at https://github.com/dorssel/usbipd-win/actions/runs/1705631433. Could you give that a go?
You will have to "force-bind" the camera with "usbipd bind --busid --force".
Also, once bound (forced or not), `wsl attach/detach` no longer require admin rights...

c0d3r43v3r, 2022-01-17:
@dorssel
Sure, NP. I will check and get back to you with the results

c0d3r43v3r, 2022-01-17:
@dorssel 
I've removed the installation of usbipd from <Add Remove programs>
I've downloaded and installed the installer from <https://github.com/dorssel/usbipd-win/actions/runs/1705631433>
I've restarted
I've done: < usbipd wsl list >
I've done: < usbipd wsl attach --busid 2-17 >  and got the following error:  < usbipd: error: WSL 'usbip' client not correctly installed. >

What to do next ?

dorssel, 2022-01-17:
See the wiki. The WSL preparation has changed. Thx for testing!

dorssel, 2022-02-11:
Original issue has been fixed in 2.0.0.

closed 2022-02-11 by dorssel