dorssel/usbipd-win #186: can not attach adb interface

junzhangsd, 2022-01-06:
WSL2 
uname -a 
Linux Ampere-Host 5.10.60.1-microsoft-standard-WSL2 #1 SMP Wed Aug 25 23:20:18 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

Windows 11 Pro

d:\>usbipd wsl list
BUSID  DEVICE                                                        STATE
6-3    Android Composite ADB Interface                               Not attached

d:\>usbipd wsl attach --busid 6-3
[sudo] password for jun:
usbip: error: Attach Request for 6-3 failed - Request Failed

usbipd: error: Failed to attach device with ID '6-3'.

usbip-win is the latest version.

Is it a known issue, any workaround  for this?

dorssel, 2022-01-06:
@junzhangsd 
Could you run the server with debugging output and post the results? See https://github.com/dorssel/usbipd-win/wiki/Troubleshooting.

junzhangsd, 2022-01-06:
d:\>usbipd server
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: d:\
fail: UsbIpServer.ConnectedClient[3]
      An exception occurred while communicating with the client:
      UsbIpServer.ConfigurationManagerException (1003): CM_Disable_DevNode returned CR_REMOVE_VETOED
         at UsbIpServer.ConfigurationManager.ThrowOnError(CONFIGRET configRet, String function) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConfigurationManager.cs:line 42
         at UsbIpServer.ConfigurationManager.TemporarilyDisabledDevice..ctor(UInt32 deviceNode) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConfigurationManager.cs:line 323
         at UsbIpServer.ConfigurationManager.TemporarilyDisabledDevice..ctor(String instanceId) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConfigurationManager.cs:line 316
         at UsbIpServer.ConnectedClient.HandleRequestImportAsync(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConnectedClient.cs:line 136

dorssel, 2022-01-06:
@junzhangsd 
Have a look at https://github.com/dorssel/usbipd-win/issues/150#issuecomment-987261481.
I think you have the same issue.

junzhangsd, 2022-01-06:
Thanks for your response. 

I do not have any host application using adb interface though.

dorssel, 2022-01-06:
@junzhangsd 

> CR_REMOVE_VETOED

This is the error from Windows Device Manager that means "someone/something has denied access to the configuration of the device because it is in use". I don't know what is using the device, but *something* is using it. Could be a Windows service, or maybe even the driver itself. You could try removing the default Windows driver, which is not a problem if you *only* want to use the device under Linux.

junzhangsd, 2022-01-07:
I managed to attach Android Bootloader Interface to WSL2/Ubuntu as shown on the logs on both Windows and Linux sides.

However, on linux side
lsusb shows the interface. 
fastboot devices command did not enumerate the devices.

if I detach the interface, on windows side, fastboot works.
C:\Windows\System32>fastboot devices
89RY0AAZF       fastboot

Sounds like linux side fastboot command did not pipe into the interface.  How do we know where the issue lies?

=======Logs=============

C:\Windows\System32>usbipd wsl list
BUSID  DEVICE                                                        STATE
6-3    Android Bootloader Interface                                  Attached - Ubuntu-20.04

C:\Windows\System32>usbipd server
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\Windows\System32
info: UsbIpServer.ConnectedClient[1]
      Client 172.17.247.57 claimed device at 6-3 (USB\VID_18D1&PID_4EE0\89RY0AAZF).

jun@Ampere-Host:~/extra/pixel3/aosp-12$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 18d1:4ee0 Google Inc. Nexus 4 (bootloader)
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

jun@Ampere-Host:~/extra/pixel3/aosp-12$ fastboot devices

dorssel, 2022-01-07:
@junzhangsd 
See #60.
Please search the issue list, other people have solved very similar issues. Now that the device is attached, the problems are usually with WSL: missing drivers, access rights, `udev`, etc. Not something that `usbipd-win` can solve, it has done its job.

junzhangsd, 2022-01-07:
Thank you so much for your help. 

It is indeed linux side udev setup for adb interface. Now fastboot works fine on linux side.

Regards,

Jun

closed 2022-01-07 by junzhangsd