dorssel/usbipd-win #187: Parent "USB Input Device" is CR_NOT_DISABLEABLE but a required child device is disableable

beanpaisley, 2022-01-07:
"USB Input Device" i am trying to connect to is indeed not disableable with right click in Device Manager.
Unsure if this is equivalent to CR_NOT_DISABLEABLE log entry i got.
This Parent USB Input Device has a number of  child devices, some of which are disableable and some are not.

I would like to connect to one of the disableable child devices, however, it is not available for sharing separated from the parent USB Input Device.
Are there plans to allow sharing child devices ?? i.e. an option to list child devices for a given parent??

Also, are there plans to include all HID devices, whether connected through USB or not??

dorssel, 2022-01-07:
@beanpaisley 
`CR_NOT_DISABLEABLE` is a regression from 1.2.0 to 1.3.0. You may have more luck with 1.2.0. The problem has been fixed in master, so you may try that instead; you can pick a recent build from https://github.com/dorssel/usbipd-win/actions/workflows/build.yml. Note that other things have changed in master which are not ready for release yet, so 1.2.0 is a safer bet.

The USBIP protocol only allows for sharing "entire devices". You are asking about only sharing individual interfaces of what is called a COMPOSITE device. This is not supported (nor will it ever be), simply because the APIs do not exist for that, neither on the Windows nor on the Linux side. It simply can't be done.

Your question about HID devices is irrelevant to `usbipd`. USBIP is "device type agnostic", it does not share HID USB devices, it shares *any* USB device. From that low level, nothing special is being done for HID devices or for any other specific class of devices. So that answer is also a "no", `usbipd` shares USB devices, not non-USB devices.

beanpaisley, 2022-01-07:
@dorssel
Appreciate your help, got it working with 1.3.1

closed 2022-01-07 by beanpaisley