dorssel/usbipd-win #195: Shared but not Attached

m4l490n, 2022-01-10:
After executing `usbipd wsl attach --busid 10-1`, This is what I get when checking the list with `wsl`
```
C:\Users\mmalagon>usbipd wsl list
BUSID  DEVICE                                                        STATE
1-6    Qualcomm QCA61x4A Bluetooth                                   Not attached
1-10   Dell ControlVault w/o Fingerprint Sensor, Microsoft Usbcc...  Not attached
10-1   JLink CDC UART Port (COM5), J-Link driver, USB Mass Stora...  Not attached
10-3   USB Input Device                                              Not attached
```

And this is what I get when checking the list without `wsl`
```
C:\Users\mmalagon>usbipd list
Present:
BUSID  DEVICE                                                        STATE
1-6    Qualcomm QCA61x4A Bluetooth                                   Not shared
1-10   Dell ControlVault w/o Fingerprint Sensor, Microsoft Usbcc...  Not shared
10-1   JLink CDC UART Port (COM5), J-Link driver, USB Mass Stora...  Shared
10-3   USB Input Device                                              Not shared
```

Why is my device "Shared" but not "Attached"?

dorssel, 2022-01-11:
"Shared" means it has been marked for "able to be attached". It is the result of `usbipd bind`. The next step is that the guest should attach. `usbipd wsl attach` is essentially a macro that runs both these things: first it executes `usbipd bind` (which was successful) and then it runs `usbip attach` within WSL. If that fails, `usbipd wsl attach` should have displayed the error. But if the device was unplugged, or WSL was restarted, or the device was ejected by WSL, then the attach is done. Now the device is "shared" again, and you will have to re-attach. Was any error displayed after `usbipd wsl attach`? If not, then the attach was successful and later the device got detached. That would be normal. "sharing" is persistent, "attach" is session based and only lasts until detach.

m4l490n, 2022-01-13:
@dorssel I had to do this on ubuntu; `sudo apt install linux-tools-generic` which installed `/usr/lib/linux-tools-5.4.0-92` and I updated this with `sudo visudo`. Only then I was able to make this work.

It wasn't working with `sudo apt install linux-tools-5.4.0-77-generic hwdata` as mentioned in the [Connecting USB devices to WSL](https://devblogs.microsoft.com/commandline/connecting-usb-devices-to-wsl/) article.

Thanks for helping!

closed 2022-01-13 by m4l490n