dorssel/usbipd-win #196: Failed to connect USB mass storage on Windows 10 with recompiled USBIP kernel

brianddk, 2022-01-11:
I'm running Windows 10 Home Version `21H1` Build `19043.1415` with usbipd-win `1.3.0` from the releases page.  I'm trying to attach a USB thumb-drive with a FAT32 and EXT4 partition on it to mount under WSL.  I can't upgrade to Win 11 due to system requirements.
```
> ver

Microsoft Windows [Version 10.0.19043.1415]

> usbipd --version
usbipd-win
1.3.0+37.Branch.master.Sha.89a4e59f384d08d953380913735f59821e63fa1f

> wsl --list --verbose
  NAME          STATE           VERSION
* Ubuntu        Stopped         2
  wsl2-usbip    Stopped         2
```
I'm running WSL2 Ubuntu distribution, and performed [the kernel recompile steps](https://github.com/dorssel/usbipd-win/wiki/WSL-support#building-your-own-usbip-enabled-wsl-2-kernel) outlined in the wiki
```
> usbipd wsl list
BUSID  DEVICE                                                        STATE
1-1    USB Mass Storage Device                                       Not attached
1-7    USB Input Device                                              Not attached
1-8    USB Input Device                                              Not attached

> usbipd wsl attach --busid 1-1
[sudo] password for wslusr:

>
```
After the connection I get plenty of messages in `dmesg` (attached) but no new block devices, just my normal harddrives.
```
$ uname -r
5.10.60.1-microsoft-standard-WSL2+
$ lsblk
NAME MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
sda    8:0    0  256G  0 disk
sdb    8:16   0  256G  0 disk /
```
Thoughts on what I can try next?

### Attachments 
* [config.gz](https://github.com/dorssel/usbipd-win/files/7844378/config.gz) - from `/proc/config.gz` after loading the new kernel
* [dmesg.before.gz](https://github.com/dorssel/usbipd-win/files/7844380/dmesg.before.gz) - `dmesg` dump from before attach
* [dmesg.after.gz](https://github.com/dorssel/usbipd-win/files/7844379/dmesg.after.gz) - `dmesg` dump from after attach

dorssel, 2022-01-11:
@brianddk 
From your config.gz:

```
# CONFIG_USB_STORAGE is not set
```

The device is attached successfully, but you don't have a WSL-side driver for it.

brianddk, 2022-01-11:
Awesome... Will rebuild and report back

Thx

brianddk, 2022-01-11:
Thx, did the kernel rebuild with `CONFIG_USB_STORAGE` and it worked.

Thx.

closed 2022-01-11 by brianddk

ZeeD, 2022-03-16:
Just a question: I'm using a "standard" kernel (`5.10.60.1-microsoft-standard-WSL2`) and I see
```bash
$ zgrep CONFIG_USB_STORAGE /proc/config.gz
# CONFIG_USB_STORAGE is not set
```
Do I need to build a custom kernel to access the mass storage?

dorssel, 2022-03-16:
@ZeeD 
Yes you do.