dorssel/usbipd-win #198: Chance to see a usbip client with verified driver here?

maxdd, 2022-01-11:
Hello,
i've seen that the only available client for windows requires to enable windows "testsigning" mode.
To my knowledge there isn't any version with a properly signed driver.
I've seen you are using VBox drivers, do we have any chance to use it also for a usbip client?

dorssel, 2022-01-11:
@maxdd 
See https://github.com/dorssel/usbipd-win/discussions/191.
TL;DR: no.

maxdd, 2022-01-11:
Hi dorssel,
i'm also using cezanne but it has a huge downside of the "testsigning" mode for now even though something is moving https://github.com/cezanne/usbip-win/issues/171
I hoped for the driver to be able to support also a client version but by reading the issue https://github.com/dorssel/usbipd-win/issues/65 it seems it is a different beast.

closed 2022-01-19 by dorssel