dorssel/usbipd-win #200: Need help: BSOD when using wsl attach

loeffelpan, 2022-01-14:
Hey, I'm using _WSL Ubuntu 20.04 on Win10 21H2_ and would like to attach my ledger nano to sideload an app.
Unfortunately I got an BSOD with following stopcode when using the _wsl attach_-command.

PNP_DETECTED_FATAL_ERROR

Can you please help me figuring out whats wrong?

dorssel, 2022-01-14:
@loeffelpan 
Woah, BSOD ... that's not good! You're the first to report such a serious issue.

Did the bluescreen report the kernel module/driver that caused the crash? It could be the VBox drivers, but also the hub driver. What type of hub is your device connected to? Can you try using a different port/hub/computer? First we need to find out if it is machine, hub, hub-driver, or device related.

What version of `usbipd-win` are you using? 1.3.0? Could you try the latest master build? You can find it here: https://github.com/dorssel/usbipd-win/actions/runs/1695796183.

loeffelpan, 2022-01-14:
After some testing with different devices and different ports/hub I would guess that issue is hub- or hub-driver related. Any device with another hub number is working fine. Any device connected with bus id 4-x leads to BSOD. Other bus id (e. g. 5-x) seem to work.
I just attached the ledger with bus id 5-2 and just **connected** an device to bus id 4-2 and got an blue screen. Maybe this isn't really related to your project.

If you want, you can close this issue. But I'm also ready for further investigation if it would be helpful for you. Your descision.

dorssel, 2022-01-14:
@loeffelpan 
When `usbipd-win` 1.2.0 attaches a device, effectively an unplug/replug is simulated (USB port reset). That explains why you are experiencing this during `wsl attach`. This strategy has changed for 1.3.0 (which does a disable/enable cycle instead) and again in the current master (which does a pause/restart cycle instead). So, you may have better luck with one of those.

However, I think we can indeed conclude that the problem is with the hub; not something that `usbipd-win` can fix. So, I will close this ticket. But I am interested to find out what the root cause is anyway. Can you locate the root port and the hub(s) in the Window Device Manager? (best select: view->by connection). Are the hubs using the standard Microsoft drivers? What happens if you "Update driver" on the hub or the root port? Finally, it could be some third-party USB filter driver. What is the output of:

```
reg query "HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}" /v UpperFilters
```

closed 2022-01-14 by dorssel

loeffelpan, 2022-01-16:
> So, you may have better luck with one of those.
Unfortunately I got #152 with 1.3.0. Didn't try master branch. Maybe this would work.
 
> Can you locate the root port and the hub(s) in the Window Device Manager? (best select: view->by connection). Are the hubs using the standard Microsoft drivers? What happens if you "Update driver" on the hub or the root port?
I'm not sure if this is the right device. _Update driver_ says that driver is up to date and it is using standard drivers as you can see here.
[image]

> Finally, it could be some third-party USB filter driver. What is the output of:
> 
> ```
> reg query "HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}" /v UpperFilters
> ```
```
HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}
    UpperFilters    REG_MULTI_SZ    EUsbHubFilter
```

dorssel, 2022-01-16:
@loeffelpan 
There is your problem:

> UpperFilters    REG_MULTI_SZ    EUsbHubFilter

Seems to be part of "flexihub". Google "flexihub" + "BSOD". You may want to uninstall that.

loeffelpan, 2022-01-16:
Wow... thanks for that (lightning fast) reply and advice. Youre right. I've used Flexihub some time ago. I will uninstall it as I don't need it for now.

Thanks