dorssel/usbipd-win #205: Unable to add tested devices to the wiki

Josverl, 2022-01-17:
Hi Frans,

I tried a few devices and they appear to work well , so I tried to add them to the tested devices page,
but think that editing is not allowed.

I tested the below two devices 
```
| f055:9800 | MicroPython Pyboard Virtual Comm Port in FS Mode| 1.1 / 2.0| :heavy_check_mark: |                                      |
| 2e8a:0005 | Pimoroni pico lipo MicroPython Board in FS mode| 1.1 / 2.0| :heavy_check_mark: |                                      |
```

Also assuming you want the community to contribute, I think would be good to add a page with the verification steps one should follow before adding a device.
Below is what I tried, but im sure that the WSL usb part could be improved.

## Device Test Procedure 
Before testing new devices, configure your machine and verify your setup with at least one other known good device.
**From a Windows prompt**
- [ ] run `wsl --list --verbose`
  Verify that you are running as WSL version 2
```
  NAME      STATE           VERSION
* Ubuntu    Running         2
```
To test a device: 
- [ ]  Plug in the USB device 
**From a Windows elevated prompt** 
- [ ] `usblist`
  Verify that the device is visible
- [ ] `usblist wsl`
  Verify that the device is visible, and listed as **'Not attached'**
- [ ] `usbipd wsl attach --busid=<BUSID>`
  Verify that the device state is now **`Shared`**

**From a WSL prompt** (Ubuntu)
For USB serial ports
- [ ] `lsusb`
  Verify that the device is visible
- [ ] `ls /dev/tty*`
  Verify that the device is visible as `/tty/S<n>` or `/tty/ACM<n>`
- [ ] `screen /dev/ttyACM0 115200` ,
  Verify that you can connect to the 

For USB Composite Devices
- [] run `lsusb --tree`  
  Verify that all expected usb interfaces are shown, such as mass storage

dorssel, 2022-01-17:
@Josverl 
Awesome feedback, thanks!
The WSL instructions come from your employer, I believe.
@benmcmorran: maybe chime in on this?

A lot will be changing in version 2.0.0 that I am preparing. Could you also give PR #164 a go? The artifact is at https://github.com/dorssel/usbipd-win/actions/runs/1705631433. It has a lot better reporting if things aren't working immediately.

Voyytec, 2022-01-19:
On the subject of new device testing feedback. 

I was able to successfully use usbipd-win with:

`ID 12d1:140c Huawei Technologies Co., Ltd. E180v `

which is a Huawei GSM modem that I used as a dongle for my freepbx virtual machine hosted on a Windows 10 OS via Hyper-V. Both USB 2.0 and 3.0 ports work well – calls, sms and ussd messages go through both ways. I managed to correctly set it up on Ubuntu 18.04.6 LTS and asterisk with FreePBX recognising it flawlessly. 

The default FreePBX Sangoma7.8 distro (derivative of CentOS) produced an error after a usbip attach -r attempt (usbip list -r lists the device and the bus properly):

```
vhci_hcd vhci_hcd: dev 1 ep7in scatterlist error -104/-121
usb usb1-port: Cannot enable. Maybe the USB cable is bad? 
usb usb1-port: unable to enumerate USB device
```

I don’t need a fix as I’m ok with running freePBX on Ubuntu – this is just to let you know that Huawei 3G dongle works with Ubuntu vm via usbipd-win.

Great piece of software that saved me a lot of time!

dorssel, 2022-01-19:
@Josverl 
@Voyytec 
Thanks for reporting working devices.
I'll add them when I get around to it. I will leave this issue open for other to add to it.

Josverl, 2022-01-19:
To save you from copy /pasting and you could consider to open up the wiki , by Deselecting the Restriction on editing.
[image]
https://docs.github.com/en/communities/documenting-your-project-with-wikis/changing-access-permissions-for-wikis

benmcmorran, 2022-01-19:
@Josverl  I added the tested devices and changes from #209 to the wiki.

dorssel, 2022-01-19:
@benmcmorran 
I've invited Jos as a collaborator, so he can directly add his (very useful!) contributions to the wiki.

closed 2022-01-20 by Josverl