dorssel/usbipd-win #206: USB Webcam detected by WSL2 but still no /dev/video0

anmol1ratn, 2022-01-18:
I followed the instructions and the output of `lsusb` command:
`Bus 001 Device 004: ID 13d3:56eb IMC Networks USB2.0 HD UVC WebCam`

Means that WSL2 has detected the webcam.

But:
```
v4l2-ctl --list-devices
Cannot open device /dev/video0, exiting.
```
```
ls -ltrh /dev/video*
ls: cannot access '/dev/video*': No such file or directory
```
Can you please tell me how to get `/dev/video0` file? Am I missing something?

dorssel, 2022-01-18:
@anmol1ratn 
By default, the WSL kernel only has USB driver for (some) USB-to-serial devices. It does not have drivers for cameras. You need to build your own kernel, see https://github.com/dorssel/usbipd-win/wiki/WSL-support.

anmol1ratn, 2022-01-19:
Hi @dorssel 

Forgive my limited knowledge, but why can't I just install the webcam driver with some sudo apt install command?

dorssel, 2022-01-19:
@anmol1ratn 
That's a complex answer. The short summary: WSL2 uses a monolithic kernel (does not support modules), as WSL instances all share the same kernel (they are essentially dockers). So any driver you want to use must be compiled into that one kernel image.

Sorry, but this is out-of-scope for `usbipd-win`. WSL issues and questions can be asked here: https://github.com/microsoft/WSL

closed 2022-01-19 by anmol1ratn

biiigwang, 2022-05-20:
@anmol1ratn Hello.Have you solved the problem?

anmol1ratn, 2022-05-20:
No, I couldn't. Compiling a kernel was overkill for my small task.