dorssel/usbipd-win #207: An exception occurred while communicating with the client

yyl-20020115, 2022-01-19:
Hi,
I bind a USB keyboard on Win10 server side and connect with usbip attach -r ... -b ...
and I got the following error on server side,                                          

fail: UsbIpServer.ConnectedClient[3]
      An exception occurred while communicating with the client:
      System.ComponentModel.Win32Exception (2): CreateFile
         at UsbIpServer.DeviceFile..ctor(String fileName) in C:\Working\usbipd-win\UsbIpServer\DeviceFile.cs:line 31
         at UsbIpServer.VBoxUsbMon..ctor() in C:\Working\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 20
         at UsbIpServer.ConnectedClient.HandleRequestImportAsync(CancellationToken cancellationToken) in C:\Working\usbipd-win\UsbIpServer\ConnectedClient.cs:line 131

Can you please help me to identify the reason of the error?
P.S. both server and client have admin rights (UAC elevated).

Best,
Yilin

dorssel, 2022-01-19:
@yyl-20020115 
It looks like the VBoxUbMon driver is not running. You can check with:
```
sc query VBoxUsbMon
```
Did you reboot after installing `usbipd-win`? Sometimes you have to, even if Windows doesn't say so...

yyl-20020115, 2022-01-20:
@dorssel
Thanks for reply!

What should I expect from the sc query? I got nothing.

I put Win10 in Test Mode and imported the Oracle cert.

Rebooted and sc query still got nothing.

Any further suggestions?
BR,
Yilin

@yyl-20020115<https://github.com/yyl-20020115>
It looks like the VBoxUbMon driver is not running. You can check with:

sc query VBoxUsbMon

Did you reboot after installing usbipd-win? Sometimes you have to, even if Windows doesn't say so...

yyl-20020115, 2022-01-20:
***@***.***
This is screen shot of my registery settings.

@yyl-20020115<https://github.com/yyl-20020115>
It looks like the VBoxUbMon driver is not running. You can check with:

sc query VBoxUsbMon

Did you reboot after installing usbipd-win? Sometimes you have to, even if Windows doesn't say so...

dorssel, 2022-01-20:
@yyl-20020115 
Changing certificates or putting Windows in test mode is absolutely not required. Please don't.
`sc query VBoxUsbMon` should output *something*. At least an error. No admin rights required either. I get this:
```
C:\>sc query VBoxUsbMon

SERVICE_NAME: VBoxUsbMon
        TYPE               : 1  KERNEL_DRIVER
        STATE              : 4  RUNNING
                                (STOPPABLE, NOT_PAUSABLE, IGNORES_SHUTDOWN)
        WIN32_EXIT_CODE    : 0  (0x0)
        SERVICE_EXIT_CODE  : 0  (0x0)
        CHECKPOINT         : 0x0
        WAIT_HINT          : 0x0

C:\>
```

yyl-20020115, 2022-01-20:
Yes, I get the output in cmd. Was in powershell, thought they’are same.

@yyl-20020115<https://github.com/yyl-20020115>
Changing certificates or putting Windows in test mode is absolutely not required. Please don't.
sc query VBoxUsbMon should output something. At least an error. No admin rights required either. I get this:

C:\>sc query VBoxUsbMon

SERVICE_NAME: VBoxUsbMon

        TYPE               : 1  KERNEL_DRIVER

        STATE              : 4  RUNNING

                                (STOPPABLE, NOT_PAUSABLE, IGNORES_SHUTDOWN)

        WIN32_EXIT_CODE    : 0  (0x0)

        SERVICE_EXIT_CODE  : 0  (0x0)

        CHECKPOINT         : 0x0

        WAIT_HINT          : 0x0

C:\>

dorssel, 2022-01-20:
@yyl-20020115 
I guess you needed a reboot, which then started VBoxUsbMon properly. Can we close this?

yyl-20020115, 2022-01-23:
Yes please close this.

Thsnk you very much!

@yyl-20020115<https://github.com/yyl-20020115>
I guess you needed a reboot, which then started VBoxUsbMon properly. Can we close this?

yyl-20020115, 2022-01-23:
@***@***.***>
I test two machines.
192.168.1.108(desktop) and 192.168.1.104(macbookpro)

When 108 is client and 104 is server, it just works correctly.
However if 104 is client and 108 is server, it gets the error I posted before.

I uninstalled VMWare or any virtual machine from both PCs.
Any idea why?
Best Regards,
Yilin

@yyl-20020115<https://github.com/yyl-20020115>
I guess you needed a reboot, which then started VBoxUsbMon properly. Can we close this?

dorssel, 2022-01-23:
@yyl-20020115 
Have you updated to 2.0.0? It was release a few days ago and provides more detailed error information.

yyl-20020115, 2022-01-23:
OK, I’ll try 2.0 now

从 Windows 版邮件<https://go.microsoft.com/fwlink/?LinkId=550986>发送

yyl-20020115, 2022-01-24:
Hi,
I got this message
device version not supported: 4.0, expected 5.0

fail: UsbIpServer.ConnectedClient[3]
      An exception occurred while communicating with the client:
      System.NotSupportedException: device version not supported: 4.0, expected 5.0
         at UsbIpServer.VBoxUsbMon.ClaimDeviceOnce(ExportedDevice device) in C:\Working\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 77
         at UsbIpServer.VBoxUsbMon.ClaimDevice(ExportedDevice device) in C:\Working\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 121
         at UsbIpServer.ConnectedClient.HandleRequestImportAsync(CancellationToken cancellationToken) in C:\Working\usbipd-win\UsbIpServer\ConnectedClient.cs:line 149

where can I find 5.0?

the vboxusbmon I used is built from vbox source. Do you have another version of the vbox source?

BR,

Yilin

@yyl-20020115<https://github.com/yyl-20020115>
Have you updated to 2.0.0? It was release a few days ago and provides more detailed error information.

dorssel, 2022-01-24:
@yyl-20020115 
`usbipd-win` ships with version 5. This machine apparently has a (rather old!) version of VirtualBox installed. Either remove the old VirtualBox, or use `usbipd-win bind --force` to bypass VBoxUsbMon altogether.

dorssel, 2022-01-24:
@yyl-20020115 
I have added a sanity check to warn about outdated VBoxUsbMon in PR #229.

yyl-20020115, 2022-01-26:
Ok all done. My problem that I used old verion of vbox
thx

@yyl-20020115<https://github.com/yyl-20020115>
I have added a sanity check to warn about outdated VBoxUsbMon in PR #229<https://github.com/dorssel/usbipd-win/pull/229>.

dorssel, 2022-01-26:
@yyl-20020115 
Can we close this issue?

yyl-20020115, 2022-01-27:
OK,pls.

@yyl-20020115<https://github.com/yyl-20020115>
Can we close this issue?

closed 2022-01-27 by dorssel