dorssel/usbipd-win #218: This access control list is not in canonical form

bacongobbler, 2022-01-20:
After upgrading from 1.3.0 to 2.0.0, I'm starting to see this:

```
info: UsbIpServer.ConnectedClient[1]
      Client 172.27.228.221 claimed device at 6-3 (USB\VID_1050&PID_0407\7&2D859D3B&0&3).
info: UsbIpServer.ConnectedClient[2]
      Client 172.27.228.221 released device at 6-3 (USB\VID_1050&PID_0407\7&2D859D3B&0&3).
fail: UsbIpServer.ConnectedClient[3]
      An exception occurred while communicating with the client:
      System.InvalidOperationException: This access control list is not in canonical form and therefore cannot be modified.
         at System.Security.AccessControl.CommonAcl.ThrowIfNotCanonical()
         at System.Security.AccessControl.CommonAcl.AddQualifiedAce(SecurityIdentifier , AceQualifier , Int32 , AceFlags , ObjectAceFlags , Guid , Guid )
         at System.Security.AccessControl.DiscretionaryAcl.AddAccess(AccessControlType , SecurityIdentifier , Int32 , InheritanceFlags , PropagationFlags )
         at System.Security.AccessControl.CommonObjectSecurity.ModifyAccess(AccessControlModification , AccessRule , Boolean& )
         at System.Security.AccessControl.CommonObjectSecurity.AddAccessRule(AccessRule )
         at System.Security.AccessControl.RegistrySecurity.AddAccessRule(RegistryAccessRule )
         at UsbIpServer.RegistryUtils.SetDeviceAsAttached(Guid guid, BusId busId, IPAddress address, String stubInstanceId) in D:\a\usbipd-win\usbipd-win\UsbIpServer\RegistryUtils.cs:line 81
         at UsbIpServer.ConnectedClient.HandleRequestImportAsync(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConnectedClient.cs:line 212
         at UsbIpServer.ConnectedClient.HandleRequestImportAsync(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConnectedClient.cs:line 242
```

reverting to #221 resolved the issue.

dorssel, 2022-01-20:
@bacongobbler 
Thanks for the report!
What is the version and language of your OS?

bacongobbler, 2022-01-20:
[image]

dorssel, 2022-01-20:
@bacongobbler 
What is the output of the following command in PowerShell:
```
Get-Acl -Path HKLM:\SOFTWARE\usbipd-win\Devices | Format-List
```

It seems that this error is related to corrupt ACLs. You could also try the 'AccessChk' utility: https://docs.microsoft.com/en-us/sysinternals/downloads/accesschk. The command would be:

```
accesschk.exe -k HKLM:\SOFTWARE\usbipd-win\Devices
```

The reason you do not see this with 1.3.0 is that only since 2.0.0 the ACLs are set so that WSL attach/detach no longer requires admin rights. But the underlying problem may be a corrupt registry...

bacongobbler, 2022-01-21:
```
Get-Acl -Path HKLM:\SOFTWARE\usbipd-win\Devices | Format-List

Path   : Microsoft.PowerShell.Core\Registry::HKEY_LOCAL_MACHINE\SOFTWARE\usbipd-win\Devices
Owner  : NT AUTHORITY\SYSTEM
Group  : NT AUTHORITY\SYSTEM
Access : BUILTIN\Users Allow  ReadKey
         BUILTIN\Administrators Allow  FullControl
         NT AUTHORITY\SYSTEM Allow  FullControl
         CREATOR OWNER Allow  FullControl
         APPLICATION PACKAGE AUTHORITY\ALL APPLICATION PACKAGES Allow  ReadKey
         S-1-15-3-1024-1065365936-1281604716-3511738428-1654721687-432734479-3232135806-4053264122-3456934681 Allow
         ReadKey
Audit  :
Sddl   : O:SYG:SYD:AI(A;CIID;KR;;;BU)(A;CIID;KA;;;BA)(A;CIID;KA;;;SY)(A;CIIOID;KA;;;CO)(A;CIID;KR;;;AC)(A;CIID;KR;;;S-1
         -15-3-1024-1065365936-1281604716-3511738428-1654721687-432734479-3232135806-4053264122-3456934681)
```

Accesschk doesn't appear to show anything useful... Nor does accesschk64.

```
.\accesschk.exe -k HKLM:\SOFTWARE\usbipd-win\Devices

Accesschk v6.14 - Reports effective permissions for securable objects
Copyright ⌐ 2006-2021 Mark Russinovich
Sysinternals - www.sysinternals.com
```

dorssel, 2022-01-24:
@bacongobbler 
That all looks correct. Can you open regedit.exe to find the GUID of one of the devices? These are all unique, e.g. on my machine:

```
Get-Acl -Path "HKLM:\SOFTWARE\usbipd-win\Devices\{474a056d-d6db-4c55-8ccb-a2c79c019f84}" | Format-List
```

dorssel, 2022-01-27:
@bacongobbler 
On second look: are you saying `accesschk.exe` works on `usbipd-win`, but not on `usbipd-win\Devices` ? That may indicate that the Devices key actually has a corrupt ACL. What does regedit.exe say when you open the access rights property page on that node?

bacongobbler, 2022-01-28:
> Can you open regedit.exe to find the GUID of one of the devices?

[image]

> On second look: are you saying `accesschk.exe` works on `usbipd-win`, but not on `usbipd-win\Devices`?

Yes.

> What does regedit.exe say when you open the access rights property page on that node?

[image]

bacongobbler, 2022-01-28:
> It seems that this error is related to corrupt ACLs.

How would I have corrupted the ACLs? Sorry, not much of a Windows user here so this is all new to me.

dorssel, 2022-01-28:
@bacongobbler 
Thanks for that screenshot. What do you see when you press "Advanced" there?

Background:
- The register is essentially a file in your hard drive
- The error you get indicates something is wrong with the registry *now*
- "The Internet" (Google) tells me that accesschk.exe should report any such errors. In your case it doesn't  even work for the Device node, which is very suspicious
- You are the only one experiencing this problem (out of approx. 7000 users)

What could have caused this? I don't know... Maybe memory corruption when the registry was written to disk. Maybe file corruption on the disk itself. Maybe an unfortunate reboot at the wrong time (although the file system journal should cover that)...

On the other hand, the Device node is created by the `usbipd-win` installer. So it could always be a bug. That's what we're trying to find out... If you're not worried about the more severe causes, you could simply try to uninstall/reinstall `usbipd-win`, but that could also hide an underlying bigger problem.

bacongobbler, 2022-01-28:
> What do you see when you press "Advanced" there?

[image]

I managed to fix the issue by completely uninstalling and re-installing usbipd-win from scratch.

When I opened the ticket, I used the .msi "Upgrade" option... Perhaps that may have something to do with it?

In either case I found a workaround, and can happily attach/detach devices without any issue.

```
PS C:\Users\me> usbipd wsl list
BUSID  DEVICE                                                        STATE
1-3    USB Input Device                                              Not attached
2-1    USB 2.0 BILLBOARD                                             Not attached
2-3    Goodix fingerprint                                            Not attached
2-4    MediaTek Bluetooth Adapter                                    Not attached
2-6    Realtek USB GbE Family Controller                             Not attached
3-1    Logitech BRIO, USB Input Device                               Not attached
4-2    Scarlett Solo USB, Focusrite Control, USB Mass Storage De...  Not attached
4-3    Audioengine D1, USB Input Device                              Not attached
4-4    USB Input Device                                              Not attached
6-2    USB Input Device                                              Not attached
6-3    USB Input Device, Microsoft Usbccid Smartcard Reader (WUDF)   Attached - Ubuntu
```

I'm happy to keep attempting to reproduce/diagnose the issue, but feel free to close if you're happy with the outcome.

dorssel, 2022-01-28:
@bacongobbler 
Good to hear you have it working now.
And thanks for the info. I will try to reproduce the upgrade. What version were you coming from? 1.3.0 to 2.0.0?
If I cannot reproduce it, I think we'll leave it here and just call it a fluke...

(PS: There is nothing special about that registry entry. It is (or should be) the same as any other. Nothing has changed since 1.0.0 in how it is created or used. And your Advanced permissions look exactly like mine. I'm out of ideas, really.)

bacongobbler, 2022-01-28:
> What version were you coming from? 1.3.0 to 2.0.0?

For the most part, yeah. I was using the artifact build from https://github.com/dorssel/usbipd-win/pull/153 so I could attach my YubiKey to WSL2, then upgraded to 2.0.0 once it became available.

I really appreciate the assistance!

dorssel, 2022-01-29:
@bacongobbler 
Can't reproduce ... closing.

closed 2022-01-29 by dorssel