dorssel/usbipd-win #219: usbipd sometimes scrambles USB serial devices charater order

eteq, 2022-01-21:
While usbipd generally is working great, I've noticed some odd behavior when connecting to USB serial devices (in this case, some Circuitpython microcontroller boards, although I see the same behavior on multiple other kinds of USB serial devices when they are connected through WSL).  Specifically, some of the characters seem to arrive out-of-order after padding through usbipd.

To be more concrete, when I connect to a board running Circuitpython, the following is the expected behavior, which I can acheive by using PuTTY in windows (without usbipd involved):
```
>>> import arg
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'arg'
>>> import arg
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'arg'
>>>
```

Whereas when I connect this to WSL2 via usbipd in ubuntu 21.10 I get things like this:
```
>>> import arg
Traceback (most recent call last):
  File "<stdin>", line 1<module>
, in ImportError: no module named 'arg'
>>> import arg
Traceback (most recent call last):
", line <stdin>1, in   File "<module>
ImportError: no module named 'arg'
>>>
```

You can see there things like the `"` in the second import error are in the wrong place, whereas in the first one the `, in` has been moved to the wrong place.

It's rather frustratingly hard to reproduce consistently, but it seems to be most often worse when a lot of information is going to or from the serial console. I can boot this same computer and device into linux and do some tests there if it would be helpful (i.e., to see if it's `usbipd` in the kernel vs `usbipd-win` that's the issue here), but I'm not sure exactly what to test.

dorssel, 2022-01-21:
@eteq 
Thanks for this most useful report!
I think you are on a very longstanding bug that will solve many issues! This is the first time I see some clear indicators of a race condition that I wasn't aware of:

USBIP sends multiple URBs (USB request blocks) that need to be handled asynchronously. That means multiple simultaneous URBs can be outstanding, and can be completed out-of-order. And "out-of-order" is the key here. This report seems to indicate that the replies *for the same endpoint* must be sent back *in-order*. Currently, the replies are sent back asynchronously as well, even for the same endpoint. When Windows completes 2 URBs for the same endpoint (in-order, in quick succession), their completion handlers may be called out-of-order. Currently, that means that the two replies will be sent back over USBIP in the wrong order! It turns out that URBs must be handled asynchronously, and can be completed out-of-order too, but *for the same endpoint* their replies must be sent back in-order.

I will make a reply queue per endpoint, so that the replies for URBs for the same endpoint will be re-sequenced back in-order. Are you willing to test that once the PR is ready?

dorssel, 2022-01-23:
@eteq 
Could you please try PR #224, the installer is at https://github.com/dorssel/usbipd-win/actions/runs/1734739747.
This may very well solve the problem.

eteq, 2022-01-28:
@dorssel - tested just now, and #224 does indeed seem to fix this (at least with the first device I tested with - I'll test a few more in the next few days and report back if any continue to show the issue.  But I doubt it - this looks like it solved it!)

dorssel, 2022-01-28:
@eteq
Thanks for testing, and the review! Merging...

PR #224 (2022-01-28): Fix reply ordering per endpoint
Fixes #219
closed 2022-01-28 by dorssel