dorssel/usbipd-win #222: Setup issue : "Service 'USBIP Device Host' (usbipd) failed to start. Verify that you have sufficient privileges to start system services"

kevin-doolaeghe, 2022-01-21:
Hi!
I need help to install the program:

The setup stops when starting the service with error message:
"Service 'USBIP Device Host' (usbipd) failed to start. Verify that you have sufficient privileges to start system services"

I am running setup with admin privileges.
When I try to start the service manually, I get this error:
"error 1068 the dependency service or group failed to start"

Thanks in advance,
Kevin

dorssel, 2022-01-21:
@kevin-doolaeghe 
The service depends on the VBoxUsbMon driver, which also gets installed. Sometimes Windows fails to start a newly installed driver if there is a reboot pending due some earlier other install. Try rebooting and installing again.

kevin-doolaeghe, 2022-01-21:
Ok, thanks for the reply.

I tried to reboot and install several times
This has not solved the problem

I think that the VBoxUsbMon isn't properly installed.

How can I check if it is correctly installed ?
Where is it located in driver manager?

Thank you !
Kevin

dorssel, 2022-01-21:
Can you install 1.3.0. it does also install the same VBoxUsbMon driver. That install should succeed. Then execute as admin
```
sc query VBoxUsbMon
sc start VBoxUsbMon
```
This should give you a reason why it won't run.

kevin-doolaeghe, 2022-01-21:
Successfully installed v1.3.0

```
sc query VBoxUsbMon
sc start VBoxUsbMon
```
The result is:

> C:\windows\system32>sc query VBoxUsbMon
>
> SERVICE_NAME: VBoxUsbMon
>        TYPE               : 1  KERNEL_DRIVER
>        STATE              : 4  RUNNING
>                                (STOPPABLE, NOT_PAUSABLE, IGNORES_SHUTDOWN)
>        WIN32_EXIT_CODE    : 0  (0x0)
>        SERVICE_EXIT_CODE  : 0  (0x0)
>        CHECKPOINT         : 0x0
>        WAIT_HINT          : 0x0
>
>C:\windows\system32>sc start VBoxUsbMon
>[SC] StartService échec(s) 1056 :
>
>Une instance du service s’exécute déjà.

Hope this can help you.
It says that another instance is already running.

dorssel, 2022-01-21:
That is weird ... the installation of VBoxUsbMon hasn't changed between 1.3.0 and 2.0.0. For some reason it failed to start after 2.0.0, but it did start for 1.3.0.

The good news is that you probably will be able to update to 2.0.0 now. Since it will leave the installation of VBoxUsbMon intact. Can you give that a try?

kevin-doolaeghe, 2022-01-21:
Update to v2.0.0 done.
Everything seems to work well.
I got the same results for the previous commands:
```
sc query VBoxUsbMon
sc start VBoxUsbMon
```

dorssel, 2022-01-21:
That's great. Are you willing to test a little bit more, so we can find the root cause of this problem?

Can you:
- uninstall `usbipd-win`
- verify that VBoxUsbMon is also uninstalled (`sc query VBoxUsbMon` should give an error)
- install 2.0.0 from scratch again

Just to confirm that you get the same problem as before. If so, I want to try and fix it. What you have done now is just workaround; I really want to fix it...

kevin-doolaeghe, 2022-01-21:
No problem!

- I uninstalled `usbipd-win` from control panel
- I ran `sc query VBoxUsbMon` as admin

It gives the following error:

> [SC] EnumQueryServicesStatus:OpenService échec(s) 1060 :
>
>Le service spécifié n’existe pas en tant que service installé.

It says that the service does not exist since it is not installed.

- I launch v2.0.0 setup but it stops again when starting the services (same messages as before)

dorssel, 2022-01-21:
Thanks. Give me a few minutes to build a PR...

kevin-doolaeghe, 2022-01-21:
yeah

dorssel, 2022-01-22:
Done (always a little tougher than expected). The build artifact is at https://github.com/dorssel/usbipd-win/actions/runs/1731351516. Thank for testing!

kevin-doolaeghe, 2022-01-22:
Hi!
This build worked for me: setup completed successfully

kevin-doolaeghe, 2022-01-22:
Nice job
Thanks for help

kevin-doolaeghe, 2022-01-22:
I close the issue

kevin-doolaeghe, 2022-01-22:
I close the issue

PR #223 (2022-01-22): Make failed service start on install non-fatal
Fixes #222
closed 2022-01-22 by kevin-doolaeghe