dorssel/usbipd-win #225: 'Failed to attach device with BUSID' when mounting a device not compatible with Windows

hofleverancier, 2022-01-23:
Trying to get my old labelprinter (DYMO 320) working on WSL, so I can use it with glabel, I have been doing this with a linux client running on unraid and that worked.

I am able to to attach other devices.
 `uname -a
Linux DESKTOP-VFES7GI 5.10.60.1-microsoft-standard-WSL2 #1 SMP Wed Aug 25 23:20:18 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux`
`usbipd --version
2.0.0+168.Branch.master.Sha.1b6fb94d0eacc51999007b95b621d32c30f7d898`
Packages are up to date.

`usbipd wsl list
BUSID  DEVICE                                                        STATE
       Unknown USB Device (Invalid Configuration Descriptor)         Not attached
2-1    USB Input Device                                              Not attached
2-5    Generic Bluetooth Radio                                       Not attached
2-8    Unknown USB Device (Invalid Configuration Descriptor)         Not attached
2-9    USB Input Device                                              Not attached
2-10   USB Input Device                                              Not attached
2-20   USB Mass Storage Device                                       Attached - Ubuntu-20.04
`

`usbipd wsl attach --busid 2-8
An error occurred mounting one of your file systems. Please run 'dmesg' for more details.
usbip: error: Attach Request for 2-8 failed - Device in error state

usbipd: error: Failed to attach device with BUSID '2-8'.
`

I don't know if you consider this a bug or not, it seems to me that the state of the device shouldn't matter as once linux sees it, it should be able to work with it.

If logs are needed, please tell me where to find them.

dorssel, 2022-01-23:
@hofleverancier 
I see multiple things that are worth investigating:

1. The very first device listed (without BUSID!) shouldn't be listed at all.
   Can you find the device in Windows Device Manager and post what it says at "Location:" in the property page?
2. The description "Unknown USB Device (Invalid Configuration Descriptor)". What is the problem code for these devices in Windows Device Manager?

You're wrong about "the state of the device shouldn't matter". Windows still needs to send URBs to the endpoints of these devices. And in order to do that, the hub must know the endpoints. The endpoints are defined in the Configuration Descriptor. So, if the Configuration Descriptor is "invalid enough" and the endpoints cannot be determined, then this is not going to work.

Nevertheless, you can get more detailed attach errors by following https://github.com/dorssel/usbipd-win/wiki/Troubleshooting.

hofleverancier, 2022-01-23:
> 1. The very first device listed (without BUSID!) shouldn't be listed at all.

Yes that seems some sort of remnant, it first appeared when I changed the problematic device to a different port, it wasn't there prior. 

> 1. Can you find the device in Windows Device Manager and post what it says at "Location:" in the property page?
```Port_#0008.Hub_#0002``` This is the device listed at 2-8

I just pulled the device entirely, and now this:
```usbipd wsl list
BUSID  DEVICE                                                        STATE
       Unknown USB Device (Invalid Configuration Descriptor)         Not attached
       Unknown USB Device (Invalid Configuration Descriptor)         Not attached
2-1    USB Input Device                                              Not attached
2-5    Generic Bluetooth Radio                                       Not attached
2-9    USB Input Device                                              Not attached
2-10   USB Input Device                                              Not attached
2-20   USB Mass Storage Device                                       Not attached
```
I guess this might be a separate bug.
The device has predictably disappeared from Windows Device Manager.

> 2\. The description "Unknown USB Device (Invalid Configuration Descriptor)". What is the problem code for these devices in Windows Device Manager?
Code 43

hofleverancier, 2022-01-23:
Alright when running the server as described in your link, and trying to attach:

```fail: UsbIpServer.ConnectedClient[3]
      An exception occurred while communicating with the client:
      System.IO.FileNotFoundException: Unable to find the specified file.
         at UsbIpServer.ConfigurationManager.GetVBoxDevice(BusId busId) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConfigurationManager.cs:line 304
         at UsbIpServer.VBoxUsb.ClaimDeviceOnce(BusId busId) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsb.cs:line 66
         at UsbIpServer.VBoxUsb.ClaimDevice(BusId busId) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsb.cs:line 76
         at UsbIpServer.VBoxUsb.ClaimDevice(BusId busId) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsb.cs:line 82
         at UsbIpServer.ConnectedClient.HandleRequestImportAsync(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConnectedClient.cs:line 157```

But based on what you said it probably will never work. I suppose the reason support was dropped for this printer is that it's doing something illegal on the port that a modern Windows version won't accept.

dorssel, 2022-01-23:
@hofleverancier 
>>The very first device listed (without BUSID!) shouldn't be listed at all.

> Yes that seems some sort of remnant, it first appeared when I changed the problematic device to a different port, it wasn't there prior.

This turned out to be a simple glitch in `wsl list`. Those should have been filtered out from the list. Fixed in #226.

hofleverancier, 2022-01-23:
@dorssel like so?

```
C:\WINDOWS\system32>usbipd wsl list
BUSID  DEVICE                                                        STATE
2-1    USB Input Device                                              Not attached
2-5    Generic Bluetooth Radio                                       Not attached
2-7    Unknown USB Device (Invalid Configuration Descriptor)         Not attached
2-9    USB Input Device                                              Not attached
2-10   USB Input Device                                              Not attached
2-20   USB Mass Storage Device                                       Attached - Ubuntu

C:\WINDOWS\system32>usbipd wsl attach --busid 2-7
usbip: error: Attach Request for 2-7 failed - Device in error state

usbipd: error: Failed to attach device with BUSID '2-7'.

C:\WINDOWS\system32>usbipd bind --busid 2-7 --force

C:\WINDOWS\system32>usbipd wsl list
BUSID  DEVICE                                                        STATE
2-1    USB Input Device                                              Not attached
2-5    Generic Bluetooth Radio                                       Not attached
2-7    Unknown USB Device (Invalid Configuration Descriptor)         Not attached
2-9    USB Input Device                                              Not attached
2-10   USB Input Device                                              Not attached
2-20   USB Mass Storage Device                                       Attached - Ubuntu

C:\WINDOWS\system32>usbipd wsl attach --busid 2-7
usbip: error: Attach Request for 2-7 failed - Device in error state

usbipd: error: Failed to attach device with BUSID '2-7'.

```

dorssel, 2022-01-23:
@hofleverancier 
Yes, it is now (supposed to) use the stub driver all the time. What does device manager say this time? And what if you run the server with full logging?

hofleverancier, 2022-01-23:
@dorssel Device Manager now calls the device "USBIP Shared Device", but still with a code 43.

I ran the attach command again with debugging, here's what it spat out.

```
C:\WINDOWS\system32>sc stop usbipd

SERVICE_NAME: usbipd
        TYPE               : 10  WIN32_OWN_PROCESS
        STATE              : 3  STOP_PENDING
                                (STOPPABLE, NOT_PAUSABLE, ACCEPTS_SHUTDOWN)
        WIN32_EXIT_CODE    : 0  (0x0)
        SERVICE_EXIT_CODE  : 0  (0x0)
        CHECKPOINT         : 0x0
        WAIT_HINT          : 0x0

C:\WINDOWS\system32>usbipd server Logging:LogLevel:Default=Trace
dbug: Microsoft.Extensions.Hosting.Internal.Host[1]
      Hosting starting
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\WINDOWS\system32
dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started
dbug: UsbIpServer.Server[1000]
      new connection from 172.25.51.213
dbug: UsbIpServer.Server[1000]
      connection closed
dbug: UsbIpServer.Server[1000]
      new connection from 172.25.51.213
dbug: UsbIpServer.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
fail: UsbIpServer.ConnectedClient[3]
      An exception occurred while communicating with the client:
      System.IO.FileNotFoundException: Unable to find the specified file.
         at UsbIpServer.ConfigurationManager.GetVBoxDevice(BusId busId) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConfigurationManager.cs:line 304
         at UsbIpServer.VBoxUsb.ClaimDeviceOnce(BusId busId) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsb.cs:line 66
         at UsbIpServer.VBoxUsb.ClaimDevice(BusId busId) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsb.cs:line 76
         at UsbIpServer.VBoxUsb.ClaimDevice(BusId busId) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsb.cs:line 82
         at UsbIpServer.ConnectedClient.HandleRequestImportAsync(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConnectedClient.cs:line 157
dbug: UsbIpServer.Server[1000]
      connection closed

```

dorssel, 2022-01-23:
@hofleverancier 
> Device Manager now calls the device "USBIP Shared Device", but still with a code 43.

OK, that's definitive then. It cannot even load the stub driver for this device. The stub driver is the one that is responsible for proxying the URBs between the hub and `usbipd-win`. If that driver does not even load, there is no way `usbipd-win` can access it. Sorry.

hofleverancier, 2022-01-23:
@dorssel Can't say we didn't try, thanks again.

closed 2022-01-23 by hofleverancier