dorssel/usbipd-win #227: [2.0.0] Failed to attach device

tadam98, 2022-01-23:
I am using:
c:> winver Version 21H2 (OS Build 22000.466 
$ uname -a
Linux MICKEY-2080TI 5.10.60.1-microsoft-standard-WSL2 #1 SMP Wed Aug 25 23:20:18 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
And I am getting an error with the FTDI cable:

```
C:\WINDOWS\system32>usbipd wsl list
BUSID  DEVICE                                                        STATE
1-1    USB Input Device                                              Not attached
2-5    Intel(R) Wireless Bluetooth(R)                                Not attached
2-8    USB Mass Storage Device                                       Not attached
6-1    USB Input Device                                              Not attached
9-2    Movidius MyriadX                                              Not attached
9-4    USB Input Device                                              Not attached
11-2   USB Serial Converter                                          Not attached
13-3   APX                                                           Not attached

usbipd: warning: Unknown USB filter 'nxusbf' may be incompatible with this software; 'bind --force' may be required.

C:\WINDOWS\system32>usbipd wsl attach --busid 11-2
usbipd: warning: Unknown USB filter 'nxusbf' may be incompatible with this software; 'bind --force' may be required.
usbip: error: Attach Request for 11-2 failed - Device busy (exported)

usbipd: error: Failed to attach device with BUSID '11-2'.
```

dorssel, 2022-01-23:
@tadam98 
> Device busy (exported)

This means something is already using the device exclusively. Probably some driver/service/software on the host.

But also:

> usbipd: warning: Unknown USB filter 'nxusbf' may be incompatible with this software; 'bind --force' may be required.

First google hit: https://forums.nomachine.com/topic/continuous-usb-issues-windows-10-x64.

tadam98, 2022-01-23:
Interesting as I am using nomachine 7.7.4 for desktop of thw wsl2. And I was doing the test in command line before conneting (of course the nomachin server was running in the background).

dorssel, 2022-01-23:
@tadam98 
Not necessarily a problem. If you never intend to use 11-2 under Windows, you can use --force:

```
usbipd bind --busid 11-2 --force
```

If you really want to share (sometimes WSL, sometimes Windows), you must fix these problems first.

tadam98, 2022-01-23:
Will try. There is also a way to tell the server not to share usb devices. I will try it as well.
https://www.nomachine.com/pdf/Enabling_and_disabling_the_NoMachine_device_sharing_services_manually.pdf

closed 2022-02-11 by dorssel

luispinheiro, 2024-07-18:
Hello guys! 
My inglish little bit, kkk... 

I'm trying to configure ESP-IDF in WSL on Windows with Ubuntu, I managed a few steps, but there are still some details that aren't working. I installed the USBIPD software, I was able to share the USB. I even managed to do Build in the terminal, but Flash, neither in the terminal nor in VSCode, is working, I tried disabling the Firewall but the problem persists. I tried to load the ESP-IDF path in Ubuntu, but I still couldn't. I tried to follow the documentation in the link below from Espressif. Can someone help me?

https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/wsl.md

[image]

[image]

[image]

[image]

[image]

[image]

 
[image]