dorssel/usbipd-win #230: usbpid-win breaks all USB access from any previous VirtualBox install

brianddk, 2022-01-25:
After install usbpid-win, all USB access through Oracle VirtualBox seems broken.

I noticed the VBoxUSB drivers in the [Drivers Folder](https://github.com/dorssel/usbipd-win/tree/6410bcc/Drivers) included in the usbpid-win repo and assume this is the reason for the contention.

Is there a procedure that exists or could be added to the Wiki for toggling between the VBoxUSB (usbpid-win) drivers and the VBoxUSB (Oracle) drivers.  It doesn't have to be simple, anything with `pnputil` or `devcon` would be fine.  I think its acceptable to say that both filters cannot be loaded at the same time, but it would still be nice to toggle between them without a reboot.

dorssel, 2022-01-25:
@brianddk 
What version of VirtualBox do you have installed? I know that `usbipd-win` up to version 0.3.1 could co-exists with recent versions of VirtualBox. But I haven't checked since...

brianddk, 2022-01-26:
I originally had `usbipd-win` version 1.3.0 on when I noticed the issue against VirtualBox version 6.0.x.  In the debug I upgraded up to VirtualBox 6.1.30 then also tried 6.0.24.  In all cases, `usbipd-win` worked fine, but any VirtualBox access to USB, either virtualized or by-filter to physical devices failed.  All other VirtualBox IO was unaffected.

I can pursue the issue with Oracle in parallel, just thought I would try here first since I saw the VBoxUSB drivers in your bundle as well.

If you can think of any `devcon` or similar commands to shed light on the contention, I'll be happy to pass on the output.

Thx.

dorssel, 2022-01-26:
@brianddk 
`usbipd-win` uses the exact same VBoxUsbMon as VirtualBox itself, in exactly the same way: ask VBoxUsbMon to filter the device and then claim it. That should coexist fine with VirtualBox, I would say. And when `usbipd-win` is "idle" (as in: no client has a device attached), it should appear to VirtualBox that `usbipd-win` isn't even there. So using both non-simultaneously should definitely work; I even think using both simultaneously should work.

FYI: the drivers currently shipped with `usbipd-win` are from VirtualBox 6.1.28, although they haven't changed for more than a year, so I would say anything 6.1.x should be fine.

dorssel, 2022-04-21:
@brianddk 
PR #354 updates the drivers to the latest VirtualBox, 6.1.34. Can you try the installer at https://github.com/dorssel/usbipd-win/actions/runs/2204188147 in combination with VirtualBox 6.1.34?

dorssel, 2022-04-28:
@brianddk 
Now released in 2.3.0.

dorssel, 2022-05-13:
Closing due to inactivity.

closed 2022-05-13 by dorssel