dorssel/usbipd-win #232: connection to WSL for Android device unstable

epiciskandar, 2022-01-26:
Windows 11 with WSL2 ubuntu 20.04, and with latest wsl update.

Connecting device is an Android 11 phone with adb enabled, once the first time `usbipd wsl attach -b a-b` works fine, then after configuring udev things, unfortunatly adb cannot recongnize it.
so, I replugged phone, `lsusb` in WSL cannot find this device(as expected), went into Powershell and re-attaching it:

```
(elevated console)
usbipd wsl attach -b a-b (since busid normally don't change, so I did not list first)
(lsusb in WSL didn't show the device)

usbipd wsl list
BUSID  DEVICE                              STATE
...
a-b      GM1900, ADB Interface     Not Shared
```

Oops?

```
usbipd list
BUSID  DEVICE                              STATE
...
a-b      GM1900, ADB Interface     Shared
```
Where did it shared to?

Other things might related:
* sometimes `usbipd list` shows a device under "Persisted:", so I had to always `usbipd unbind -a`, but next time it will appear again
* occasionly `usbipd wsl attach` works again, but rarely.
* `wsl --shutdown` didn't help much
* simple USB devices works fine, such as keyboards, USB storage sticks
* change another Android phone don't help

what would lead to this situation? would be that adb server(the phone side) always reconnecting when plugged in?

dorssel, 2022-01-26:
@epiciskandar 
Your mixing terminology here. When you run `wsl attach` for the first time, it includes a `bind`, because you first have to share the device. When a device is shared, it is registered with `usbipd-win` and available for attaching. This only needs to be done once. There is no need to `unbind`, unless you never want to attach the device again.

So after your first `wsl attach`, the binding worked fine, and the device is now shared (available for clients). Then is tries the `attach` itself, which failed. I am sure some form of error was displayed: what did it say?

Again, you don't (ever) have to `unbind` or `bind` again, the device is already shared correctly.

The "Persisted" means the device is not currently connected, but `usbipd-win` remembers it and the device is again Shared as soon as you connect it, so you can again attach it.

What version of `usbipd-win` are you using? Version 2.0.0 provides a lot more feedback for errors.

epiciskandar, 2022-01-27:
Thanks, you give lots information for me! Now I can understand what `bind` and `persist` for.

Yes I am using latest 2.0.0 version, and it really did not complaining anything, commands just done silently succeed, but actually not. I have checked Application logs in Windows Event Viewer, still nothing weird.

should I look into WSL's system logging? but I don't know what shall I notice to.

I've made a screenshot here:
[image]

dorssel, 2022-01-27:
@epiciskandar 
I suspect that the device gets attached alright, but then immediately detached again. What does /var/log/syslog say? And you mentioned event log, does it report any attach/detach?

epiciskandar, 2022-01-27:
Yes, from syslog, it just disconnect immediately:
```
 kernel: [94342.535170] vhci_hcd vhci_hcd.0: pdev(0) rhport(0) sockfd(3)
 kernel: [94342.535172] vhci_hcd vhci_hcd.0: devid(196610) speed(3) speed_str(high-speed)
 kernel: [94342.535197] vhci_hcd vhci_hcd.0: Device attached
 kernel: [94342.602044] vhci_hcd: connection closed
 kernel: [94342.602123] vhci_hcd: stop threads
 kernel: [94342.602125] vhci_hcd: release socket
 kernel: [94342.602128] vhci_hcd: disconnect device
```
the Windows side did not have any event log found, did the logs saved to other places?

dorssel, 2022-01-27:
@epiciskandar 
The service logs to the Application log in Windows Event Viewer. For example:

```
Category: UsbIpServer.ConnectedClient
EventId: 1

Client 127.0.0.1 claimed device at 1-6 (USB\VID_8087&PID_0A2B\5&18054507&0&6).
```

It also logs client communication errors (if any). Could it be simply that your kernel does not have a driver for this device? So it attaches, realizes the device is not supported, and then immediately detaches again? See https://github.com/dorssel/usbipd-win/wiki/WSL-support#building-your-own-usbip-enabled-wsl-2-kernel.

epiciskandar, 2022-01-27:
> Could it be simply that your kernel does not have a driver for this device?
Unlikely, sometimes it works, and previously on Windows 10(which before Windows Update upgrading this PC), this device also works with this WSL instance(though still not stable, but occasionly works after retry).
>  #building-your-own-kernel
But I think this is worth trying, might have more clue could be found.

And at last, no, no Application logs like that example, neither communication errors.
anyway, I will continue digging into this problem and try more scenarios or actions.

dorssel, 2022-01-27:
@epiciskandar 
If it has worked before, then the kernel is fine. So no need to build your own (save a lot of time).

Not getting any Application logs is an issue. Until that gets fixed (and first I have to find the root cause...), you can get debug logging following: https://github.com/dorssel/usbipd-win/wiki/Troubleshooting

dorssel, 2022-01-27:
@epiciskandar 
Missing log has been fixed in master.

dorssel, 2022-01-29:
@epiciskandar 
Could you try the latest master build? It is at https://github.com/dorssel/usbipd-win/actions/runs/1763859846.

epiciskandar, 2022-02-02:
Yep, but sorry I am in vacation(Chinese New Year) so I can try that only after 8th Feb.

dorssel, 2022-02-02:
@epiciskandar 
No problem. 
新年快乐！

epiciskandar, 2022-02-05:
There's good news and bad news.

Good news: on another Windows 10 PC, with two different Android phones, attaching action and connections works fine!

Bad news: I found a situation could generate BSD with usbipd, but actually the fault is in VBoxUSB.sys, I would make another issue for this problem.

for abouting connection problems related to this issue ,still need to wait until my vacation finished. sorry for this again.

epiciskandar, 2022-02-06:
I've reproduced this problem now, with `usbipd server` and verbose logging:

`usbipd wsl attach -b x-y`

```
dbug: UsbIpServer.Server[1000]
      new connection from 172.23.237.65
dbug: UsbIpServer.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
fail: UsbIpServer.ConnectedClient[3]
      An exception occurred while communicating with the client:
      System.ComponentModel.Win32Exception (433): DeviceIoControl returned error ERROR_NO_SUCH_DEVICE
         at UsbIpServer.VBoxUsb.ClaimDeviceOnce(BusId busId) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsb.cs:line 38
         at UsbIpServer.VBoxUsb.ClaimDevice(BusId busId) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsb.cs:line 76
         at UsbIpServer.ConnectedClient.HandleRequestImportAsync(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConnectedClient.cs:line 161
dbug: UsbIpServer.Server[1000]
      connection closed
```

looks like the device immediately disconnected.

dorssel, 2022-02-06:
@epiciskandar 

That is without `--force` I guess? Could you try PR #247 *with* `usbipd bind --force`? Artifact is at https://github.com/dorssel/usbipd-win/actions/runs/1798620074. Maybe that allows attaching this device...

qvoid, 2022-02-21:
Meet the same issue. Adb device is attached and detached immediately.
WSL2 kernel version is :
```
Linux LAPTOP-Q8BEABA5 5.10.60.1-microsoft-standard-WSL2 #1 SMP Wed Aug 25 23:20:18 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
```
And udevrules is configured following [instruction](https://github.com/M0Rf30/android-udev-rules)

usbipd server log is below:
```
info: UsbIpServer.ConnectedClient[1]
      Client 172.23.106.84 claimed device at 5-4 (USB\VID_2207&PID_0006\E485C444A838A059).
info: UsbIpServer.ConnectedClient[2]
      Client 172.23.106.84 released device at 5-4 (USB\VID_2207&PID_0006\E485C444A838A059).
```

dmesg from WSL2 is :
```
[ 4560.701266] vhci_hcd vhci_hcd.0: pdev(0) rhport(0) sockfd(3)
[ 4560.701268] vhci_hcd vhci_hcd.0: devid(327684) speed(3) speed_str(high-speed)
[ 4560.701289] vhci_hcd vhci_hcd.0: Device attached
[ 4560.730720] vhci_hcd: connection closed
[ 4560.730815] vhci_hcd: stop threads
[ 4560.730817] vhci_hcd: release socket
[ 4560.730820] vhci_hcd: disconnect device
``
But the other Android board with the same USB idVendor didn't have this problem:

```
[ 5189.268176] vhci_hcd vhci_hcd.0: pdev(0) rhport(0) sockfd(3)
[ 5189.268178] vhci_hcd vhci_hcd.0: devid(327684) speed(3) speed_str(high-speed)
[ 5189.268196] vhci_hcd vhci_hcd.0: Device attached
[ 5189.611888] usb 1-1: new high-speed USB device number 6 using vhci_hcd
[ 5189.761977] usb 1-1: SetAddress Request (6) to port 0
[ 5189.802456] usb 1-1: New USB device found, idVendor=2207, idProduct=0006, bcdDevice= 3.10
[ 5189.802458] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 5189.802459] usb 1-1: Product: Avaya Vantage
[ 5189.802460] usb 1-1: Manufacturer: Avaya
```
But this board can't not get ADB works, run `adb devices` many times and get different result : 
```
➜  ~ adb devices
* daemon not running; starting now at tcp:5037
* daemon started successfully
List of devices attached
21WZ502003CK    offline

➜  ~ adb shell
adb: no devices/emulators found
➜  ~ adb devices
List of devices attached
21WZ502003CK    authorizing

➜  ~ adb devices
List of devices attached
21WZ502003CK    offline

➜  ~ adb devices
List of devices attached
21WZ502003CK    authorizing

➜  ~ adb devices
List of devices attached
21WZ502003CK    authorizing

➜  ~ adb devices
List of devices attached

➜  ~ adb devices
List of devices attached
21WZ502003CK    offline
```

qvoid, 2022-02-21:
In my case, looks like the connection lost immediately after connected has something to do with Android Sleep setting and  device firmware bug, If I connect the Android board with host ADB, it will automatically disconnect after about 1 minute. But If I turn screen always on, the connection is stable. And usbipd works well when attach it to WSL2. But still ADB inside WSL2 can't connect to the Android board. Maybe it is the udev configuration problem.

qvoid, 2022-02-21:
Trying to access to Android board connected to Windows shared by usbipd on a Debian machine (not a virtual machine) via network also failed.  Device can be attached and can be found with `lsusb`. But disconnected immediately. Here is the dmesg log:
```
[ 1330.864306] vhci_hcd vhci_hcd.0: pdev(0) rhport(0) sockfd(3)
[ 1330.864316] vhci_hcd vhci_hcd.0: devid(393220) speed(3) speed_str(high-speed)
[ 1330.864342] vhci_hcd vhci_hcd.0: Device attached
[ 1331.098486] usb 3-1: new high-speed USB device number 3 using vhci_hcd
[ 1331.226501] usb 3-1: SetAddress Request (3) to port 0
[ 1331.259376] usb 3-1: New USB device found, idVendor=2207, idProduct=0006, bcdDevice= 3.10
[ 1331.259385] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1331.259390] usb 3-1: Product: rk3588_s
[ 1331.259394] usb 3-1: Manufacturer: rockchip
[ 1331.259398] usb 3-1: SerialNumber: fec195c0e78a7264
[ 1337.276701] vhci_hcd: unlink->seqnum 35
[ 1337.276732] vhci_hcd: the urb (seqnum 35) was already given back
[ 1342.308294] vhci_hcd: unlink->seqnum 38
[ 1342.308302] vhci_hcd: the urb (seqnum 38) was already given back
[ 1342.490946] vhci_hcd: connection closed
[ 1342.491415] vhci_hcd: stop threads
[ 1342.491463] vhci_hcd: release socket
[ 1342.491481] vhci_hcd: disconnect device
[ 1342.491595] usb 3-1: USB disconnect, device number 3
```
The kernel and Debian version info is :
```
# lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 10 (buster)
Release:        10
Codename:       buster
# uname -a
Linux 4.19.0-18-amd64 #1 SMP Debian 4.19.208-1 (2021-09-29) x86_64 GNU/Linux
```
The windows side log :
```
info: UsbIpServer.ConnectedClient[1]
      Client 192.168.180.10 claimed device at 6-4 (USB\VID_2207&PID_0006\FEC195C0E78A7264).
info: UsbIpServer.ConnectedClient[2]
      Client 192.168.180.10 released device at 6-4 (USB\VID_2207&PID_0006\FEC195C0E78A7264).
```

dorssel, 2022-03-16:
@epiciskandar 
@qvoid 
An issue was fixed in the current master build that may (no guarantees!) resolve these problems. You can find the installer here: https://github.com/dorssel/usbipd-win/actions/runs/1991685584. Could you give that a try?

epiciskandar, 2022-03-16:
That's good news! I will try this new version tomorrow ASAP

epiciskandar, 2022-03-17:
> @epiciskandar @qvoid An issue was fixed in the current master build that may (no guarantees!) resolve these problems. You can find the installer here: https://github.com/dorssel/usbipd-win/actions/runs/1991685584. Could you give that a try?

Way much better! 90% attaching actions succeeded immediately, barely needs a retry. Good job!
Is this also resolving #248?

epiciskandar, 2022-03-17:
> barely needs a retry
When this situation occurs, `usbipd` got this error message:

```
fail: UsbIpServer.ConnectedClient[3]
      An exception occurred while communicating with the client:
      UsbIpServer.ConfigurationManagerException (1003): CM_Register_Notification returned CR_INVALID_DATA
         at UsbIpServer.ConfigurationManager.ThrowOnError(CONFIGRET configRet, String function) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConfigurationManager.cs:line 42
         at UsbIpServer.ConnectedClient.HandleRequestImportAsync(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConnectedClient.cs:line 201
         at UsbIpServer.ConnectedClient.HandleRequestImportAsync(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConnectedClient.cs:line 246
```

qvoid, 2022-03-17:
Thank you.
The connection to WSL looks like more stable. Although devices already turn out at `lsusb` and `dmesg`,  I can't find devices by ADB. Could be my udev configuration problem.

epiciskandar, 2022-03-21:
New situation here, @dorssel 

After using the computer for days, the device can't being connected again, even after some recover attempts, like reconnecting, restarting Android device, restarting `usbipd server`, and restarting `wsl` with `wsl --shutdown`.

I'm gonna restarting computer, so, I leave the error message here:

```
<executing usbipd attach -b 1-10>

dbug: UsbIpServer.Server[1000]
      new connection from 172.22.54.128 <WSL instance IP address>
dbug: UsbIpServer.Server[1000]
      connection closed
dbug: UsbIpServer.Server[1000]
      new connection from 172.22.54.128
dbug: UsbIpServer.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT

<a noticeable pause here>

fail: UsbIpServer.ConnectedClient[3]
      An exception occurred while communicating with the client:
      UsbIpServer.ConfigurationManagerException (1003): CM_Query_And_Remove_SubTree returned CR_REMOVE_VETOED: PNP_VetoOutstandingOpen, USB\VID_2717&PID_FF48&MI_01\6&30bcec4f&0&0001
         at UsbIpServer.ConfigurationManager.RestartingDevice..ctor(UInt32 deviceNode) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConfigurationManager.cs:line 395
         at UsbIpServer.ConfigurationManager.RestartingDevice..ctor(String instanceId) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConfigurationManager.cs:line 378
         at UsbIpServer.ConnectedClient.HandleRequestImportAsync(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConnectedClient.cs:line 148
dbug: UsbIpServer.Server[1000]
      connection closed
```

qvoid, 2022-03-21:
> @qvoid Your earlier report was that devices would disconnect quickly after attaching. Do I understand correctly that the devices no longer auto-disconnect, and that the attachment itself is stable now?

Sorry, almost forgot what I reported.
Yes, the devices no longer auto-disconnect after I change sleep options at Android board.
But still can't find any devices after attach Android board to WSL2 through usbipd-win. This can be the UDEV rules configuration problem. 
Thank you for your great work!

epiciskandar, 2022-03-21:
Hi @qvoid, I'm not sure what is your environment looks like, but I'm willing to share my workarounds under Ubuntu 20.04, after executing `usbipd wsl attach`, I have aliased commands listed below. hopes this would give you some help.

```
❯ alias setup_usb_device
setup_usb_device='sudo service udev restart && sudo udevadm control --reload-rules && sudo udevadm trigger && adb kill-server && adb devices'
```

dorssel, 2022-03-21:
@epiciskandar 

>  UsbIpServer.ConfigurationManagerException (1003): CM_Query_And_Remove_SubTree returned CR_REMOVE_VETOED: PNP_VetoOutstandingOpen, USB\VID_2717&PID_FF48&MI_01\6&30bcec4f&0&0001

```
PNP_VetoOutstandingOpen == The requested operation was rejected because of outstanding open handles.
```

Something is holding a lock to the device. Not usbipd-win, since you restarted that. Probably some other service/software. If you are not using the device under Windows at all, I recommend `usbipd bind --force`. That should solve it.

qvoid, 2022-03-22:
@epiciskandar Thank you. I will give it a try. But not so soon. I already switched my working environment to Ubuntu running directory at host, no WSL2.

epiciskandar, 2022-03-22:
> Something is holding a lock to the device. 

Yep, I noticed that adb server is holding the device. after killing that daemon process, binding works again.

Since the connection problem looks being resolved, we should close this issue now.

closed 2022-03-22 by dorssel