dorssel/usbipd-win #233: feature request: add support for windows on arm

elpatax, 2022-01-26:
I'm using a Surface Pro X (SQ2) running Windows 11 Home Insider Preview; Version: Dev; OS build: 22538.1010.
The installer (usbipd-win 2.0.0) fails with exit code: 1603 and "rolls back actions". 
I suppose only x86_x64 is supported for now. I haven't found any other issues for targeting arm64 nor information about supported platforms.
Is it possible to also make a release for windows on arm devices?

Thank you.

dorssel, 2022-01-27:
@elpatax 
Sorry, not going to happen. `usbipd-win` requires (signed) USB drivers to "stub" the devices. The only open source drivers I could find that do the job are those from VirtualBox. See https://forums.virtualbox.org/viewtopic.php?t=92857.

closed 2022-01-27 by dorssel

elpatax, 2022-01-27:
I understand. Thank you for explaining why it's not possible for now.

sidd-kishan, 2022-07-06:
Windows on Arm64 support was added in the cezanne/usbip-win [Add support for ARM64 build](https://github.com/cezanne/usbip-win/commit/bf2d056e4b72946598213d8a88b66c2e810c82f9) Please try to implement the same.

theironrobin, 2022-07-11:
what is the difference between cezanne's repo and this one?

sidd-kishan, 2022-07-15:
The Cezanne's repo also has source for his driver required to stub the usb devices which drivers are not present.

theironrobin, 2022-07-15:
I would also love if we found a way to use usb devices on windows on arm WSL2

xmow49, 2024-09-10:
> I would also love if we found a way to use usb devices on windows on arm WSL2

Have you found a new way to use USB in WSL on arm ?

Molier, 2024-11-18:
Also interested in the update on this issue. Running WOA on a copilot plus pc and would want to integrate usb devices in the existing excellent wsl service.

dorssel, 2025-03-24:
ARM64 support is part of version 5.0.0, which has just been released.