dorssel/usbipd-win #236: `usbipd wsl list` fails with Unhandled exception

helzerr, 2022-01-27:
`C:\WINDOWS\system32>usbipd wsl list
BUSID  DEVICE                                                        STATE
Unhandled exception. System.AggregateException: One or more errors occurred. (DEVPKEY_Device_LocationInfo returned 'Port_#000A.Hub_#0001', expected form 'Port_#0123.Hub_#4567')
 ---> System.NotSupportedException: DEVPKEY_Device_LocationInfo returned 'Port_#000A.Hub_#0001', expected form 'Port_#0123.Hub_#4567'
   at UsbIpServer.ConfigurationManager.GetBusId(UInt32 deviceNode) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConfigurationManager.cs:line 174
   at UsbIpServer.UsbDevice.GetAll() in D:\a\usbipd-win\usbipd-win\UsbIpServer\UsbDevice.cs:line 30
   at UsbIpServer.CommandHandlers.UsbIpServer.ICommandHandlers.WslList(IConsole console, CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\CommandHandlers.cs:line 526
   at UsbIpServer.Program.<>c__DisplayClass10_0.<<Run>b__35>d.MoveNext() in D:\a\usbipd-win\usbipd-win\UsbIpServer\Program.cs:line 446
--- End of stack trace from previous location ---
   at System.CommandLine.Invocation.AnonymousCommandHandler.InvokeAsync(InvocationContext )
   at System.CommandLine.Invocation.InvocationPipeline.<>c__DisplayClass4_0.<<BuildInvocationChain>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass22_0.<<UseParseErrorReporting>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass15_0.<<UseHelp>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass26_0.<<UseVersionOption>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass24_0.<<UseTypoCorrections>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c.<<UseSuggestDirective>b__23_0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass21_0.<<UseParseDirective>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c.<<UseDebugDirective>b__8_0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c.<<RegisterWithDotnetSuggest>b__7_0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c.<<CancelOnProcessTermination>b__1_0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Invocation.InvocationPipeline.InvokeAsync(IConsole )
   at System.CommandLine.Parsing.ParseResultExtensions.InvokeAsync(ParseResult , IConsole )
   at System.CommandLine.Parsing.ParserExtensions.InvokeAsync(Parser , String[] , IConsole )
   --- End of inner exception stack trace ---
   at System.Threading.Tasks.Task.ThrowIfExceptional(Boolean )
   at System.Threading.Tasks.Task``1.GetResultCore(Boolean )
   at System.Threading.Tasks.Task``1.get_Result()
   at UsbIpServer.Program.Run(IConsole optionalTestConsole, ICommandHandlers commandHandlers, String[] args) in D:\a\usbipd-win\usbipd-win\UsbIpServer\Program.cs:line 504
   at UsbIpServer.Program.Main(String[] args) in D:\a\usbipd-win\usbipd-win\UsbIpServer\Program.cs:line 117`

dorssel, 2022-01-27:
@helzerr 
Looks like you have got an incompatible hub:

> Port_#000A.Hub_#0001', expected form 'Port_#0123.Hub_#4567

The port and bus numbers reported by hubs are supposed to be decimal; yours reports hexadecimal apparently.
Can you locate the hub in device manager? What drivers are installed for it (looks like not the default Microsoft drivers...).

It could also be an incompatible filter driver. What does the following command report:

```
reg query "HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}" /v UpperFilters
```

helzerr, 2022-01-28:
My system has a few USB hubs, most using the Microsoft driver, with a
couple using AMD and NVIDIA drivers (motherboard is AMD, video card has USB
ports as well)

With "-v UpperFilters", no results - I've included the output without "-v
UpperFilters" for comparison:

C:\WINDOWS\system32>reg query
"HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}"
/v UpperFilters

ERROR: The system was unable to find the specified registry key or value.

C:\WINDOWS\system32>reg query
"HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}"

HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}
    Class    REG_SZ    USB
    ClassDesc    REG_SZ    @%SystemRoot%\System32\SysClass.Dll,-3025
    IconPath    REG_MULTI_SZ    %SystemRoot%\system32\setupapi.dll,-20
    NoInstallClass    REG_SZ    1
    LowerLogoVersion    REG_SZ    5.2
    LastDeleteDate    REG_BINARY    2EF6491A0C93D701

HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}\0000
HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}\0001
HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}\0002
HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}\0003
HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}\0004
HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}\0005
HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}\0006
HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}\0007
HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}\0008
HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}\0009
HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}\0010
HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}\0011
HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}\0012
HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}\0013
HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}\0014
HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}\Properties

dorssel, 2022-01-28:
@helzerr 
OK, so it's not a filter. Then it must be one of those hubs. Unfold each hub in the device manager, and check the property page of the USB devices directly under them (no need to go deeper than 1 level under the hub). The property page should give a "Location:". One of those devices has 'Port_#000A.Hub_#0001' as location. The device is OK, but the hub should not have assigned that location. What hub is that? What drivers is it using? Can you update the drivers (of the hub, that is)? Is the device removable, or embedded in your machine?

helzerr, 2022-01-28:
Device description: USB Composite Device
Location information: Port_#000A.Hub_#0001
Driver node strong name:
usb.inf:392c3d53461eda3a:Composite.Dev.NT:10.0.19041.488:USB\COMPOSITE

It appears to be using the latest Microsoft Usbccgp.sys driver.

Disabling the device did not help.

Uninstalling the device, moved the problem to a different device...

Unhandled exception. System.AggregateException: One or more errors
occurred. (DEVPKEY_Device_LocationInfo returned 'Port_#000C.Hub_#0001',
expected form 'Port_#0123.Hub_#4567')

I'm unable to find any device referencing Port_#000C, but I did find
another related to Port_#000D which I also uninstalled; turns out, that was
my mouse...

helzerr, 2022-01-28:
The devices with seemingly hex port numbers, appear to be under the AMD
USB3.1 Root Hub, using AMD driver 1.0.5.3 - it's built into my motherboard.
Attempting to update the driver, "The best drivers for your device are
already installed"

helzerr, 2022-01-28:
I uninstalled that USB host controller device and its associated driver, upon reboot Windows installed driver version 10.0.19041.1320; I'm pleased to report `usbipd wsl list` works perfectly now!

dorssel, 2022-01-28:
@helzerr 
Good to hear that we found the root problem (AMD, non-standard location assignment). Do I understand that the hub works correctly now that the default MS drivers are installed for it? Are all connected devices working properly?

If so: congratz! Please close the issue.
Hope you can now use the device you intended in WSL. If not, feel free to open a new issue for that.

helzerr, 2022-01-28:
Yes, the hub and all devices are fully functional with the MS driver, and I'm able to attach devices to WSL distributions.

Thank you for your assistance troubleshooting this matter! Until I tried to list those USB devices, I had no idea I was running such a wonky USB driver!

closed 2022-01-28 by helzerr