dorssel/usbipd-win #238: CR_NOT_DISABLED

yyl-20020115, 2022-01-28:
Hi dorssel,

I got this error when I try to use usbipd-win with an USB-Keyboard,
fail: UsbIpServer.ConnectedClient[3]
      An exception occurred while communicating with the client:
      UsbIpServer.ConfigurationManagerException (1003): CM_Disable_DevNode returned CR_NOT_DISABLEABLE
         at UsbIpServer.ConfigurationManager.ThrowOnError(CONFIGRET configRet, String function) in C:\Working\usbipd-win-org\UsbIpServer\ConfigurationManager.cs:line 42
         at UsbIpServer.ConfigurationManager.TemporarilyDisabledDevice..ctor(UInt32 deviceNode) in C:\Working\usbipd-win-org\UsbIpServer\ConfigurationManager.cs:line 327
         at UsbIpServer.ConfigurationManager.TemporarilyDisabledDevice..ctor(String instanceId) in C:\Working\usbipd-win-org\UsbIpServer\ConfigurationManager.cs:line 320
         at UsbIpServer.ConnectedClient.HandleRequestImportAsync(CancellationToken cancellationToken) in C:\Working\usbipd-win-org\UsbIpServer\ConnectedClient.cs:line 145

Seems it is not allowed to disable keyboard... is it?
Anything wrong I've done? or any suggestion?

Best Regards,
Yilin

dorssel, 2022-01-28:
@yyl-20020115 
See https://github.com/dorssel/usbipd-win/issues/187.
Are you using 1.3.0? If so, then please upgrade to 2.0.0.

yyl-20020115, 2022-01-28:
C:\Working\usbipd-win\UsbIpServer\bin\x64\Debug\net6.0-windows8.0\win-x64>usbipd -v
usbipd-win
2.0.1+17.Branch.master.Sha.54157ab141eaae4d224f83130230b2b6b24f234f

C:\Working\usbipd-win\UsbIpServer\bin\x64\Debug\net6.0-windows8.0\win-x64>

@yyl-20020115<https://github.com/yyl-20020115>
See #187<https://github.com/dorssel/usbipd-win/issues/187>.
Are you using 1.3.0? If so, then please upgrade to 2.0.0.

dorssel, 2022-01-28:
@yyl-20020115 
> UsbIpServer.ConfigurationManagerException (1003): CM_Disable_DevNode returned CR_NOT_DISABLEABLE

This code (and any call to CM_Disable_DevNode) was removed on December 8, long before the release of 2.0.0.

> 2.0.1+17.Branch.master.Sha.54157ab141eaae4d224f83130230b2b6b24f234f

This version does definitely not contain any references to CM_Disable_DevNode.

I suggest you completely uninstall `usbipd-win` and perform a clean install of either 2.0.0, or of the latest master build at https://github.com/dorssel/usbipd-win/actions/runs/1763859846

closed 2022-03-12 by dorssel