dorssel/usbipd-win #243: Alternative to `usbipd wsl` subcommand

tommasomazzoni, 2022-02-01:
Microsoft Windows [version 10.0.22000.438]
Ubuntu 20.04.3 LTS (GNU/Linux 5.10.16.3-microsoft-standard-WSL2 x86_64)

My list of USB devices:
```
C:\windows\system32>usbipd wsl list
BUSID  DEVICE                                                        STATE
2-3    Périphérique série USB (COM4)                                 Not attached
2-6    Integrated Webcam                                             Not attached
2-8    Dell ControlVault w/ Fingerprint Touch Sensor, Lecteur de...  Not attached
2-10   Intel(R) Wireless Bluetooth(R)                                Not attached
3-3    Realtek USB GbE Family Controller                             Not attached
4-1    Périphérique d’entrée USB                                     Not attached
4-2    Périphérique d’entrée USB                                     Not attached
4-5    Périphérique d’entrée USB                                     Not attached
5-1    Périphérique d’entrée USB, Billboard Interface                Not attached
```

Apparently my wls2 distribution can not be run on elevated prompt for some reasons, so uspid attach doesn't work either:
```
C:\windows\system32>usbipd wsl attach --busid=2-3
usbipd: error: No default WSL distribution exists.
```

In another issue I found an alternative solution:
> do not use the usbipd wsl subcommand, but instead use usbipd bind and within WSL use usbip attach. In this case the check won't happen (and therefore won't fail).

So I did:
```
C:\windows\system32>usbipd bind -b 2-3
C:\windows\system32>usbipd list
Connected:
BUSID  DEVICE                                                        STATE
2-3    Périphérique série USB (COM4)                                 Shared
2-6    Integrated Webcam                                             Not shared
2-8    Dell ControlVault w/ Fingerprint Touch Sensor, Lecteur de...  Not shared
2-10   Intel(R) Wireless Bluetooth(R)                                Not shared
3-3    Realtek USB GbE Family Controller                             Not shared
4-1    Périphérique d’entrée USB                                     Not shared
4-2    Périphérique d’entrée USB                                     Not shared
4-5    Périphérique d’entrée USB                                     Not shared
5-1    Périphérique d’entrée USB, Billboard Interface                Not shared

Persisted:
GUID                                  DEVICE

```
The device 2-3 seems properly shared.
On my wsl I did:
```
$ usbip attach -b 2-3
usage: usbip attach <args>
    -r, --remote=<host>      The machine with exported USB devices
    -b, --busid=<busid>    Busid of the device on <host>
    -d, --device=<devid>    Id of the virtual UDC on <host>
```
Same result for:
`usbip attach --busid=2-3`

Is there something wrong with the syntax? Or am I missing something?

dorssel, 2022-02-01:
@tommasomazzoni 
You will have to set the "remote" host with -r, specifying the IP address of your Windows host.

But don't rule out `usbipd wsl attach` yet. Since version 2.0.0, you no longer have to run it elevated! Just use your normal user account, and your WSL distros will be found. This is much easier, as the IP address of the host will change after every boot.

tommasomazzoni, 2022-02-01:
Thank you for the quick reply!
Ok, I missed that one had to specify the host explicitly. Makes sense.
I have usbpid-win 2.1.0 installed on my system, but the command usbpid is not recognized in a non-elevated terminal. Is that normal?

dorssel, 2022-02-01:
@tommasomazzoni 
>I have usbpid-win 2.1.0 installed on my system, but the command usbpid is not recognized in a non-elevated terminal. Is that normal?

Ah, I missed that one. You are confusing "elevated" with "other (admin) user". You are talking about the latter. I didn't realize the installer only installed the PATH variable for the installing user. Most people only require elevation (which is still the same user, but with more privileges). However, you installed as a different user altogether, probably because your normal account is not a member of the admin group. In that case:

1) Apparently the PATH is set for that (other) user only
2) Running `wsl attach` won't work for that (other) user, because WSL instances are *per user*

You can fix (1) by adding `%ProgramFiles%\usbipd-win` to your PATH yourself. And (2) no longer applies, as `wsl attach` can now be run as a normal user.

I'll see if I can fix the installer by adding to the PATH for all users (if that is even possible...)

dorssel, 2022-02-01:
@tommasomazzoni 
Just checked: the PATH variable is already altered system-wide. Maybe you need to open a *new* command prompt after the installation to make it effective? Can you check?

tommasomazzoni, 2022-02-01:
Thanks for checking that out! 
I am actually a bit confused about the difference between a user with admin privileges and an admin user on Windows...
Anyway, indeed I can run the following in a new "normal" terminal :
```
C:\Users\tommaso.mazzoni>usbipd wsl attach --busid=2-3
usbipd: error: Access denied; this operation requires administrator privileges.
usbipd: info: The first time you attach a device to WSL requires administrator privileges; subsequent attaches will succeed with standard user privileges.
```
Is it what I should expect? If so, I'm basically back at square one, cause I can not run `usbipd wsl attach --busid=2-3` in a terminal with admin privileges.

dorssel, 2022-02-01:
@tommasomazzoni 
You are nearly there!
What you need is:
- as Admin : `usbipd bind --busid=2-3`
  (This is necessary, because your normal user is not part of the admin group; this does not involve WSL)
- as user: `usbipd wsl attach --busid=2-3`
  (This is now possible, because the device is already bound in the previous step)

tommasomazzoni, 2022-02-01:
That's it. Thank you for the clarification!
Great project!

PR #255 (2022-02-10): Improve access check
- differentiate between elevation (UAC) and standard/admin users

Fixes #243
Fixes #244
closed 2022-02-01 by tommasomazzoni