dorssel/usbipd-win #244: Using usbipd and WSL2 with separate Windows user and admin accounts?

pearson, 2022-02-02:
I've run into a problem using `usbpid` with WSL2 as I have separate Windows users and admin accounts. 

The issue appears to be that the Linux distribution is installed under the user account, but the use of `usbipd` requires the admin account. Using `wsl -l` will show my installed distribution in a regular command prompt, but the same command will show "Windows Subsystem for Linux has no installed distributions." from an admin command prompt.

Where this is a problem is in the first use of `usbipd wsl attach`:

- From an admin command prompt the command generates the error:
  - `usbipd: error: No default WSL distribution exists.`

- From a user command prompt the command generates the error:
  - `usbipd: error: Access denied; this operation requires administrator privileges.`
  - `usbipd: info: The first time you attach a device to WSL requires administrator privileges; subsequent attaches will succeed with standard user privileges.`

Is there a work-around for this? That "first time" administrator privileges requirement is getting in the way.

Thanks!

dorssel, 2022-02-02:
@pearson 
Workaround: https://github.com/dorssel/usbipd-win/issues/243#issuecomment-1027040799

pearson, 2022-02-02:
@dorssel 
Thanks for the reply! However, I was already using `usbipd wsl attach` as user, not `usbipd bind`.

I just stumbled onto the solution, though. I hadn't logged into the admin account since installing the distribution, and on logging in I was presented with the Ubuntu username/password setup.

The issue with `usbipd wsl attach` giving the admin privileges error may be related to an issue with the WSL kernel, unless a failed use of `usbipd wsl attach` from an admin command prompt is all that's needed (see below). 

In case this is useful, here's what I did:

1. Login as the admin account and do the Ubuntu setup
2. From the user account, `wsl -l` to confirm that there was now a default Linux distro. There was.
3. Admin prompt, user account: `usbipd wsl attach --busid 5-1`. This failed with the error `usbipd: error: WSL kernel is not USBIP capable; update with 'wsl --update'.`
4. Admin prompt, user account: `wsl --update`
5. Admin prompt, user account: `wsl --shutdown`
6. User prompt, user account: `usbipd wsl attach --busid 5-1`
7. Success

dorssel, 2022-02-02:
@pearson 
That's not what was wrong, or what happened. But anyway, glad you have it working.
I'll think about adding another sanity check and different instructions in case the WSL user is not a member of the Administrators group.

pearson, 2022-02-02:
@dorssel , I'm curious what the problem was, but I could wait for the sanity check / additional instructions next time I install. 😄

dorssel, 2022-02-02:
@dorssel 
OK, I'll elaborate, but you have to promise to test the PR once I have it!

`wsl attach` first does a number of sanity checks: is WSL installed, does the distro (if specified) exist, is it version 2, is it running, etc.
Then it checks if the device is already bound. That needs to be done only once. Once bound, future `wsl attach` will skip this. This is the step that requires Admin rights. Most users are a member of Administrators, so they just need elevation. However, some user use a "standard user" account, they require to do the bind as a *different* (admin) user. But this time, you cannot use `wsl attach` as WSL uses *per user* distros. So running `wsl attach` as a different (admin) user, cannot find any distros, and it will fail at the previously mentioned sanity checks.

This is why you got "distro not found" as admin, and "admin rights required" as user.

The solution in this case is: as admin, do not use `wsl attach`, but bind manually with `bind`. Now the user can attach with `wsl attach` as no further admin rights are required.

So now you are at the point: the usual sanity checks are done (WSL installed, distro exists, version 2, running), device already bound (no admin rights required) -> on to the next step: sanity checks *within* the distro: is the kernel the right version, is usbip client installed, can the network be reached. If all is well: attach!

What was necessary for you was: use `bind` as admin, then `wsl attach` as user, and then the rest of your issues (with proper feedback).

Why did I make all these sanity checks? To make users help themselves... It is almost complete, just missing: proper instructions in case the user is not a member of the admin group.

TL;DR: get ready to test that sanity check soon!

pearson, 2022-02-02:
> @dorssel OK, I'll elaborate, but you have to promise to test the PR once I have it!

Absolutely! 😃 I'm happy to help out. I really appreciate being able to use USB under WSL2! I also do appreciate the information. Obviously the interactions between WSL and Windows is something new to me, so some learning is a good thing.

Where my thinking went wrong here was in my understanding of distribution installs for separate user accounts. After reading the following big note on Microsoft's [Set up a WSL development environment](https://docs.microsoft.com/en-us/windows/wsl/setup/environment#set-up-your-linux-username-and-password) page, I'd assumed that the admin account simply didn't have a distribution installed and available. I was surprised to see that Ubuntu setup when I logged into the admin account.

> Linux distributions installed with WSL are a per-user installation and can't be shared with other Windows user accounts.

What might be worth mentioning is that I never did use a `bind` command. I've scrolled back through my administrator prompt to confirm that. The `usbipd wsl attach --busid 5-1` mentioned above is the command I used from the administrator prompt.

dorssel, 2022-02-02:
@pearson 
OK, now I am confused.
Is your user a member of the Administrators group? That means: do you use elevation (UAC) to gain admin privileges? Or do you have a standard user account, and do you login as a different user (username+password) to use admin rights?

pearson, 2022-02-02:
@dorssel 

I use a "standard" account, and when Windows requires administrator privileges, the UAC dialog box needs the administrator user name and password. (It's not the "click yes or no" UAC box one would see if the account was in the administrator's group.)

dorssel, 2022-02-02:
@pearson 
Ah, yes, that is what I thought.
But are you saying you had to install the distro as that admin (== different user), and now you can use it as the standard user? I would expect that the admin user now has a distro, and the standard user has none. Or do they both have (different) ones?

pearson, 2022-02-02:
I only installed the distro as the user. When I logged into the admin account, I was surprised to see the Ubuntu setup going on -- it is a separate account.

dorssel, 2022-02-02:
@pearson 
That is kind of weird indeed. So now the standard user has a distro, and the admin? Can it see it? What happens if you do `wsl --list` as admin?

pearson, 2022-02-02:
@dorssel , perhaps something else is going on here. The WSL install instructions had stated that a default Ubuntu would be installed, yet there was no distribution in my user account. I then installed Ubuntu 20.04 from the user account, and I've assumed that what I saw from the admin account was the same distribution.

When I check with `wsl -l` on each account, the user account is showing `Ubuntu-20.04 (Default)` while the admin account is showing `Ubuntu (Default)`. 

I'm now thinking that the admin distro is the one that was installed with WSL. I just never saw it because I hadn't logged into the admin account.

That said, it's confusing as to why `usbipd` is working. I'm guessing that your sanity check isn't checking that both the admin account and the user account are referring to the same default distribution? Eg. `usbipd` passed the sanity check on the admin account, set some flag, and then the user account could use it, even though they're working with different distributions?

dorssel, 2022-02-02:
@pearson 
It is starting to make sense now. You now have 2 distros, one in the admin account (which you can remove if you don't need it), and one in the user account (which is the one you are after).
`usbipd` doesn't care about WSL. (in fact: it was originally not even designed for WSL). It just needs an admin to "approve" (==bind) the use of a device by a client (any client!). All that `usbipd wsl` does is: look up the distro, and run the `usbip` client command *within* that distro. That's why it is perfectly fine to do the bind as whatever user with admin rights, and then you can use attach with whatever (other) user to whatever distro.

dorssel, 2022-02-02:
@pearson 
BTW: I am still going to provide a little better feedback for your case: admin rights required (for first bind) + not a member of Adminitrators -> display "you need to run `usbipd bind` from an admin account before you can use `wsl attach` for this device."
(or something similar...)

pearson, 2022-02-02:
@dorssel Thanks for all the info here, and sorry to have led you down the wrong path. 

When the RC comes out, I'll spin up a Windows 11 VM and see what it does -- and if I can still make it work without using `usbipd bind`. 😄

PR #255 (2022-02-10): Improve access check
- differentiate between elevation (UAC) and standard/admin users

Fixes #243
Fixes #244
closed 2022-02-02 by pearson