dorssel/usbipd-win #245: Failed to attached device with BUSID

cbeauregard-boreas, 2022-02-04:
I am on Windows 10 and I have Debian 11 for WSL2 distro (5.10.60.1-microsoft-standard-WSL2).
I have installed usbip on my Debian. Here is my problem.

```
PS C:\Users\CedrikBeauregard> usbipd wsl list
BUSID  DEVICE                                                        STATE
1-2    ST-Link Debug, USB Mass Storage Device, STMicroelectronic...  Not attached
1-3    USB Serial Device (COM16)                                     Not attached
1-4    USB Mass Storage Device                                       Not attached
1-6    Integrated Webcam                                             Not attached
1-10   Qualcomm QCA61x4A Bluetooth                                   Not attached

PS C:\Users\CedrikBeauregard> usbipd wsl attach --busid 1-3
usbip: error: Attach Request for 1-3 failed - Device in error state

usbipd: error: Failed to attach device with BUSID '1-3'.
PS C:\Users\CedrikBeauregard>
```

Here is logs:

```
PS C:\Users\CedrikBeauregard> usbipd server Logging:LogLevel:Default=Trace
dbug: Microsoft.Extensions.Hosting.Internal.Host[1]
      Hosting starting
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\Users\CedrikBeauregard
dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started
dbug: UsbIpServer.Server[1000]
      new connection from 192.168.87.34
dbug: UsbIpServer.Server[1000]
      connection closed
dbug: UsbIpServer.Server[1000]
      new connection from 192.168.87.34
dbug: UsbIpServer.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
fail: UsbIpServer.ConnectedClient[3]
      An exception occurred while communicating with the client:
      System.IO.FileNotFoundException: Unable to find the specified file.
         at UsbIpServer.ConfigurationManager.GetVBoxDevice(BusId busId) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConfigurationManager.cs:line 304
         at UsbIpServer.VBoxUsb.ClaimDeviceOnce(BusId busId) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsb.cs:line 66
         at UsbIpServer.VBoxUsb.ClaimDevice(BusId busId) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsb.cs:line 76
         at UsbIpServer.VBoxUsb.ClaimDevice(BusId busId) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsb.cs:line 82
         at UsbIpServer.ConnectedClient.HandleRequestImportAsync(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConnectedClient.cs:line 161
dbug: UsbIpServer.Server[1000]
      connection closed
```

dorssel, 2022-02-04:
What is the version of usbipd?
You may have to use `usbipd bind --force`. That will probably work, but I'm interested why this would be necessary in your case...

cbeauregard-boreas, 2022-02-04:
usbipd-win 2.1.0

Using `usbipd bind --busid 1-3 --force` and `usbipd wsl attach --busid 1-3` does not work. Here are the logs

```
PS C:\Users\CedrikBeauregard> usbipd server Logging:LogLevel:Default=Trace
dbug: Microsoft.Extensions.Hosting.Internal.Host[1]
      Hosting starting
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\Users\CedrikBeauregard
dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started
dbug: UsbIpServer.Server[1000]
      new connection from 172.27.219.247
dbug: UsbIpServer.Server[1000]
      connection closed
dbug: UsbIpServer.Server[1000]
      new connection from 172.27.219.247
dbug: UsbIpServer.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
info: UsbIpServer.ConnectedClient[1]
      Client 172.27.219.247 claimed device at 1-3 (USB\VID_0483&PID_5740\204330713850).
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=37, flags=512, length=64, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 256 0 64
trce: UsbIpServer.AttachedClient[1001]
      actual: 18, requested: 64
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=38, flags=512, length=18, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 256 0 18
trce: UsbIpServer.AttachedClient[1001]
      actual: 18, requested: 18
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=39, flags=512, length=10, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 1536 0 10
dbug: UsbIpServer.AttachedClient[1000]
      USBSUP_XFER_STALL -> EPIPE -> -32
trce: UsbIpServer.AttachedClient[1001]
      actual: 0, requested: 10
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=40, flags=512, length=10, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 1536 0 10
dbug: UsbIpServer.AttachedClient[1000]
      USBSUP_XFER_STALL -> EPIPE -> -32
trce: UsbIpServer.AttachedClient[1001]
      actual: 0, requested: 10
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=41, flags=512, length=10, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 1536 0 10
dbug: UsbIpServer.AttachedClient[1000]
      USBSUP_XFER_STALL -> EPIPE -> -32
trce: UsbIpServer.AttachedClient[1001]
      actual: 0, requested: 10
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=42, flags=512, length=9, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 512 0 9
trce: UsbIpServer.AttachedClient[1001]
      actual: 9, requested: 9
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=43, flags=512, length=75, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 512 0 75
trce: UsbIpServer.AttachedClient[1001]
      actual: 75, requested: 75
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=44, flags=512, length=255, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 768 0 255
trce: UsbIpServer.AttachedClient[1001]
      actual: 4, requested: 255
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=45, flags=512, length=255, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 770 1033 255
trce: UsbIpServer.AttachedClient[1001]
      actual: 48, requested: 255
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=46, flags=512, length=255, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 769 1033 255
trce: UsbIpServer.AttachedClient[1001]
      actual: 40, requested: 255
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=47, flags=512, length=255, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 771 1033 255
trce: UsbIpServer.AttachedClient[1001]
      actual: 26, requested: 255
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=48, flags=0, length=0, ep=0
dbug: UsbIpServer.AttachedClient[1000]
      Trapped SET_CONFIGURATION: 1
info: UsbIpServer.ConnectedClient[2]
      Client 172.27.219.247 released device at 1-3 (USB\VID_0483&PID_5740\204330713850).
dbug: UsbIpServer.ConnectedClient[1000]
      Unbind or unplug while attached
fail: UsbIpServer.ConnectedClient[3]
      An exception occurred while communicating with the client:
      System.ComponentModel.Win32Exception (87): DeviceIoControl returned error ERROR_INVALID_PARAMETER
         at UsbIpServer.AttachedClient.HandleSubmitAsync(UsbIpHeaderBasic basic, UsbIpHeaderCmdSubmit submit, CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\AttachedClient.cs:line 303
         at UsbIpServer.AttachedClient.RunAsync(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\AttachedClient.cs:line 517
         at UsbIpServer.ConnectedClient.HandleRequestImportAsync(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConnectedClient.cs:line 223
         at UsbIpServer.ConnectedClient.HandleRequestImportAsync(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConnectedClient.cs:line 246
dbug: UsbIpServer.Server[1000]
      connection closed
```

cbeauregard-boreas, 2022-02-04:
I tried with a USB Mass storage and it is working. So it seems to be related with my device, which is a COM port on Windows.

dorssel, 2022-02-05:
@cbeauregard-boreas 

>dbug: UsbIpServer.AttachedClient[1000]
>      Trapped SET_CONFIGURATION: 1
>dbug: UsbIpServer.ConnectedClient[1000]
>      Unbind or unplug while attached

Looks like your device "resets" or triggers a "replug" when its configuration is changed. The device originally attaches correctly, and WSL is communicating with it correctly, but when the Linux driver sets the configuration it "disappears".

Can you post the output of syslog or dmesg?
And can you try PR #247? The installer is at https://github.com/dorssel/usbipd-win/actions/runs/1798620074.

hegza, 2022-02-12:
Similar looking re-plug situation here, but I found an interesting workaround for my case and thought I'd share it. I'm using usbipd to patch a USB-connected Longan Nano board to WSL2 for flashing with dfu-util.

Basically, there's two modes the board boots in that I shall call DEFAULT and FLASH. I need to get access in WSL2 while it's in FLASH mode, but rebooting the board always puts it in DEFAULT, **unless** one is pressing the flash button.

If I interrupt the usbipd attach command by resetting-the board in FLASH mode quickly, it manages to patch through to WSL2 without failing.

## Proceedings
1. Connect board while pressing a button to get it connected in FLASH mode
    - Device is available in Windows through e.g. `dfu-util --list` as expected
    - Not yet available in WSL2 `dfu-util --list` as expected
2. In Windows: `usbipd wsl attach --busid=<BUSID>` -> device re-connects in DEFAULT mode which causes the command to fail similar to this issue.

Now, if I use the button to quickly reset the board in FLASH mode at step 2 while usbipd is working, the patch-through succeeds and the device becomes available to dfu-util in both Windows and WSL2.

A more convenient solution for my case would be if I, or the tool, was able to set up a callback or a daemon to trigger that particular device to always automatically use the settings that usbipd actively achieves.

dorssel, 2022-02-12:
@hegza 
Thanks for the detailed report!

I am still trying to figure out at what point exactly such devices "restart into the other mode". Is it a timer? is it when the USB port resets? Or is it when a specific URB is sent to them.

Could you try PR https://github.com/dorssel/usbipd-win/pull/247? The installer is at https://github.com/dorssel/usbipd-win/actions/runs/1798620074. And then `usbipd bind --force` the device before attaching to WSL? This will *not* load a new driver upon attach (the driver is permanently loaded by --force), will *not* reset the USB port, and will *not* unconfigure the device before use (that's what the PR is for). I wonder if in that case, the replug won't happen...

hegza, 2022-02-15:
Attempted as instructed. Results in a different sort of failure. Based on the USB beeps on Windows I think it works as you describe. Proceedings:

0. Device connected in FLASH mode at bus-id 1-22. Visible to Windows, not visible to Linux.

1. `usbipd bind --force --busid=1-22`

This seems to reset the USB, which power-cycles the device, which reboots the device in DEFAULT. We recall that the device is only available to Windows in FLASH, so we're fine with that and move to next step.

2. Reset the device in FLASH

Deducing from Windows beeps, the driver is now indeed **not** unloaded (no disconnect sound) **unlike** in previous test case. After reset in FLASH mode, we can also now see it listed in Windows with e.g. `dfu-util --list` but not yet in WSL as expected, and same as in last test case.

3. `usbipd wsl attach --busid=1-22`

Suspiciously long wait, after which...
> `**USB connect sound**`
> usbip: error: Attach Request for 1-22 failed - Device in error state
> usbipd: error: Failed to attach device with BUSID '1-22'.

Additionally, the device vanishes from Windows (e.g. `dfu-util --list`) **specifically** after this step. Returns to windows with "reset in FLASH" (but not WSL as usual).

cBeaur, 2022-03-08:
Your were right, here is the dmesg output from wsl:
`[96167.339420] usb 1-1: New USB device found, idVendor=0483, idProduct=5740, bcdDevice= 2.00
[96167.339423] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[96167.339464] usb 1-1: Product: Boreas DevKit (BOS0614)
[96167.339481] usb 1-1: Manufacturer: Boreas Technologies
[96167.339482] usb 1-1: SerialNumber: 204330713850
[96167.600342] vhci_hcd: connection closed
[96167.600487] vhci_hcd: stop threads
[96167.600489] vhci_hcd: release socket
[96167.600492] vhci_hcd: disconnect device
[96172.370365] usb 1-1: can't set config #1, error -110
[96172.370752] usb 1-1: USB disconnect, device number 2`

I installed the PR you mentionned (247)

cBeaur, 2022-03-08:
Here is syslog:
```
Mar  8 15:09:21 DESKTOP-J8HGU5V kernel: [96780.792910] vhci_hcd vhci_hcd.0: pdev(0) rhport(0) sockfd(3)
Mar  8 15:09:21 DESKTOP-J8HGU5V kernel: [96780.792913] vhci_hcd vhci_hcd.0: devid(65538) speed(2) speed_str(full-speed)
Mar  8 15:09:21 DESKTOP-J8HGU5V kernel: [96780.792954] vhci_hcd vhci_hcd.0: Device attached
Mar  8 15:09:21 DESKTOP-J8HGU5V kernel: [96781.070384] vhci_hcd: vhci_device speed not set
Mar  8 15:09:21 DESKTOP-J8HGU5V kernel: [96781.140147] usb 1-1: new full-speed USB device number 3 using vhci_hcd
Mar  8 15:09:21 DESKTOP-J8HGU5V kernel: [96781.220099] vhci_hcd: vhci_device speed not set
Mar  8 15:09:21 DESKTOP-J8HGU5V kernel: [96781.290242] usb 1-1: SetAddress Request (3) to port 0
Mar  8 15:09:21 DESKTOP-J8HGU5V kernel: [96781.324374] usb 1-1: config 1 has an invalid interface number: 2 but max is 1
Mar  8 15:09:21 DESKTOP-J8HGU5V kernel: [96781.324376] usb 1-1: config 1 has an invalid interface number: 3 but max is 1
Mar  8 15:09:21 DESKTOP-J8HGU5V kernel: [96781.324377] usb 1-1: config 1 has no interface number 0
Mar  8 15:09:21 DESKTOP-J8HGU5V kernel: [96781.324377] usb 1-1: config 1 has no interface number 1
Mar  8 15:09:21 DESKTOP-J8HGU5V kernel: [96781.327365] usb 1-1: New USB device found, idVendor=0483, idProduct=5740, bcdDevice= 2.00
Mar  8 15:09:21 DESKTOP-J8HGU5V kernel: [96781.327368] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Mar  8 15:09:21 DESKTOP-J8HGU5V kernel: [96781.327369] usb 1-1: Product: Boreas DevKit (BOS0614)
Mar  8 15:09:21 DESKTOP-J8HGU5V kernel: [96781.327370] usb 1-1: Manufacturer: Boreas Technologies
Mar  8 15:09:21 DESKTOP-J8HGU5V kernel: [96781.327371] usb 1-1: SerialNumber: 204330713850
Mar  8 15:09:22 DESKTOP-J8HGU5V kernel: [96781.564008] vhci_hcd: connection closed
Mar  8 15:09:22 DESKTOP-J8HGU5V kernel: [96781.564468] vhci_hcd: stop threads
Mar  8 15:09:22 DESKTOP-J8HGU5V kernel: [96781.564472] vhci_hcd: release socket
Mar  8 15:09:22 DESKTOP-J8HGU5V kernel: [96781.564478] vhci_hcd: disconnect device
Mar  8 15:09:27 DESKTOP-J8HGU5V kernel: [96786.370590] usb 1-1: can't set config #1, error -110
Mar  8 15:09:27 DESKTOP-J8HGU5V kernel: [96786.370774] usb 1-1: USB disconnect, device number 3
```

dorssel, 2022-03-09:
@cBeaur 
There seems to be a problem with the device's configuration descriptor. Looks like it exposes 2 interfaces, but as number 2 and 3 instead of 0 and 1. Interface 0 is essential; every device must have it.
Can you provide the output of `usbview.exe` (part of Windows SDK)?

cBeaur, 2022-03-14:
Oh ok I can see it with usbview.exe. The problem must be on my side then, because we write our descriptor for our board. I will let you know if I can fix it.

cBeaur, 2022-03-16:
@dorssel 
This is like you mentioned, my device did not have the good interfaces. I forced it to have at least 0 as interface # and it is connecting to my wsl!

dorssel, 2022-03-16:
@cBeaur 
Good to hear. Can we close this issue? Feel free to open a new one if needed in the future.

closed 2022-03-17 by cBeaur

pansila, 2024-01-02:
Just FYI, I encountered the same issue, and found it's caused the USB hub which connects my USB mass storage device. After removing the hub and connects my USB device to  the PC's USB port, it workd.

twaldecker, 2025-03-17:
Thanks @pansila 

I plugged my USB2.0 device to another port on my Computer probably wired differently