dorssel/usbipd-win #251: usbipd: error: WSL 'usbip' client not correctly installed

JohnBerg60, 2022-02-07:
After reinstalling my wsl2 Debian I get ```usbipd: error: WSL 'usbip' client not correctly installed```. Very strange, it worked flawless for quite some time.

The server console shows no errors, and from the wsl2 side I can use ```usbip list --remote="$(hostname).local"``` with this output:

```
Exportable USB devices
======================
 - JOHN-LEGION5.local
        4-2: STMicroelectronics : ST-LINK/V2.1 (0483:374b)
           : USB\VID_0483&PID_374B\0667FF544949878667163348
           : Miscellaneous Device / ? / Interface Association (ef/02/01)
           :  0 - Vendor Specific Class / Vendor Specific Subclass / Vendor Specific Protocol (ff/ff/ff)
           :  1 - Mass Storage / SCSI / Bulk-Only (08/06/50)
           :  2 - Communications / Abstract (modem) / AT-commands (v.25ter) (02/02/01)
           :  3 - CDC Data / Unused / unknown protocol (0a/00/00)
```

and this request is also shown correlecty on the server console:
```
dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started
dbug: UsbIpServer.Server[1000]
      new connection from 172.27.45.95
dbug: UsbIpServer.ConnectedClient[1000]
      Received opcode: OP_REQ_DEVLIST
dbug: UsbIpServer.Server[1000]
      connection closed
```

I can share the device from the wsl side: ```usbip attach --remote="$(hostname).local" --busid=4-2``` but not from the windows side.

What am I doing wrong?

dorssel, 2022-02-07:
@JohnBerg60 
You moved from 1.x to 2.x. See release notes: https://github.com/dorssel/usbipd-win/releases/tag/v2.0.0.

JohnBerg60, 2022-02-07:
I was on 1.3.1. I've upgraded to 2.1.0. The error message is slighly changed: 
```
usbipd: error: WSL 'usbip' client not correctly installed. See https://github.com/dorssel/usbipd-win/wiki/WSL-support for the latest instructions.
```

As I am on Debian WSL, maybe WSL [issue#7652](https://github.com/microsoft/WSL/issues/7652) ?

dorssel, 2022-02-07:
The usbip client command must be correctly installed for user root. The previous instructions only made it work for the default user.

You reported that usbip can be executed correctly as user. Make it so that after `sudo su -` (i.e. from a root shell) you can also run usbip. Perhaps you need to add it to root's path. Or adjust the update-alternative for your distro.

JohnBerg60, 2022-02-07:
no problem on the wsl2 side:
```
root@JOHN-LEGION5:[/mnt/c/Users/John]: whoami
root
root@JOHN-LEGION5:[/mnt/c/Users/John]: usbip list --remote="$(hostname).local"
Exportable USB devices
======================
 - JOHN-LEGION5.local
        2-2: STMicroelectronics : ST-LINK/V2.1 (0483:374b)
           : USB\VID_0483&PID_374B\0667FF544949878667163348
           : Miscellaneous Device / ? / Interface Association (ef/02/01)
           :  0 - Vendor Specific Class / Vendor Specific Subclass / Vendor Specific Protocol (ff/ff/ff)
           :  1 - Mass Storage / SCSI / Bulk-Only (08/06/50)
           :  2 - Communications / Abstract (modem) / AT-commands (v.25ter) (02/02/01)
           :  3 - CDC Data / Unused / unknown protocol (0a/00/00)
```

dorssel, 2022-02-07:
On your distro, what is the output of `usbip version`?

dorssel, 2022-02-07:
Weird. This looks all OK. One more test. What is the output from *Windows* side:
```
wsl --user root -- usbip version
```

JohnBerg60, 2022-02-07:
```
C:\Users\John>wsl --user root -- usbip version
/bin/sh: usbip: not found
```

That is very strange .. but consistent  .. it works from the wsl side, but windows has no access ..

dorssel, 2022-02-07:
Are you perhaps running `wsl` (Windows side) as a different user?

dorssel, 2022-02-07:
More testing: what is the output (as root) for `which usbip`?

JohnBerg60, 2022-02-07:
No .. I have only 1 account on my machine.

output: ```/usr/sbin/usbip```

dorssel, 2022-02-07:
What is the output of (Windows side): `wsl --user root -- set`?

JohnBerg60, 2022-02-07:
Wooh .. thats a rather long one. I assume you are looking for ```WSL_DISTRO_NAME='docker-desktop'``` ?

I assume this is not correct ( remember: I removed the Debian Distro, and installed a clean one)

JohnBerg60, 2022-02-07:
Running ```wsl --set-default Debian``` solved the issue!

My Debian instal was corrupted, so I removed it, and reinstalled it from the MS Store. But I am also running Docker, and Windows decided in its ultimate wisdom that Docker should be my default distro ..

closed 2022-02-07 by JohnBerg60

JohnBerg60, 2022-02-07:
Nice side effect: as there is no way to get usb support in a docker container running linux on a windows host, this seems to work now, without any additional install in the docker image.
I was flabergasted to see:
```
root@09081f5c10cc:[/src]: lsusb
Bus 002 Device 001: ID 1d6b:0003
Bus 001 Device 002: ID 0483:374b
Bus 001 Device 001: ID 1d6b:0002
```
where Device 002 is my ST-Link adapter. As said: I havent installed any usb related software (except lsusb) installed in the docker container. 

Thanks for all your help!

dorssel, 2022-02-07:
No problem. Glad you have it working. I'll add an informational message in case:

- the user has more than 1 distro
- AND the user did not specify a specific distro with `usbipd wsl --distribution`

that will say: "Using default distribution '\<NAME\>'".
Too late for you, but it may help others...

xiaolongba, 2022-03-26:
Hello @dorssel 

I also ran into this issue,

```c
PS C:\WINDOWS\system32> usbipd wsl attach --busid=8-1
usbipd: error: WSL 'usbip' client not correctly installed. See https://github.com/dorssel/usbipd-win/wiki/WSL-support for the latest instructions.
```

the all commands you mentioned above, as follows:

1. root user
```
root@e17a6e226adb:/workspaces/hello_world# whoami
root
```
2. usbip version
```
root@e17a6e226adb:/workspaces/hello_world# usbip version
usbip (usbip-utils 2.0)
```
3. 
```
PS C:\WINDOWS\system32> wsl --user root -- usbip version
/bin/bash: usbip: command not found
```
4. 
```
root@e17a6e226adb:/workspaces/hello_world# which usbip
/usr/sbin/usbip
```

5. 
```
PS C:\WINDOWS\system32> wsl --user root -- set
BASH=/bin/bash
BASHOPTS=checkwinsize:cmdhist:complete_fullquote:extquote:force_fignore:globasciiranges:hostcomplete:interactive_comments:progcomp:promptvars:sourcepath
BASH_ALIASES=()
BASH_ARGC=()
BASH_ARGV=()
BASH_CMDS=()
BASH_EXECUTION_STRING=set
BASH_LINENO=()
BASH_SOURCE=()
BASH_VERSINFO=([0]="5" [1]="0" [2]="16" [3]="1" [4]="release" [5]="x86_64-pc-linux-gnu")
BASH_VERSION='5.0.16(1)-release'
DIRSTACK=()
EUID=0
GROUPS=()
HOME=/root
HOSTNAME=LAPTOP-GUPHVV02
HOSTTYPE=x86_64
IFS=$' \t\n'
LANG=C.UTF-8
LOGNAME=root
MACHTYPE=x86_64-pc-linux-gnu
NAME=LAPTOP-GUPHVV02
OPTERR=1
OPTIND=1
OSTYPE=linux-gnu
PATH='/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/usr/lib/wsl/lib:/mnt/d/Python39/Scripts/:/mnt/d/Python39/:/mnt/c/Program Files (x86)/Common Files/Oracle/Java/javapath:/mnt/c/windows/system32:/mnt/c/windows:/mnt/c/windows/System32/Wbem:/mnt/c/windows/System32/WindowsPowerShell/v1.0/:/mnt/c/windows/System32/OpenSSH/:/mnt/c/Program Files (x86)/NVIDIA Corporation/PhysX/Common:/mnt/c/Program Files (x86)/HP/HP Performance Advisor:/mnt/d/PuTTY/:/mnt/c/WINDOWS/system32:/mnt/c/WINDOWS:/mnt/c/WINDOWS/System32/Wbem:/mnt/c/WINDOWS/System32/WindowsPowerShell/v1.0/:/mnt/c/WINDOWS/System32/OpenSSH/:/mnt/c/Program Files/Intel/WiFi/bin/:/mnt/c/Program Files/Common Files/Intel/WirelessCommon/:/mnt/c/Program Files (x86)/IVI Foundation/VISA/WinNT/Bin/:/mnt/c/Program Files/IVI Foundation/VISA/Win64/Bin/:/mnt/c/Program Files (x86)/IVI Foundation/VISA/WinNT/Bin:/mnt/d/Git/cmd:/mnt/c/Program Files (x86)/STMicroelectronics/STM32 ST-LINK Utility/ST-LINK Utility:/mnt/c/Program Files (x86)/IVI Foundation/IVI/bin:/mnt/c/Program Files/IVI Foundation/IVI/bin:/mnt/d/CMake/bin:/mnt/c/Program Files/Nordic Semiconductor/nrf-command-line-tools/bin/:/mnt/d/doxygen/bin:/mnt/d/Mscgen:/mnt/d/stlink-1.3.0-win64/bin:/mnt/c/Program Files (x86)/Intel/Intel(R) Management Engine Components/DAL:/mnt/c/Program Files/Intel/Intel(R) Management Engine Components/DAL:/mnt/c/ProgramData/chocolatey/bin:/mnt/d/nodejs/:/mnt/c/Program Files/usbipd-win/:/mnt/c/Program Files/Docker/Docker/resources/bin:/mnt/c/ProgramData/DockerDesktop/version-bin:/mnt/c/Users/wireless_tech/AppData/Local/Microsoft/WindowsApps:/mnt/d/Microsoft VS Code/bin:/mnt/c/Users/wireless_tech/AppData/Roaming/npm'
PPID=574
PS4='+ '
PWD=/mnt/c/WINDOWS/system32
SHELL=/bin/bash
SHELLOPTS=braceexpand:hashall:interactive-comments
SHLVL=1
TERM=xterm-256color
UID=0
USER=root
WSLENV=
WSL_DISTRO_NAME=Ubuntu-20.04
WSL_INTEROP=/run/WSL/574_interop
_=/bin/bash
```

can you give me your hand?

thanks in advance

dorssel, 2022-03-26:
@xiaolongba 
It looks like you are using separate admin/standard user accounts. Could that be it?
See #282.
If this is the case, run `usbipd wsl attach` as your normal user instead of as admin.

xiaolongba, 2022-03-26:
@dorssel 

I followed the instructions you mentioned above, but the issue is still exist, as below:
[image]

anything I missed?

dorssel, 2022-03-26:
@xiaolongba 
The last command (`usbipd wsl attach`) should *not* be run from an admin terminal, but instead from a normal terminal for your standard user account.

(this is assuming that admin/standard user issues is indeed the problem you are having).

xiaolongba, 2022-03-26:
@dorssel 

Ok, it seems that the issue is still present, as below:
[image]

dorssel, 2022-03-26:
@xiaolongba 
On the right-side terminal (normal user), what is the output of:
```
wsl --user root usbip version
```

xiaolongba, 2022-03-26:
@dorssel 

the output as below:
```c
PS C:\Users\wireless_tech> wsl --user root usbip version
/bin/bash: usbip: command not found
PS C:\Users\wireless_tech>
```

dorssel, 2022-03-26:
@xiaolongba 
There is your problem:
You have been mixing up WSL instances. They are *per user*. Since you use different users for normal Windows use (standard user) and admin (a *different* user with admin right). They both have their own WSL instances. You have set up `usbip` on one of the admin WSL instances, but not on your WSL instance for your normal user.

See #244.

You will see that `wsl --list` may even produce different results for the two users. Or not, if they just happen to have the same distros installed. But in reality, they are *different* instances.

I recommend: use the standard user WSL instances only. In fact, remove the admin user WSL instances. And only use `wsl` and `usbipd wsl` commands as standard user from now on.

xiaolongba, 2022-03-26:
@dorssel 

I type in the `wsl --list` on the admin prompt and user prompt, the results are the same:
[image]

I also checked the user account I used, it is a member of the Administrators group. so to use usbipd, I need to change the current account to a standard user? right?

dorssel, 2022-03-26:
@xiaolongba 
OK, so that's not it. You are using a user that is itself a member of Administrators.

Earlier you reported that:
```
root@e17a6e226adb:/workspaces/hello_world# usbip version
usbip (usbip-utils 2.0)
```

but also
```
PS C:\WINDOWS\system32> wsl --user root -- usbip version
/bin/bash: usbip: command not found
```

So, when logged in as root within Linux it works, but when running as root from Windows it does not?
Can you reconfirm by running from the same terminal:

```
windows> wsl --user root -- usbip version
```

and 

```
windows> wsl --root
linux# usbip version
```

xiaolongba, 2022-03-26:
> windows> wsl --user root -- usbip version

> windows> wsl --root
[image]

the root user name for win is wireless_tech, not root

> linux# usbip version

[image]

dorssel, 2022-03-26:
I've corrected the command, should be

wsl --user root

Please try again.

xiaolongba, 2022-03-26:
> wsl --user root

[image]

dorssel, 2022-03-26:
Obviously followed by the usbip version command...

xiaolongba, 2022-03-26:
> Obviously followed by the usbip version command...

sorry, what do you mean?

xiaolongba, 2022-03-26:
Hello @dorssel Bro.

maybe I missed some important info, based on your tips, I tried to type in the command `wsl --user root` on admin prompt, and then it goes into the shell of Linux, and then I tried to type in the command `usbip --version`, but it stills showed me 
```
-sh: usbip: not found
```

However, I usbip I installed based on the docker image, it is also based on WSL2. as below:
[image]

so it seems that the usbip I installed locates at the WSL of the docker image, not the local wsl. so if my understanding is correct, how to move to wsl of the docker image, not the local wsl?

dorssel, 2022-03-26:
@xiaolongba 
Ah. So the problem has been found. You installed `usbip` for the docker image, but not in the root WSL, where it needs to be.

You will probably have to do:
```
windows> wsl --user root
linux# apt install <whatever_you_need_for_your_distro>
linux# <optional_command_to_ensure_that_usbip_is_in_the_path_of_root>
```

The instructions for Ubuntu are on the wiki page. Other distros may require different instructions.
Note: the `usbipd wsl attach` must be done on the root WSL. Then the device will be available in docker as well (as long as you start docker *after* the attach).

xiaolongba, 2022-03-26:
@dorssel 
big thanks, bro. another issue is coming, when I tried to type in `apt install linux-tools-5.4.0-77-generic hwdata` in the root WSL, it showed me the following error:

[image]

that made me crazy, so do I need to install an extra WSL? such as ubuntu 20.4 LTS

dorssel, 2022-03-26:
@xiaolongba 
What distro are you using? `apt` is for debian-like distros. Maybe you need `yum`? What package manager did you use to install usbip in the docker?

xiaolongba, 2022-03-26:
@dorssel 

as you see, I dont install any distro, there are already present after installing the docker desktop for windows app, as below:
[image]

so it should be Debian also, but without apt or apt-get.

> What package manager did you use to install usbip in the docker?

I use the command I mentioned above, `apt install linux-tools-5.4.0-77-generic hwdata`, it works well. based on the results, it seems that the root WSL is different from the WSL docker used.

dorssel, 2022-03-26:
@xiaolongba 
I'm getting really confused. You are still using administrator command prompts...
From a normal command prompt (standard user rights), what is the output of:
```
windows> wsl whoami
windows> wsl usbip version
windows> wsl which usbip
windows> wsl --user root set
windows> wsl --user root usbip version
```

I am not familiar with docker desktop. Maybe search the other issues/discussions, also the closed ones. Other people may have solved this and are better able to help you.

xiaolongba, 2022-03-26:
@dorssel I am also really confused. the outputs as below:

```c
Windows PowerShell
Copyright (C) Microsoft Corporation. All rights reserved.

Try the new cross-platform PowerShell https://aka.ms/pscore6

PS C:\Users\wireless_tech> wsl whoami
root
PS C:\Users\wireless_tech> wsl usbip version
/bin/sh: usbip: not found
PS C:\Users\wireless_tech> wsl which usbip
PS C:\Users\wireless_tech> wsl --user root set
HOME='/root'
HOSTNAME='LAPTOP-GUPHVV02'
HOSTTYPE='x86_64'
IFS='
'
LANG='en_US.UTF-8'
LINENO=''
LOGNAME='root'
NAME='LAPTOP-GUPHVV02'
OPTIND='1'
PATH='/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/usr/lib/wsl/lib:/tmp/docker-desktop-root/mnt/host/d/Python39/Scripts/:/tmp/docker-desktop-root/mnt/host/d/Python39/:/tmp/docker-desktop-root/mnt/host/c/Program Files (x86)/Common Files/Oracle/Java/javapath:/tmp/docker-desktop-root/mnt/host/c/windows/system32:/tmp/docker-desktop-root/mnt/host/c/windows:/tmp/docker-desktop-root/mnt/host/c/windows/System32/Wbem:/tmp/docker-desktop-root/mnt/host/c/windows/System32/WindowsPowerShell/v1.0/:/tmp/docker-desktop-root/mnt/host/c/windows/System32/OpenSSH/:/tmp/docker-desktop-root/mnt/host/c/Program Files (x86)/NVIDIA Corporation/PhysX/Common:/tmp/docker-desktop-root/mnt/host/c/Program Files (x86)/HP/HP Performance Advisor:/tmp/docker-desktop-root/mnt/host/d/PuTTY/:/tmp/docker-desktop-root/mnt/host/c/WINDOWS/system32:/tmp/docker-desktop-root/mnt/host/c/WINDOWS:/tmp/docker-desktop-root/mnt/host/c/WINDOWS/System32/Wbem:/tmp/docker-desktop-root/mnt/host/c/WINDOWS/System32/WindowsPowerShell/v1.0/:/tmp/docker-desktop-root/mnt/host/c/WINDOWS/System32/OpenSSH/:/tmp/docker-desktop-root/mnt/host/c/Program Files/Intel/WiFi/bin/:/tmp/docker-desktop-root/mnt/host/c/Program Files/Common Files/Intel/WirelessCommon/:/tmp/docker-desktop-root/mnt/host/c/Program Files (x86)/IVI Foundation/VISA/WinNT/Bin/:/tmp/docker-desktop-root/mnt/host/c/Program Files/IVI Foundation/VISA/Win64/Bin/:/tmp/docker-desktop-root/mnt/host/c/Program Files (x86)/IVI Foundation/VISA/WinNT/Bin:/tmp/docker-desktop-root/mnt/host/d/Git/cmd:/tmp/docker-desktop-root/mnt/host/c/Program Files (x86)/STMicroelectronics/STM32 ST-LINK Utility/ST-LINK Utility:/tmp/docker-desktop-root/mnt/host/c/Program Files (x86)/IVI Foundation/IVI/bin:/tmp/docker-desktop-root/mnt/host/c/Program Files/IVI Foundation/IVI/bin:/tmp/docker-desktop-root/mnt/host/d/CMake/bin:/tmp/docker-desktop-root/mnt/host/c/Program Files/Nordic Semiconductor/nrf-command-line-tools/bin/:/tmp/docker-desktop-root/mnt/host/d/doxygen/bin:/tmp/docker-desktop-root/mnt/host/d/Mscgen:/tmp/docker-desktop-root/mnt/host/d/stlink-1.3.0-win64/bin:/tmp/docker-desktop-root/mnt/host/c/Program Files (x86)/Intel/Intel(R) Management Engine Components/DAL:/tmp/docker-desktop-root/mnt/host/c/Program Files/Intel/Intel(R) Management Engine Components/DAL:/tmp/docker-desktop-root/mnt/host/c/ProgramData/chocolatey/bin:/tmp/docker-desktop-root/mnt/host/d/nodejs/:/tmp/docker-desktop-root/mnt/host/c/Program Files/Docker/Docker/resources/bin:/tmp/docker-desktop-root/mnt/host/c/ProgramData/DockerDesktop/version-bin:/tmp/docker-desktop-root/mnt/host/c/Program Files/usbipd-win/:/tmp/docker-desktop-root/mnt/host/c/Users/wireless_tech/AppData/Local/Microsoft/WindowsApps:/tmp/docker-desktop-root/mnt/host/d/Microsoft VS Code/bin:/tmp/docker-desktop-root/mnt/host/c/Users/wireless_tech/AppData/Roaming/npm'
PPID='19580'
PS1='\w \$ '
PS2='> '
PS4='+ '
PWD='/tmp/docker-desktop-root/mnt/host/c/Users/wireless_tech'
SHELL='/bin/sh'
SHLVL='1'
TERM='xterm-256color'
USER='root'
WSLENV=''
WSL_DISTRO_NAME='docker-desktop'
WSL_INTEROP='/run/WSL/19580_interop'
PS C:\Users\wireless_tech> wsl --user root usbip version
/bin/sh: usbip: not found
PS C:\Users\wireless_tech>

```

I think the problem is caused by the default distro of docker image, I will try to switch to Ubuntu 20.4 LTS and try again.

I will feedback to you when I resolved it, anyway, big big thanks for your support, you are nice man.

thanks again.

xiaolongba, 2022-03-27:
@dorssel Bro,

Finally, I resolved this issue. and Now it works well. let me summarize my case here.

1. root cause
My pc has not installed WSL before, but the WSL is exist already after installing the docker desktop windows app, so I misunderstood that the local WSL is the same as the WSL docker used. but the truth is that they are different. at least, I installed usbip in the local WSL, but the WSL docker used has not such a usbip. so it returned the error info `usbipd: error: WSL 'usbip' client not correctly installed. See https://github.com/dorssel/usbipd-win/wiki/WSL-support for the latest instructions.
` when typing in the `usbipd wsl attach --busid=8-1` in standard user cmd prompt. actually, the usbip has been installed in the WSL docker used, not the local WSL; However, the command `usbipd wsl attach --busid=8-1` has been called in the local WSL, of course it returned with the error info as the local WSL has not installed usbip.

2. workaround
    1. installing the local WSL manually with  `wsl --install -d <distribution name>`, such as  `wsl --install -d Ubuntu`
    2. set the default distro, such as  `wsl --set-default Ubuntu-20.04`  
    3. follow the instructions which are from [USB/IP client tools](https://github.com/dorssel/usbipd-win/wiki/WSL-support#usbip-client-tools)
    4. and then type in the command `usbipd bind --busid 8-1 --force` in the admin prompt, where 8-1 is the device I would like to bind
    5. type in another command `usbipd wsl attach --busid 8-1` in the standard user prompt, where 8-1 is the device I would like to attach
    6. and then switch to WSL with root rights, the command is `wsl --user root` and then type in ` dmesg | tail`, the output as below:
  ```
[ 3018.496089] vhci_hcd: vhci_device speed not set
[ 3018.565897] usb 1-1: new full-speed USB device number 5 using vhci_hcd
[ 3018.645856] vhci_hcd: vhci_device speed not set
[ 3018.715855] usb 1-1: SetAddress Request (5) to port 0
[ 3018.752468] usb 1-1: New USB device found, idVendor=303a, idProduct=1001, bcdDevice= 1.01
[ 3018.752469] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 3018.752470] usb 1-1: Product: USB JTAG/serial debug unit
[ 3018.752471] usb 1-1: Manufacturer: Espressif
[ 3018.752472] usb 1-1: SerialNumber: 7C:DF:A1:BB:E6:60
[ 3018.754450] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
```

dorssel, 2022-03-27:
@xiaolongba 
Well done!
For my information: what does `wsl --list` show now?

xiaolongba, 2022-03-28:
> For my information: what does `wsl --list` show now?

```
PS C:\WINDOWS\system32> wsl --list
Windows Subsystem for Linux Distributions:
Ubuntu-20.04 (Default)
docker-desktop-data
docker-desktop
```
by the way. it seems that the COM port has lost on windows side after using `usbipd bind --busid <busid>` even if I unplug and plug in again, I have to call `usbipd unbind --busid <budis>` else I cannot find the COM port on the computer management

dorssel, 2022-03-28:
> by the way. it seems that the COM port has lost on windows side after using usbipd bind --busid <busid> even if I unplug and plug in again, I have to call usbipd unbind --busid <budis> else I cannot find the COM port on the computer management

You reported earlier that you used `bind --force`, and this is exactly what `--force` means. If you want to use the device with the host when not attached to a client, then leave out `--force`.

xiaolongba, 2022-03-28:
@dorssel 

Big thanks bro, clear now. 

thanks again.

nickcollins, 2022-04-22:
> @dorssel Bro,
> 
> Finally, I resolved this issue. and Now it works well. let me summarize my case here.
> 
> 1. root cause
>    My pc has not installed WSL before, but the WSL is exist already after installing the docker desktop windows app, so I misunderstood that the local WSL is the same as the WSL docker used. but the truth is that they are different. at least, I installed usbip in the local WSL, but the WSL docker used has not such a usbip. so it returned the error info `usbipd: error: WSL 'usbip' client not correctly installed. See https://github.com/dorssel/usbipd-win/wiki/WSL-support for the latest instructions. ` when typing in the `usbipd wsl attach --busid=8-1` in standard user cmd prompt. actually, the usbip has been installed in the WSL docker used, not the local WSL; However, the command `usbipd wsl attach --busid=8-1` has been called in the local WSL, of course it returned with the error info as the local WSL has not installed usbip.
> 2. workaround
>    
>    1. installing the local WSL manually with  `wsl --install -d <distribution name>`, such as  `wsl --install -d Ubuntu`
>    2. set the default distro, such as  `wsl --set-default Ubuntu-20.04`
>    3. follow the instructions which are from [USB/IP client tools](https://github.com/dorssel/usbipd-win/wiki/WSL-support#usbip-client-tools)
>    4. and then type in the command `usbipd bind --busid 8-1 --force` in the admin prompt, where 8-1 is the device I would like to bind
>    5. type in another command `usbipd wsl attach --busid 8-1` in the standard user prompt, where 8-1 is the device I would like to attach
>    6. and then switch to WSL with root rights, the command is `wsl --user root` and then type in ` dmesg | tail`, the output as below:
>    
>    ```
>    [32964.815885] usb 1-1: new full-speed USB device number 8 using vhci_hcd
>    [32964.896035] vhci_hcd: vhci_device speed not set
>    [32964.965934] usb 1-1: SetAddress Request (8) to port 0
>    [32965.003968] usb 1-1: New USB device found, idVendor=10c4, idProduct=ea60, bcdDevice= 1.00
>    [32965.003971] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
>    [32965.003972] usb 1-1: Product: CP2104 USB to UART Bridge Controller
>    [32965.003973] usb 1-1: Manufacturer: Silicon Labs
>    [32965.003974] usb 1-1: SerialNumber: 02446348
>    [32965.006859] cp210x 1-1:1.0: cp210x converter detected
>    [32965.010186] usb 1-1: cp210x converter now attached to ttyUSB0
>    ```

Thank you so much in this whole universe you are my only hero

vpathuis, 2022-05-12:
This all works, great detailed explanation. But... this still leaves the question how to attach an USB device to a docker-desktop wsl. Since usbipd won't install in the docker-desktop wsl, the attach won't work (as you've shown). Has someone been able to mount an USB-device to a docker container in Windows?

dorssel, 2022-05-12:
@vpathuis 
There is only one kernel instance. All distros and docker instances use that very same kernel.

What I understand (without using docker myself): you should install a normal distro alongside your (desktop)dockers. Then you can install the `usbip` tools on that "normal" distro (e.g. Ubuntu), just for running the user-mode tools. That will also be the target distro for `usbip wsl attach` (either make it the default distro or use `--distribution`). When a device is attached, it will attach to the one-and-only kernel and magically become available in all distros as well as the docker instances (as they all share that same kernel). Within each instance, you may have to (re)start udev to update the instance's view of the new hardware.

vpathuis, 2022-05-13:
That's so great! Thanks a lot.
In the mean time I've been experimenting with the attached usb devices (in Ubuntu for now). For an usb drive, using dmesg | tail I get:
`[ 3094.687086] vhci_hcd vhci_hcd.0: pdev(0) rhport(0) sockfd(3)
[ 3094.687088] vhci_hcd vhci_hcd.0: devid(65553) speed(5) speed_str(super-speed)
[ 3094.687123] vhci_hcd vhci_hcd.0: Device attached
[ 3095.029507] usb 2-1: SetAddress Request (2) to port 0
[ 3095.029515] usb 2-1: new SuperSpeed Gen 1 USB device number 2 using vhci_hcd
[ 3095.065533] usb 2-1: New USB device found, idVendor=8564, idProduct=1000, bcdDevice= a.00
[ 3095.065535] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 3095.065536] usb 2-1: Product: Mass Storage Device
[ 3095.065536] usb 2-1: Manufacturer: JetFlash
[ 3095.065537] usb 2-1: SerialNumber: 693062409`

I'm having a hard time recognizing where the usb is mounted on.

dorssel, 2022-05-13:
@vpathuis 
Your kernel is missing USB storage drivers, which are not part of the default WSL kernel. You will have to compile your own kernel to include those. See the wiki page on WSL and search through the issue list on "USB storage".
(And possibly: add your request to the ones already on the microsoft/WSL issue list to include such drivers be default in the future).

kevinzhai80, 2022-05-20:
> @dorssel Bro,
>
> Finally, I resolved this issue. and Now it works well. let me summarize my case here.
>
> 1. root cause
>    My pc has not installed WSL before, but the WSL is exist already after [...]

Hi, thanks for your plan. This is effect.
I have another issue, do you have occurs? as following:
at wsl ubuntu 20.04, I enter command:
usbipd wsl list

output:

WARNING: usbipd not found for kernel 5.10.60.1-microsoft

  You may need to install the following packages for this specific kernel:
    linux-tools-5.10.60.1-microsoft-standard-WSL2
    linux-cloud-tools-5.10.60.1-microsoft-standard-WSL2

  You may also want to install one of the following packages to keep up to date:
    linux-tools-standard-WSL2
    linux-cloud-tools-standard-WSL2

but, I don't install any tools it show.
Did you occurr this issue?

dorssel, 2022-05-20:
@kevinzhai80 
`usbipd wsl XXX` is supposed to be called on Windows, **not** within WSL on Linux.

kevinzhai80, 2022-05-20:
> @kevinzhai80 `usbipd wsl XXX` is supposed to be called on Windows, **not** within WSL on Linux.

I find I enter this command in powershell also get the same show.

dorssel, 2022-05-20:
@kevinzhai80 
That cannot be, the error:

> WARNING: usbipd not found for kernel 5.10.60.1-microsoft

only occurs when you run `usbipd` on Linux. What exactly is the error when you run `usbipd wsl list` on PowerShell (Windows)?

Alembertcn, 2022-08-15:
it do not work for me i

Alembertcn, 2022-08-15:
> Hi, thanks for your plan. This is effect. I have another issue, do you have occurs? as following: at wsl ubuntu 20.04, I enter command: usbipd wsl list
>
> output:
>
> WARNING: usbipd not found for kernel [...]

it do not work for me

team-orangeBlue, 2023-03-21:
Didn't work for me either. Everything installs but no luck. Tried debian 11 and Ubuntu 20.04.6. Both fail at the main error.

Device added with bind --force. Kernel 5.10.90. dmesg|tail returns some PCI bus stuff and PCI bars which is definitely not what I care about.

Maybe I should recompile my kernel for USBIPD or use a Windows 10 install instead?

CNXT-Simon, 2023-05-17:
> I will just post it hear in case anyone stumbles over it. Ubuntu Terminal:
> 
> ```shell
> update-alternatives --install /usr/local/bin/usbip usbip $(command -v ls /usr/lib/linux-tools/*/usbip | tail -n1) 20
> ```
> 
> Powershell Terminal:
> 
> ```shell
> wsl --shutdown 
> ```
> 
> Start the WSL Ubuntu terminal again. After all the install `linux-tools-x.x.x` and doing `update-alternatives...` I had to restart wsl itself, as even with the `update-alternatives` command I still got the "hey kernel miss match please install linux-tools-...-WSL2", which of course doesn't work on wsl.
> 
> Now `usbip` works.

I have encountered the same error message, the ```wsl -shutdown``` works. Thanks for your information.

khurramar, 2023-06-25:
> @dorssel Bro,
>
> Finally, I resolved this issue. and Now it works well. let me summarize my case here.
>
> 1. root cause
>    My pc has not installed WSL before, but the WSL is exist already after [...]

Always installed WSL and distros via the command line wsl --install or from the store. always worked without any issues on Windows 11. This is the first time on a new Windows 11 PC that I am stuck with this error. Have tried uninstalling and reinstalling the distros but the error persists when using the "usbipd wsl attach" command. Can't figure out what what's the issue.

wmattias, 2023-07-01:
> > @dorssel Bro,
> > Finally, I resolved this issue. and Now it works well. let me summarize my case here.
> > 
> > 1. root cause
> >    My pc has not installed WSL before, but the WSL is exist already after installing the docker desktop windows app, so I misunderstood that the local WSL is the same as the WSL docker used. but the truth is that they are different. at least, I installed usbip in the local WSL, but the WSL docker used has not such a usbip. so it returned the error info `usbipd: error: WSL 'usbip' client not correctly installed. See https://github.com/dorssel/usbipd-win/wiki/WSL-support for the latest instructions. ` when typing in the `usbipd wsl attach --busid=8-1` in standard user cmd prompt. actually, the usbip has been installed in the WSL docker used, not the local WSL; However, the command `usbipd wsl attach --busid=8-1` has been called in the local WSL, of course it returned with the error info as the local WSL has not installed usbip.
> > 2. workaround
> >    
> >    1. installing the local WSL manually with  `wsl --install -d <distribution name>`, such as  `wsl --install -d Ubuntu`
> >    2. set the default distro, such as  `wsl --set-default Ubuntu-20.04`
> >    3. follow the instructions which are from [USB/IP client tools](https://github.com/dorssel/usbipd-win/wiki/WSL-support#usbip-client-tools)
> >    4. and then type in the command `usbipd bind --busid 8-1 --force` in the admin prompt, where 8-1 is the device I would like to bind
> >    5. type in another command `usbipd wsl attach --busid 8-1` in the standard user prompt, where 8-1 is the device I would like to attach
> >    6. and then switch to WSL with root rights, the command is `wsl --user root` and then type in ` dmesg | tail`, the output as below:
> >    
> >    ```
> >    [32964.815885] usb 1-1: new full-speed USB device number 8 using vhci_hcd
> >    [32964.896035] vhci_hcd: vhci_device speed not set
> >    [32964.965934] usb 1-1: SetAddress Request (8) to port 0
> >    [32965.003968] usb 1-1: New USB device found, idVendor=10c4, idProduct=ea60, bcdDevice= 1.00
> >    [32965.003971] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> >    [32965.003972] usb 1-1: Product: CP2104 USB to UART Bridge Controller
> >    [32965.003973] usb 1-1: Manufacturer: Silicon Labs
> >    [32965.003974] usb 1-1: SerialNumber: 02446348
> >    [32965.006859] cp210x 1-1:1.0: cp210x converter detected
> >    [32965.010186] usb 1-1: cp210x converter now attached to ttyUSB0
> >    ```
> 
> Always installed WSL and distros via the command line wsl --install or from the store. always worked without any issues on Windows 10. This is the first time on a new Windows 11 PC that I am stuck with this error. Have tried uninstalling and reinstalling the distros but the error persists when using the "usbipd wsl attach" command. Can't figure out what what's the issue.

Same here - It was working before, but it suddenly stopped working. Getting the same error when i try to attach.

aschuhardt, 2023-09-03:
For anyone else who ends up here, this is the command I ended up having to use (after installing `linux-tools-5.10` and `usbip`):

```
sudo update-alternatives --install /usr/local/bin/usbip usbip /lib/modules/5.10.0-22-amd64/kernel/drivers/usb/usbip 20
```

PR #253 (2022-02-07): Report default distro used
- but only if there is more than 1

Fixes #251