dorssel/usbipd-win #257: Built my own kernel, but can't get it to display as a video device

hakanErgin, 2022-02-09:
Hi.
I built my own kernel following the instructions and my USB camera shows as `Attached - wsl2-usbip` successfully. It also shows when I enter `lsusb`.

Problem is, the device won't show as a video device under `/dev/videoX` directories. I believe that's where I should find it, no? To be clear, there is no directory that starts with 'video' under '/dev/'.

_My goal is to use it on a openCV project_

This is a USB webcam, however I tried with my laptop's built in webcam with the same results. 

There's a part where I doubt it worked properly, at the last step of building the kernel where the instructions say:

`[wsl2]
kernel=c:\\users\\<user>\\usbip-bzImage`

My windows username is two words with a whitespace in between. Can i just write it as `kernel=c:\\users\\user name\\usbip-bzImage` or should I escape it like `kernel=c:\\users\\user\ name\\usbip-bzImage` (which I have tried, can't tell if it did anything)

Thanks in advance!

akuropka, 2022-02-13:
No need to escape,
```
[wsl2]
kernel=C:\\Users\\A B\\bzImage
```
works.

dhutter, 2022-03-07:
bear in mind that WSL2 comes without udev-rules. So either you have to load all required modules by hand (modprobe) or more convenient, you have to compile them already into the kernel. Also firmware isn't loaded dynamically from /lib/firmware but has also to be included into the kernel during compile time. Hope that helps.

dorssel, 2022-03-26:
Closing due to lack of response from OP.

closed 2022-03-26 by dorssel