dorssel/usbipd-win #258: VBoxUsbMon error when attempting to attache devices

GarryMcKee, 2022-02-10:
Hi Team,

I am currently going through this document: https://devblogs.microsoft.com/commandline/connecting-usb-devices-to-wsl/
When I get to the _ step I get the following error: `The VBoxUsbMon driver is currently not running; a reboot should fix that.`
I have tried a few reboots but can't seem to solve the, issue, is there another manual step required to install VBoxUsbMon? A separate Virtual Box install or anything like that? 

Thanks in advance!

**Software versions:**
Windows 11 Version 21H2
usbipd-win 2.1.0

**Linux Kernel Version:**
Linux DESKTOP 5.10.60.1-microsoft-standard-WSL2 #1 SMP Wed Aug 25 23:20:18 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

**wsl -l -v output:**
  NAME      STATE           VERSION
* Ubuntu    Running         2

GarryMcKee, 2022-02-10:
@dorssel  should this be ran from a Windows prompt or from the WSL Linux shell?

On Windows I'm getting this:
```
C:\WINDOWS\system32>sc query VBoxUsbMon
[SC] EnumQueryServicesStatus:OpenService FAILED 1060:

The specified service does not exist as an installed service.

C:\WINDOWS\system32>sc start VBoxUsbMon
[SC] StartService: OpenService FAILED 1060:

The specified service does not exist as an installed service.
```

Don't have VirtualBox installed separately no,

dorssel, 2022-02-10:
There must have gone something wrong with the installer then. Try uninstalling `usbipd-win` and reinstall. How did you install? winget, or msi? Was there no error message?

GarryMcKee, 2022-02-10:
Got it! Seems like PowerToys PowerLauncher had a lock on something in needed and I needed to give it admin rights to close it during the install. Appreciate the nigh on realtime assistance @dorssel!

closed 2022-02-10 by GarryMcKee