dorssel/usbipd-win #259: super-speed-plus (ASUS ROG STRIX Arion) fails to attach

Fredolx, 2022-02-12:
Kernel: 5.10.60.1-microsoft-standard-WSL2 #1 SMP Wed Aug 25 23:20:18 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
Ubuntu WSL 2.0
USBIPD 2.1.0

When trying to attach my ASUS ROG STRIX Arion with ```usbipd wsl attach``` (it gets detected as ```USB Attached SCSI (UAS) Mass Storage Device``` with ```wsl --list --verbose```), I get these error messages:

```
usbip: error: import device
usbipd: error: Failed to attach device with BUSID '1-22'.
```
After running ```dmesg | tail```:
```
vhci_hcd: Failed attach request for unsupported USB speed: super-speed-plus
```

Thank you

dorssel, 2022-02-12:
@Fredolx 
Duplicate of #137 

I investigated a bit. It turns out that the Linux kernel (not just WSL, but *all* Linux kernels) does not support USB 3.1 Gen2 (super-speed-plus) over USBIP. `usbipd-win` does, so it reports super-speed-plus, but then Linux cannot handle it.

See Linux source code:
https://elixir.bootlin.com/linux/v5.10.60/source/drivers/usb/usbip/vhci_sysfs.c#L288

To resolve, you could connect this device to a different (non 3.1 Gen2) port, maybe you have a 3.0 or even 2.0 port available? Alternatively, you could connect a 3.0 or 2.0 hub between the port and the device.

Would you be willing to test a version that simply lies about the speed? I mean `usbipd-win` reporting super-speed instead of super-speed-plus. I have no idea if that could even work. I know for a fact that lying about USB1/USB2/USB3 does *not* work, but maybe lying about 3.0/3.1/3.1Gen2 works...

dorssel, 2022-02-12:
@Fredolx 
I've created PR #260 that may fix the problem. The installer is at https://github.com/dorssel/usbipd-win/actions/runs/1833543256.
Could you test that?

dorssel, 2022-03-12:
Closing due to lack of response.

closed 2022-03-12 by dorssel