dorssel/usbipd-win #261: Unable to locate and install usbip user-space tools for Kali in WSL2

amcamillo, 2022-02-13:
Been following Dorssel's and this [blogs](https://devblogs.microsoft.com/commandline/connecting-usb-devices-to-wsl/)' guide to setup usbip on a Kali Linux WSL2 - but I've had no luck.

I did came across another issue from this repo where kali was mentioned, and after performing windows updates to get the latest WSL kernel, the issue persisted.
I have enabled advanced update settings in windows update, pulled and installed all updates available from windows.

THis is my WSL kernel version:
└─$ uname -a
Linux AWLT 5.10.60.1-microsoft-standard-WSL2 #1 SMP Wed Aug 25 23:20:18 UTC 2021 x86_64 GNU/Linux

I have setup usbipd-win 2 and can get a list of attached usb devices from the CMD prompt.

However, when attempting to attach the usbipd bus to WSL, I get this error:
usbipd: error: WSL 'usbip' client not correctly installed. See https://github.com/dorssel/usbipd-win/wiki/WSL-support for the latest instructions.

Which I believe is related to the fact that I cannot setup the required tools in the Kali WSL instance.
THis is the error I'm getting when attempting to setup the linux tools in Kali:
└─$ sudo apt install linux-tools-5.4.0-77-generic hwdata
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package linux-tools-5.4.0-77-generic
E: Couldn't find any package by glob 'linux-tools-5.4.0-77-generic'

Any suggestions? Thanks in advance.

dorssel, 2022-02-13:
@amcamillo 
You need to install whatever package for Kali that installs the `usbip` program. First google hit: https://installlion.com/kali/kali/main/u/usbip/install/index.html
If that has changed for Kali (article above is from 2017), you can always compile your own kernel and `usbip` following https://github.com/dorssel/usbipd-win/wiki/WSL-support#building-your-own-usbip-enabled-wsl-2-kernel

akuropka, 2022-02-13:
`apt install usbip` does the trick.

closed 2022-03-12 by dorssel