dorssel/usbipd-win #262: Connecting a Device but Being Unable to Connect

silver2row, 2022-02-13:
Hello,

I have an Ubuntu Distro on the WSL2 instance on Win 11.

I have not been able to make this software work due to these requirements:

```

WARNING: usbipd not found for kernel 5.10.60.1-microsoft

  You may need to install the following packages for this specific kernel:
    linux-tools-5.10.60.1-microsoft-standard-WSL2
    linux-cloud-tools-5.10.60.1-microsoft-standard-WSL2

  You may also want to install one of the following packages to keep up to date:
    linux-tools-standard-WSL2
    linux-cloud-tools-standard-WSL2

```

I have not been able to install these tools on the WSL2 instance on Win 11 to get this library, usbipd-win, to work correctly. Is there any other workaround known to-date?

Seth

silver2row, 2022-02-13:
I got it to work. Way to go!

Seth

P.S. Nice lib. for WSL2 and building!

closed 2022-02-13 by silver2row

AntumArk, 2022-02-26:
How were you able to fix it? I am getting the same error just for Win10