dorssel/usbipd-win #264: Device on error mode

willzyx-hub, 2022-02-14:
Hi dorsel i try to attach my pendrive to wsl but it gave me error

Microsoft Windows [Version 10.0.22000.469]
(c) Microsoft Corporation. All rights reserved.

C:\Windows\system32>usbipd list
Connected:
BUSID  DEVICE                                                        STATE
2-3    USB Mass Storage Device                                       Shared
3-1    USB Input Device                                              Not shared
3-2    USB Input Device                                              Not shared
5-2    Intel(R) Centrino(R) Wireless-N + WiMAX 6150                  Not shared
5-4    TOSHIBA Web Camera - MP                                       Not shared

Persisted:
GUID                                  DEVICE

usbipd: warning: USB filter 'UsbDk' is known to be incompatible with this software; 'bind --force' will be required.

C:\Windows\system32>usbipd wsl attach -b 2-3
usbipd: warning: USB filter 'UsbDk' is known to be incompatible with this software; 'bind --force' will be required.
usbip: error: Attach Request for 2-3 failed - Device in error state

usbipd: error: Failed to attach device with BUSID '2-3'.

dorssel, 2022-02-14:
@willzyx-hub 
Well ... it says you should use `bind --force`, but you didn't ... why not?
```
usbipd bind --force --busid 2-3
```

willzyx-hub, 2022-02-14:
> @willzyx-hub Well ... it says you should use `bind --force`, but you didn't ... why not?
> 
> ```
> usbipd bind --force --busid 2-3
> ```

What to do after that ?

dorssel, 2022-02-14:
```
usbipd wsl attach -b 2-3
```

willzyx-hub, 2022-02-14:
> ```
> usbipd wsl attach -b 2-3
> ```

hey thanks its fixed my problem

closed 2022-02-14 by willzyx-hub

Kuhicop, 2022-03-15:
I had the same problem but after running `usbipd wsl attach -b 1-2` I get this error:

```
PS C:\WINDOWS\system32> usbipd wsl attach --busid 1-2
usbip: error: Attach Request for 1-2 failed - Device in error state

usbipd: error: Failed to attach device with BUSID '1-2'.
```

dorssel, 2022-03-15:
@Kuhicop 
Do I understand correctly that you used `bind --force`? Did it respond with a warning that a reboot may be required?
What is the output of `usbipd list`?