dorssel/usbipd-win #265: usbip: error: recv op_common usbip: error: query

spequer, 2022-02-14:
Hi,
I've successful connected the usbipd wsl. but failed after reboot.
with usbipd-win-2.1.0

> command
C:\windows\system32>usbipd wsl attach -b 4-3
usbip: error: recv op_common
usbip: error: query
usbipd: error: Failed to attach device with BUSID '4-3'.

> server trace
C:\windows\system32>usbipd server Logging:LogLevel:Default=Trace
dbug: Microsoft.Extensions.Hosting.Internal.Host[1]
      Hosting starting
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\windows\system32
dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started
dbug: UsbIpServer.Server[1000]
      new connection from 172.21.138.92
dbug: UsbIpServer.Server[1000]
      connection closed
dbug: UsbIpServer.Server[1000]
      new connection from 172.21.138.92
dbug: UsbIpServer.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
fail: UsbIpServer.ConnectedClient[3]
      An exception occurred while communicating with the client:
      UsbIpServer.ConfigurationManagerException (1003): CM_Query_And_Remove_SubTree returned CR_REMOVE_VETOED: PNP_VetoDevice, FTDIBUS\VID_0403+PID_6001+A904CVM0A\0000
         at UsbIpServer.ConfigurationManager.RestartingDevice..ctor(UInt32 deviceNode) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConfigurationManager.cs:line 395
         at UsbIpServer.ConfigurationManager.RestartingDevice..ctor(String instanceId) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConfigurationManager.cs:line 378
         at UsbIpServer.ConnectedClient.HandleRequestImportAsync(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConnectedClient.cs:line 148
dbug: UsbIpServer.Server[1000]
      connection closed

spequer, 2022-02-14:
another issue we could found is that
after issue 
>usbipd wsl attach 

there were some device found in device manager like this:

[image]

spequer, 2022-02-14:
It works after reboot. 
I can't reproduce it.

dorssel, 2022-02-14:
@spequer 
The actual "error" is CR_REMOVE_VETOED. This just means that something on the Windows host is already exclusively using the device. This could be either the original driver, some service, or some software application. In other words: the device is busy.

Either close the software that is already using the device, or (if you do not intend to use the device on the host ever, only from WSL), you can use `usbipd bind --force`.

What is weird is that `usbipd` reports this back to the Linux side, which should have nicely said "device busy". Instead, your `usbip` client says:

> usbip: error: recv op_common

That is not normal. What distro are you using? And which package did you install for `usbip`?

dorssel, 2022-02-14:
@spequer 
About the left-over devices after failed attach. That is not normal either. What is the error code for those? (see property page of the device). Can you verify that those left-over devices are gone after a reboot?

spequer, 2022-02-15:
@dorssel 
[image]
[image]
[image]
[image]

spequer, 2022-02-15:
@dorssel
I've tried usbipd with a new installed windows and it works.

On the computer with issue, we've install zadig to switch usb driver for openocd connection.
We've also installed virtualbox, virtualbox w/ extension to run ubuntu.

I may figure out what happened later if I have spare time..

dorssel, 2022-02-15:
@spequer 
I think there may be a conflict between VirtualBox (full installation) and usbipd-win. What version of VirtualBox do you have installed? See also #230.

spequer, 2022-02-16:
@dorssel 

I removed VirtualBox. Server trace is different. 

>server trace

C:\windows\system32>usbipd server Logging:LogLevel:Default=Trace
dbug: Microsoft.Extensions.Hosting.Internal.Host[1]
      Hosting starting
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\windows\system32
dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started
dbug: UsbIpServer.Server[1000]
      new connection from 192.168.50.105
dbug: UsbIpServer.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
fail: UsbIpServer.ConnectedClient[3]
      An exception occurred while communicating with the client:
      System.IO.FileNotFoundException: Unable to find the specified file.
         at UsbIpServer.ConfigurationManager.GetVBoxDevice(BusId busId) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConfigurationManager.cs:line 304
         at UsbIpServer.VBoxUsb.ClaimDeviceOnce(BusId busId) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsb.cs:line 66
         at UsbIpServer.VBoxUsb.ClaimDevice(BusId busId) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsb.cs:line 76
         at UsbIpServer.VBoxUsb.ClaimDevice(BusId busId) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsb.cs:line 82
         at UsbIpServer.ConnectedClient.HandleRequestImportAsync(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConnectedClient.cs:line 161
dbug: UsbIpServer.Server[1000]
      connection closed

Like this:
[image]
[image]
[image]
[image]

dorssel, 2022-02-16:
Now that you have removed VirtualBox, can you do a "Repair" install on `usbipd-win`. That should fix the driver issue. Because now it seems there are no drivers at all.

dorssel, 2022-03-26:
Closing due to lack of response.

closed 2022-03-26 by dorssel