dorssel/usbipd-win #270: Can attach iPhone 6s to WSL

d0n13, 2022-02-15:
WSL 2
UPSIPD 2.1.0
Ubuntu 21.10

Hi, when I attach an iPhone 6s and attempt to attach to WSL I get the following:

```
usbipd wsl attach -b 2-3
usbip: error: Attach Request for 2-3 failed - Device in error state

usbipd: error: Failed to attach device with BUSID '2-3'.
```

This is what the server logs. Looks like it can't find a file. This phone worked before on older versions of usbip. Currently using 2.1.0
```
info: UsbIpServer.ConnectedClient[1]
      Client 192.168.217.251 claimed device at 2-4 (USB\VID_05AC&PID_12A8\00008030000E28A63C80802E).
fail: UsbIpServer.ConnectedClient[3]
      An exception occurred while communicating with the client:
      System.IO.FileNotFoundException: Unable to find the specified file.
         at UsbIpServer.ConfigurationManager.GetVBoxDevice(BusId busId) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConfigurationManager.cs:line 304
         at UsbIpServer.VBoxUsb.ClaimDeviceOnce(BusId busId) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsb.cs:line 66
         at UsbIpServer.VBoxUsb.ClaimDevice(BusId busId) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsb.cs:line 76
         at UsbIpServer.VBoxUsb.ClaimDevice(BusId busId) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsb.cs:line 82
         at UsbIpServer.ConnectedClient.HandleRequestImportAsync(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConnectedClient.cs:line 161
```
AN iPhone 11 and works ok and any Android phone I have plugged in. Just two iPhone 6s devices are causing this issue...

Any thoughts?
Thanks
Donie

dorssel, 2022-02-15:
@d0n13 
Note: you're logs have mixed BUSIDs (2-3 vs. 2-4). Ignoring that... You can try
```
usbipd bind --force --busid 2-4
```
to permanently load the driver. With --force you can no longer use the device on the host, but attaching to WSL should succeed.

d0n13, 2022-02-15:
Hi @dorssel, thank you. That worked. Can you explain what it's doing here and why force works around the problem? I notice the device would move to the Persisted List when I did "usbipd list". Don't fully understand why devices are persisted.

Can I detect this state in code as I'm trying to attach the device pragmatically and I would need to know if I needed to do this force option before committing to a bind. I am trying to automatically attach devices to the WSL host and detect that it is successful.

dorssel, 2022-02-15:
If you want auto-detection of the device, then --force is definitely the way to go. First the "normal" (without --force):

- The device has its normal hardware ID, say VID_1234:PID_5678
- Windows loads the normal driver and the host can use it
- You do a normal bind. Nothing (!) happens yet, apart from some internal administration that now allows clients to ask for the device.
- A client does 'attach'
- The VBoxUsbMon driver/service unplugs the device (USB port reset) and then *modifies* the hardware ID to, say, VID_CAFE:PID_BABE
- When the port reset completes, Windows now thinks there is a completely different device attached and will load the driver for it, which happens to be VBoxUsb
- The VBoxUsb driver is used to forward client URBs to the device
- When you detach, the reverse happens: port reset, VBoxUsbMon stops lying about the device ID, normal driver loads

This is "magic".

With --force it is all much simpler:

- The device has its normal hardware ID, say VID_1234:PID_5678
- Windows loads the normal driver and the host can use it
- You do a bind --force. The driver is immediately *replaced* by VBoxUsb. No lying about the hardware ID or anything, just a driver "update".
- Now, the host cannot use the device, since it is no longer a serial port, or a webcam, but instead a "VBoxUsb" (that Windows doesn't know how to use).
- Client does attach: no USB port resets, no hardware ID tweaking, just immediately using the already loaded VBoxUsb

As you can see: the --force is much more reliable. In fact, VBoxUsbMon (the one that "tweaks" the hardware ID) is not even involved anymore. Downside: the Windows host cannot use the device at all, it is only available to USBIP clients.

Now you also understand why autodetecting the device *without* force is troublesome, as there are really 2 devices involved; the original, and the one that VBoxUsbMon lies about. *With* force there is just the one device, with its original device ID.

dorssel, 2022-02-15:
@d0n13 
Let me ask you: are you monitoring the device so you can do "auto-attach" to WSL as soon as the device is plugged in?

If so, I highly recommend it the other way around. Run a never-ending script in WSL that just does `usbip attach` every few seconds. No harm done if the device is already attached, no harm done if the device is currently not plugged in, and auto-attach as soon as it is available.

d0n13, 2022-02-15:
Yea, but I only want to takes phones across. And to do the attach we would have to know the busid. I can single out iPhones fairly easily as the vendor and product id's are limited. It's different for Androids, there are so many manufacturers. 

So we attach the devices and to detect they are Android we run adb commands on them to check if they respond. So we are running an app to do the bind and attach as required.

dorssel, 2022-02-16:
@d0n13 
The original issue is resolved, right? If so, please close the issue. Feel free to open any new issue for further help.

d0n13, 2022-02-16:
Ok, I'm now having an issue related to the attaching of an iPhone. I remove the device and unbind so that the current state is as below.

When I bind I'm getting an exception. Not sure what has changed....

```
c:\ usbipd list
Connected:
BUSID  DEVICE                                                        STATE
2-1    Logitech USB Input Device, USB Input Device                   Not shared
2-4    Apple Mobile Device USB Composite Device                      Not shared
2-14   Intel(R) Wireless Bluetooth(R)                                Not shared

Persisted:
GUID                                  DEVICE

PS C:\Users\donie> usbipd bind --force --busid 2-4
Unhandled exception. System.AggregateException: One or more errors occurred. (DIIDFLAG_INSTALLNULLDRIVER)
 ---> System.ComponentModel.Win32Exception (0x80004005): DIIDFLAG_INSTALLNULLDRIVER
   at UsbIpServer.NewDev.ThrowOnError(BOOL success, String message) in D:\a\usbipd-win\usbipd-win\UsbIpServer\NewDev.cs:line 21
   at UsbIpServer.NewDev.ForceVBoxDriver(String originalInstanceId) in D:\a\usbipd-win\usbipd-win\UsbIpServer\NewDev.cs:line 62
   at UsbIpServer.CommandHandlers.Bind(BusId busId, Boolean wslAttach, Boolean force, IConsole console) in D:\a\usbipd-win\usbipd-win\UsbIpServer\CommandHandlers.cs:line 152
   at UsbIpServer.CommandHandlers.UsbIpServer.ICommandHandlers.Bind(BusId busId, Boolean force, IConsole console, CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\CommandHandlers.cs:line 164
   at UsbIpServer.Program.<>c__DisplayClass10_1.<<Run>b__3>d.MoveNext() in D:\a\usbipd-win\usbipd-win\UsbIpServer\Program.cs:line 165
--- End of stack trace from previous location ---
   at System.CommandLine.Invocation.AnonymousCommandHandler.InvokeAsync(InvocationContext )
   at System.CommandLine.Invocation.InvocationPipeline.<>c__DisplayClass4_0.<<BuildInvocationChain>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass22_0.<<UseParseErrorReporting>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass15_0.<<UseHelp>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass26_0.<<UseVersionOption>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass24_0.<<UseTypoCorrections>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c.<<UseSuggestDirective>b__23_0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass21_0.<<UseParseDirective>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c.<<UseDebugDirective>b__8_0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c.<<RegisterWithDotnetSuggest>b__7_0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c.<<CancelOnProcessTermination>b__1_0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Invocation.InvocationPipeline.InvokeAsync(IConsole )
   at System.CommandLine.Parsing.ParseResultExtensions.InvokeAsync(ParseResult , IConsole )
   at System.CommandLine.Parsing.ParserExtensions.InvokeAsync(Parser , String[] , IConsole )
   --- End of inner exception stack trace ---
   at System.Threading.Tasks.Task.ThrowIfExceptional(Boolean )
   at System.Threading.Tasks.Task`1.GetResultCore(Boolean )
   at System.Threading.Tasks.Task`1.get_Result()
   at UsbIpServer.Program.Run(IConsole optionalTestConsole, ICommandHandlers commandHandlers, String[] args) in D:\a\usbipd-win\usbipd-win\UsbIpServer\Program.cs:line 504
   at UsbIpServer.Program.Main(String[] args) in D:\a\usbipd-win\usbipd-win\UsbIpServer\Program.cs:line 117
```

d0n13, 2022-02-16:
None of my iPhones will attach now. Android phones are ok. This looks like it's trying to update drivers etc. Is there anything I can supply which will help you to track this down?

dorssel, 2022-02-16:
If you unbind, Window will install the default Plug-and-Play driver, probably some Apple driver. My guess is that that required a reboot, so the device is now in state "driver installer, reboot required". Then, if you bind again, *removing* that driver fails, as its installation hasn't completed yet. You probably have to reboot first, to finish the driver installation, and only then can you bind again, removing the driver and installing VBoxUsb instead.

Installing VBoxUsb does not require a reboot, and neither does uninstalling it. So it must be the Windows default driver for the device that requires it. That is not something usbipd-win can fix (or even detect).

dorssel, 2022-02-16:
PS: what I would do is
- use `bind --force`
- never `unbind` a connected device
- only `unbind --guid` for devices that are currently not connected (those are the ones listed as "Persisted"). This unbind should never require a reboot, as the device is not even present.

dorssel, 2022-02-16:
Could you try PR #274? It won't fix anything, but hopefully gives a little bit more error detail. Installer is at https://github.com/dorssel/usbipd-win/actions/runs/1853781085.

d0n13, 2022-02-17:
@dorssel I have it installed but I'm not seeing any extra detail in the exception that is thrown. Where is the last windows error logged to?

dorssel, 2022-02-17:
*IF* we are lucky, then then number 0x80004005:
```
---> System.ComponentModel.Win32Exception (0x80004005): DIIDFLAG_INSTALLNULLDRIVER
```
is different now...

mvf, 2022-02-22:
I'm having the same issue with two devices. `bind` works but `attach` fails with `Device in error state` and `bind --force` fails with `---> System.ComponentModel.Win32Exception (0x80004005): DIIDFLAG_INSTALLNULLDRIVER`. Installing a driver fixed the problem for one device, but for the other there is no Windows driver available to me. After a reboot `bind`/`attach` worked with the driver-less device, but after replugging the `Device in error state` is back. #274 doesn't seem to add any new information:
```
C:\Users\User>usbipd --version
2.1.1-PullRequest0274.23+Branch.pull-274-merge.Sha.0d3f93fe62f32cefb5af602d06090865d8ecd3a3

C:\Users\User>usbipd bind --force --busid=6-3
Unhandled exception. System.AggregateException: One or more errors occurred. (DIIDFLAG_INSTALLNULLDRIVER)
 ---> System.ComponentModel.Win32Exception (0x80004005): DIIDFLAG_INSTALLNULLDRIVER
   at UsbIpServer.NewDev.ThrowOnError(BOOL success, String message) in D:\a\usbipd-win\usbipd-win\UsbIpServer\NewDev.cs:line 21
   at UsbIpServer.NewDev.ForceVBoxDriver(String originalInstanceId) in D:\a\usbipd-win\usbipd-win\UsbIpServer\NewDev.cs:line 62
   at UsbIpServer.CommandHandlers.Bind(BusId busId, Boolean wslAttach, Boolean force, IConsole console) in D:\a\usbipd-win\usbipd-win\UsbIpServer\CommandHandlers.cs:line 173
   at UsbIpServer.CommandHandlers.UsbIpServer.ICommandHandlers.Bind(BusId busId, Boolean force, IConsole console, CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\CommandHandlers.cs:line 185
   at UsbIpServer.Program.<>c__DisplayClass10_1.<<Run>b__3>d.MoveNext() in D:\a\usbipd-win\usbipd-win\UsbIpServer\Program.cs:line 164
--- End of stack trace from previous location ---
   at System.CommandLine.Invocation.AnonymousCommandHandler.InvokeAsync(InvocationContext )
   at System.CommandLine.Invocation.InvocationPipeline.<>c__DisplayClass4_0.<<BuildInvocationChain>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass20_0.<<UseParseErrorReporting>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass13_0.<<UseHelp>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass24_0.<<UseVersionOption>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass22_0.<<UseTypoCorrections>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c.<<UseSuggestDirective>b__21_0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass19_0.<<UseParseDirective>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c.<<RegisterWithDotnetSuggest>b__6_0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c.<<CancelOnProcessTermination>b__1_0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Invocation.InvocationPipeline.InvokeAsync(IConsole )
   at System.CommandLine.Parsing.ParseResultExtensions.InvokeAsync(ParseResult , IConsole )
   at System.CommandLine.Parsing.ParserExtensions.InvokeAsync(Parser , String[] , IConsole )
   --- End of inner exception stack trace ---
   at System.Threading.Tasks.Task.ThrowIfExceptional(Boolean )
   at System.Threading.Tasks.Task`1.GetResultCore(Boolean )
   at System.Threading.Tasks.Task`1.get_Result()
   at UsbIpServer.Program.Run(IConsole optionalTestConsole, ICommandHandlers commandHandlers, String[] args) in D:\a\usbipd-win\usbipd-win\UsbIpServer\Program.cs:line 502
   at UsbIpServer.Program.Main(String[] args) in D:\a\usbipd-win\usbipd-win\UsbIpServer\Program.cs:line 116
```
On another machine which is still running usbipd-win `1.2.1+11.Branch.mas.Sha.cf5c7e0d3e9254cbdae349d2ebc0b3e72171e9b4` (doesn't have `--force` yet), `attach` works sometimes, but it can also fail with `usbip: error: Attach Request for 1-1 failed - Request Failed`.

d0n13, 2022-02-22:
@dorssel I don't see any new error codes in the exception. Is the same even with the new version from #274

dorssel, 2022-03-10:
@d0n13 
No, I cannot reproduce (or explain) this.
The error code 0x80004005 just means: E_FAIL. That's not very helpful; I was hoping the PR would result in a more specific error code, but it obviously it didn't.
The call that fails is `DiInstallDevice`, see https://docs.microsoft.com/en-us/windows/win32/api/newdev/nf-newdev-diinstalldevice. DIIDFLAG_INSTALLNULLDRIVER is used to remove the current driver, and as far as I understand it should never fail (at least not with this error code).

dorssel, 2022-03-11:
@d0n13 
I found an issue with `DiInstallDevice`. Reported it here: https://github.com/microsoft/win32metadata/issues/826
I'll work around it, so we (hopefully) get the more detailed error.

d0n13, 2022-03-11:
That’s great. It’s blocking me at the moment. If we can help let me know.

mvf, 2022-03-11:
Great news, thanks for looking into this! Also happy to run test builds.

dorssel, 2022-03-12:
@d0n13 
@mvf 
I've updated the PR, new artifact at https://github.com/dorssel/usbipd-win/actions/runs/1973025503. This one should tell us *why* it fails.

d0n13, 2022-03-12:
@dorssel  ok, I might get some time tomorrow to test

d0n13, 2022-03-14:
@dorssel I have this build and installed now. 

When an iPhone is plugged in I'm getting this list:
```
Connected:
BUSID  DEVICE                                                        STATE
1-10   Goodix fingerprint                                            Not shared
1-11   Integrated Webcam, Camera DFU Device                          Not shared
1-14   Intel(R) Wireless Bluetooth(R)                                Not shared
3-2    Realtek USB GbE Family Controller                             Not shared
4-5    Realtek USB Audio                                             Not shared
4-6    Logitech USB Input Device, USB Input Device                   Not shared
4-7    Apple Mobile Device USB Composite Device                      Not shared

Persisted:
GUID                                  DEVICE

PS C:\Users\donie.kelly> usbipd bind --force -b 4-7
usbipd: warning: A reboot may be required before the changes take effect.
```
Not getting an error message like before. Even after reboot and plugging phone in again I get this message.
Can I do anything else to track this down?

dorssel, 2022-03-14:
@d0n13 
I notice that your original report (failing `bind --force`) had USB hub 2, but your latest reports have USB hub 1, 3, and 4.
You do realize that all of this binding business is quite hub-dependent? I'm afraid you are changing more than 1 variable between tests.

The original problem, the one with with hub 2, was that `bind --force` did not work at all. That's what the PR is trying to debug, i.e. to find out *why* it fails.

Your latest report, the one with hubs 1, 3, and 4, all look normal to me. It is very common for a driver change to require a reboot. And if `bind --force` tells you a reboot may be required, then I don't consider it an issue if a subsequent attach without reboot fails. That's expected behavior.

Can you please try the PR in the original situation, where the `bind --force` fails with DIIDFLAG_INSTALLNULLDRIVER?

d0n13, 2022-03-14:
@dorssel Yes, sorry. I changed dev machine and I have a different USB setup. So I removed the docking station. Should be a little simpler now.

So I rebooted...

```
PS C:\Users\donie> usbipd list
Connected:
BUSID  DEVICE                                                        STATE
1-10   Goodix fingerprint                                            Not shared
1-11   Integrated Webcam, Camera DFU Device                          Not shared
1-14   Intel(R) Wireless Bluetooth(R)                                Not shared

GUID                                  DEVICE
552a22c4-bb98-4a9c-95b6-5051063a6443  Apple Mobile Device USB Composite Device
5eac7c04-dfa3-45d0-97e3-d0bfd1d3d879  Apple Mobile Device USB Composite Device
```
Plugged in the iPhone as it's on the persisted list it should bind automatically
```
PS C:\Users\donie> usbipd listv
Connected:
BUSID  DEVICE                                                        STATE
1-3    Apple Mobile Device USB Composite Device                      Shared (forced)
1-10   Goodix fingerprint                                            Not shared
1-11   Integrated Webcam, Camera DFU Device                          Not shared
1-14   Intel(R) Wireless Bluetooth(R)                                Not shared

Persisted:
GUID                                  DEVICE
5eac7c04-dfa3-45d0-97e3-d0bfd1d3d879  Apple Mobile Device USB Composite Device

PS C:\Users\donie.kelly> usbipd wsl attach -b 1-3
PS C:\Users\donie.kelly> wsl lsusb
Bus 001 Device 002: ID 05ac:12a8 Apple, Inc. iPhone 5/5C/5S/6/SE
```
Connects to WSL ok
```
PS C:\Users\donie.kelly> usbipd wsl detach -b 1-3
PS C:\Users\donie.kelly> usbipd list
Connected:
BUSID  DEVICE                                                        STATE
1-3    Apple Mobile Device USB Composite Device                      Shared (forced)
1-10   Goodix fingerprint                                            Not shared
1-11   Integrated Webcam, Camera DFU Device                          Not shared
1-14   Intel(R) Wireless Bluetooth(R)                                Not shared

Persisted:
GUID                                  DEVICE
5eac7c04-dfa3-45d0-97e3-d0bfd1d3d879  Apple Mobile Device USB Composite Device

PS C:\Users\donie.kelly> usbipd unbind -b 1-3
PS C:\Users\donie.kelly> usbipd list
Connected:
BUSID  DEVICE                                                        STATE
1-3    Apple Mobile Device USB Composite Device                      Not shared
1-10   Goodix fingerprint                                            Not shared
1-11   Integrated Webcam, Camera DFU Device                          Not shared
1-14   Intel(R) Wireless Bluetooth(R)                                Not shared

Persisted:
GUID                                  DEVICE
5eac7c04-dfa3-45d0-97e3-d0bfd1d3d879  Apple Mobile Device USB Composite Device
```
Unbind and detached the phone and try again...
```
PS C:\Users\donie.kelly> usbipd bind --force -b 1-3
usbipd: warning: A reboot may be required before the changes take effect.
PS C:\Users\donie.kelly> usbipd wsl attach -b 1-3
usbip: error: Attach Request for 1-3 failed - Device in error state

usbipd: error: Failed to attach device with BUSID '1-3'.
```
So it won't bind the second time. Not getting the error we are looking for either with the drivers. This is obviously behaving differently for me atm.

riverar, 2022-03-14:
> win32meta

Fixed this up for you in metadata, should be fixed in the next release. (I just don't know what that will be yet.)

mvf, 2022-03-14:
I also tried the PR today on the "other" machine from before, the one that was running an old `usbipd-win` version that didn't have `--force` yet. I couldn't uninstall the driver from the "patient zero" machine today because it was being used. Apologies if these findings are not relevant.

So I plugged in a device for which there was no driver installed. After a fresh `usbipd-win` install, `bind` succeeds (with or without `--force`). An `unbind` after a plain `bind` succeeds, but on `unbind` after a `bind --force`, `DiInstallDevice` fails with `0xE0000203` aka `ERROR_NO_DRIVER_SELECTED`:
```
C:\WINDOWS\system32>usbipd list
Connected:
BUSID  DEVICE                                                        STATE
1-1    QUSB_BULK_CID:0404_SN:BD0A388F                                Shared (forced)
...

C:\WINDOWS\system32>usbipd unbind --busid=1-1
Unhandled exception. System.AggregateException: One or more errors occurred. (DiInstallDevice)
 ---> System.ComponentModel.Win32Exception (0xE0000203): DiInstallDevice
   at UsbIpServer.Tools.ThrowOnError(BOOL success, String message) in D:\a\usbipd-win\usbipd-win\UsbIpServer\Tools.cs:line 116
   at UsbIpServer.NewDev.UnforceVBoxDriver(String originalInstanceId) in D:\a\usbipd-win\usbipd-win\UsbIpServer\NewDev.cs:line 129
   at UsbIpServer.CommandHandlers.UsbIpServer.ICommandHandlers.Unbind(BusId busId, IConsole console, CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\CommandHandlers.cs:line 272
   at UsbIpServer.Program.<>c__DisplayClass10_3.<<Run>b__13>d.MoveNext() in D:\a\usbipd-win\usbipd-win\UsbIpServer\Program.cs:line 292
--- End of stack trace from previous location ---
```
Afterwards the device name is `USBIP Shared Device` and it's `Not shared`:
```
C:\WINDOWS\system32>usbipd list
Connected:
BUSID  DEVICE                                                        STATE
1-1    USBIP Shared Device                                           Not shared
```
Trying a plain `bind` after this triggers the same exception:
```
C:\WINDOWS\system32>usbipd bind --busid=1-1
Unhandled exception. System.AggregateException: One or more errors occurred. (DiInstallDevice)
 ---> System.ComponentModel.Win32Exception (0xE0000203): DiInstallDevice
   at UsbIpServer.Tools.ThrowOnError(BOOL success, String message) in D:\a\usbipd-win\usbipd-win\UsbIpServer\Tools.cs:line 116
   at UsbIpServer.NewDev.UnforceVBoxDriver(String originalInstanceId) in D:\a\usbipd-win\usbipd-win\UsbIpServer\NewDev.cs:line 129
   at UsbIpServer.CommandHandlers.Bind(BusId busId, Boolean wslAttach, Boolean force, IConsole console) in D:\a\usbipd-win\usbipd-win\UsbIpServer\CommandHandlers.cs:line 173
   at UsbIpServer.CommandHandlers.UsbIpServer.ICommandHandlers.Bind(BusId busId, Boolean force, IConsole console, CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\CommandHandlers.cs:line 185
   at UsbIpServer.Program.<>c__DisplayClass10_1.<<Run>b__3>d.MoveNext() in D:\a\usbipd-win\usbipd-win\UsbIpServer\Program.cs:line 164
--- End of stack trace from previous location ---
```
Afterwards, the device is still reported as `USBIP Shared Device` but it's `Shared (forced)`. `attach` always succeeds as long as the device is shared. There were no `A reboot maybe required` messages. All this is reproducible after reinstalling `usbipd-win`.

d0n13, 2022-03-14:
Was just looking for this error code and came across this. I don't think it's relevant due to no *'s in the section of your VBoxUSB.inf file but maybe there is something here that might help?

https://social.msdn.microsoft.com/Forums/en-US/f399760e-6a7e-4c2a-8cee-82787bd9f8e8/updatedriverforplugandplaydevices-fails-with-0xe0000203-since-windows-10-1809

d0n13, 2022-03-14:
@mvf I noticed that the iPhone was named as USBIP Shared Device too at one point. When I did the unbind it went away and I haven't see since.

I tried to reproduce what you are getting above but it's working ok for me. Can't get it to throw that exception yet... which is annoying as it was happening all the time at one point last week....

dorssel, 2022-03-15:
@mvf 
Hi. Your case is interesting... This time it fails at *uninstalling* the stub driver (whereas all the other problems were with *installing*). Some follow up questions:

1) What device type is this?
2) If you normally connect this device to Windows for the first time, does it install a driver? Or does it show as "unknown device" in device manager with a question mark?
3) If normally a driver is installed, could you find and post the .inf file for it?

dorssel, 2022-03-15:
@mvf 
@d0n13 
Microsoft has fixed the problem with DiInstallDevice. I noticed in the fix (see https://github.com/microsoft/win32metadata/pull/827/files) that they fixed something else as well; something I hadn't realized. The broken definition also mixed up Ansi vs. Unicode. That may have caused the exception in the first place. Now that I work around the first problem (not setting last error code), I inadvertently fixed the second problem too. That may be the reason that the exception isn't showing at all any longer.

mvf, 2022-03-16:
@dorssel 
>1. What device type is this?

It's a Qualcomm Snapdragon device in QDL mode (for flashing). Bricked Snapdragon Android phones also seem to switch to this mode:

    05c6:9008 Qualcomm, Inc. Gobi Wireless Modem (QDL mode)

>2. If you normally connect this device to Windows for the first time, does it install a driver? Or does it show as "unknown device" in device manager with a question mark?

It doesn't install a driver. In device manager it looks like this:
[image]
I found [this driver on the Microsoft Update Catalog](https://www.catalog.update.microsoft.com/Search.aspx?q=qualcomm%20qdloader%209008) and installed it on the other machine to work around the failing `bind --force`.

dorssel, 2022-03-16:
@mvf 
That's what I thought. I want to solve it on the usbipd-win side, without resorting to installing (unwanted) drivers. Would you be willing to perform some more testing later?

dorssel, 2022-03-16:
@d0n13
Good to see the new error reporting is working.

>  ---> System.ComponentModel.Win32Exception (5): DIIDFLAG_INSTALLNULLDRIVER

5 == ERROR_ACCESS_DENIED

0xe0434352 == Anything broken with registry, .NET installation, missing files, etc. This usually means your Windows installation is corrupted. Google for possible repair actions. Not something usbipd-win can fix for you.

mvf, 2022-03-16:
@dorssel 
> Would you be willing to perform some more testing later?

Sure, but from this Friday the hardware will be unavailable to me until April 4th.

dorssel, 2022-03-16:
@mvf 
The current master build should solve the `--unbind` problem. Can you arrange it again such that no default Windows driver is available? Like before? The installer is at https://github.com/dorssel/usbipd-win/actions/runs/1994623081. Then do `bind --force` and `---unbind` again. This should now succeed without error, returning the device to "Other devices" with question mark, just as Windows would do for a new device. Thx.

mvf, 2022-03-17:
@dorssel Confirmed, with 5e60aaf57ec9a5925260c7afd88ce5d680e65842, `unbind` succeeds after `bind --force` and the device shows as driverless in device manager. Also switched back and forth again to make sure it still fails on the old build. Thanks!

d0n13, 2022-03-22:
@dorssel Is it possible that our antivirus software is causing problems with unloading the drivers during --force? We use forticlient. Our software seems to work ok on machines that don't have it installed. But I have not had a wide enough test yet to prove that out. Just wondering what you think?

[image]

dorssel, 2022-03-22:
@d0n13 
Erm ... yes, that may explain things.
See for example: https://www.reddit.com/r/fortinet/comments/krpph5/fortideviceguardsys/

dorssel, 2022-03-26:
Closing.
This issue mentions (too) many different issues. Most were solved, including the original issue.
Remaining issue (fortinet) opened as new issue.

closed 2022-03-26 by dorssel