dorssel/usbipd-win #273: Detect what devices are attached to WSL from Windows

d0n13, 2022-02-16:
Thank you @dorssel. I'm trying to understand the complexities of this and you are being very helpful in previous tickets. Thanks you.

I'm trying to build up a view of the connected devices in a Windows UI and I'm running into some issues on the best way to do this. Because devices are persisted and you plug out and back in, the device is automatically attached to WSL but the `usbipd` list command does not reflect it? Here is the fact that the devices are actually attached to WSL. I also looked in the registry and all the entries do not contain anything that gives me a hint of their current state afaics.

ie: Linux shows devices as per normal and the are "attached"...

```
$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 024: ID 18d1:4ee6 Google Inc. Nexus/Pixel Device (PTP + debug)
Bus 001 Device 023: ID 05ac:12a8 Apple, Inc. iPhone 5/5C/5S/6/SE
Bus 001 Device 022: ID 05ac:12a8 Apple, Inc. iPhone 5/5C/5S/6/SE
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
But, the usbipd list command just shows this. It's not showing the three devices 2-3, 2-4, and 2-7 as attached. I was trying to use that as a way to display the connected devices in my UI. 

```
C:\ usbipd list
Connected:
BUSID  DEVICE                                                        STATE
2-1    Logitech USB Input Device, USB Input Device                   Not shared
2-3    Apple Mobile Device USB Composite Device                      Shared (forced)
2-4    Apple Mobile Device USB Composite Device                      Shared
2-7    Pixel, ADB Interface                                          Shared
2-14   Intel(R) Wireless Bluetooth(R)                                Not shared

Persisted:
GUID                                  DEVICE
574aba3a-b6b8-4ad2-abad-b6151d52977d  Pixel
5b147eb7-0cb0-46e3-bc3c-334d906ff400  Apple Mobile Device USB Composite Device
```

Then I tried to use the usbip on the Linux side to see if I could get the list of attached devices there in a form that is useful but it's not terribly friendly to parse ie:

```
$ usbip port
Imported USB devices
====================
libusbip: error: fopen
libusbip: error: read_record
Port 00: <Port in Use> at High Speed(480Mbps)
       Apple, Inc. : iPhone 5/5C/5S/6/SE (05ac:12a8)
       1-1 -> unknown host, remote port and remote busid
           -> remote bus/dev 002/003
libusbip: error: fopen
libusbip: error: read_record
Port 01: <Port in Use> at High Speed(480Mbps)
       Apple, Inc. : iPhone 5/5C/5S/6/SE (05ac:12a8)
       1-2 -> unknown host, remote port and remote busid
           -> remote bus/dev 002/004
libusbip: error: fopen
libusbip: error: read_record
Port 02: <Port in Use> at High Speed(480Mbps)
       Google Inc. : Nexus/Pixel Device (PTP + debug) (18d1:4ee6)
       1-3 -> unknown host, remote port and remote busid
           -> remote bus/dev 002/007
```

QUESTION: 
Do we have any nice way to extract the devices that are actually connected to the WSL side in an easy to digest way for code to parse?

dorssel, 2022-02-16:
In the situation described above, the output of `usbipd list` *should have been* "Attached". If it really isn't, then that's a bug. We are talking about version 2.1.0, right?

If you want to see which WSL instance they are connected to, you could instead use `usbipd wsl list`, which will (or should) also report "Attached" but also includes the distro that the devices are attached to.

More programmatically, you could parse Windows EventLog, which will log every attach and detach.

So: can you give me a sequence of attached/detaches that leads to some devices being attached, but not reported by `usbipd list`? Then I can try to reproduce it (and fix it, hopefully).

d0n13, 2022-02-16:
Hi @dorssel 

Yes, I'm using 2.1.0

I had to reboot the machine shortly after this situation as the Linux side stopped letting me do an 'lsusb' and would stall with no output.

Previous to that it was showing three attached devices as above even though I had detached them. Linux still thought they were connected.

The reboot fixed that but I'm still seeing odd behaviour. Let me try to do get some things logged for you as I need to double check I'm doing it correctly this side.

Thanks

dorssel, 2022-02-16:
That makes sense. If the Linux kernel is confused (and I recommend keeping an eye on syslog), then the devices were not really attached at all. To `usbipd` it is simply: TCP connection up -> attached, TCP disconnected -> detached. Hardly anything that could go wrong there.

dorssel, 2022-02-16:
PS: instead of reboot, next time only try to restart WSL (`wsl --shutdown`). If that fixes it, you know the problem was Linux-side, not Windows.

d0n13, 2022-02-16:
I did and it didn't fix it. Maybe the problem was the usbipd service?

dorssel, 2022-02-16:
That could be, but unlikely. Like I said: to `usbipd` attached/detached is just the same as TCP connected/disconnected. The logic is very simple for that.

dorssel, 2022-03-16:
@d0n13 
> I did and it didn't fix it. Maybe the problem was the usbipd service?

I don't think it is possible that Linux thought the devices are still attached after the Linux kernel rebooted.

I'm closing this issue. Looks like Linux was confused about the attach-state. `usbipd-win` did nothing wrong here.

closed 2022-03-16 by dorssel