dorssel/usbipd-win #278: Would USBIPD-win help to get WSL2 working with my bluetooth adapter?

hansmbakker, 2022-02-19:
I would like to use my laptop's bluetooth module from inside WSL2. I can attach the device as shown below, but bluetooth support in Ubuntu 20.04 (WSL) seems missing. Also I read that kernel recompilation might be needed if the WSL kernel doesn't support a device.

Has anybody gotten a bluetooth adapter working inside WSL2? (a lot of people here would be helped: https://github.com/Microsoft/WSL/issues/242)

output from
> usbipd wsl list
```
BUSID  DEVICE                                                        STATE
2-9    Xbox Wireless Adapter for Windows                             Not attached
2-10   Intel(R) Wireless Bluetooth(R)                                Attached - Ubuntu
5-1    Surface Ethernet Adapter #2                                   Not attached
7-5    USB-invoerapparaat                                            Not attached
8-4    Realtek USB GbE Family Controller                             Not attached
10-3   Microsoft USB Wheel Mouse Optical                             Not attached
10-5   Surface Dock 2 Audio, USB-invoerapparaat                      Not attached
12-1   USB 2.0 BILLBOARD                                             Not attached
```

dorssel, 2022-02-19:
@hansmbakker 
Yes, I have 2 Bluetooth adapters working; one really old Belkin, and one integrated in a Dell laptop. I tested discovery, pairing, and audio.

However, I used them on a Hyper-V VM with a normal ubuntu kernel. I'm pretty sure for WSL you will have to compile your own kernel. But if you have the right drivers included, I'm pretty sure it'll work fine.

hansmbakker, 2022-02-19:
Oh that sounds promising!

However, I also discovered that `bluez` (for `hcitool`) was not installed in WSL2's Ubuntu 20.04 distribution, and I had problems installing it (I got the exact same error message as in this stackoverflow post: [Failed to open connection to system message bus : Failed to connect to socket /var/run/dbus/system_bus_socket](https://stackoverflow.com/questions/64380418/failed-to-open-connection-to-system-message-bus-failed-to-connect-to-socket-v)).

Following the suggestions in that stackoverflow post did not help solving the issue unfortunately.  `hcitool` became available but it could find no adapter device, and I'm not sure if recompiling the kernel would be enough seeing that error about dbus (which bluez relies on)

dorssel, 2022-02-19:
I just did:
```
sudo apt install dbus
sudo service start dbus
dbus-monitor --system
```
on Windows 10 WSL2. Seems to work just fine...

hansmbakker, 2022-02-19:
Odd.. On a clean, but updated version of Ubuntu 20.04.3 LTS on Windows 11 / WSL2 I get:
```
root@DESKTOP-9ULOVQP:/# apt install dbus
Reading package lists... Done
Building dependency tree
Reading state information... Done
dbus is already the newest version (1.12.16-2ubuntu2.1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
root@DESKTOP-9ULOVQP:/# service start dbus
start: unrecognized service
```

dorssel, 2022-02-19:
I had the service command reversed. See fixed version.

hansmbakker, 2022-02-20:
Can happen :)

With a new kernel (I had to enable bluetooth support and its device drivers) I managed to have WSL2 detect the bluetooth adapter. Initially `hci0`'s state is `down`, so I had to run `hciconfig hci0 up`.

However, `hcitool lescan` does not work still.
> hcitool lescan
```
Set scan parameters failed: Input/output error
```

`bluetoothctl` hangs with
```
Waiting to connect to bluetoothd...
```

> hciconfig hci0
```
hci0:   Type: Primary  Bus: USB
        BD Address: 80:38:FB:0F:83:DA  ACL MTU: 1021:4  SCO MTU: 96:6
        UP RUNNING
        RX bytes:2839 acl:0 sco:0 events:184 errors:0
        TX bytes:1807 acl:0 sco:0 commands:184 errors:0
```

> dmesg
```
[  523.959717] vhci_hcd vhci_hcd.0: pdev(0) rhport(0) sockfd(3)
[  523.959720] vhci_hcd vhci_hcd.0: devid(131082) speed(2) speed_str(full-speed)
[  523.959763] vhci_hcd vhci_hcd.0: Device attached
[  524.239715] vhci_hcd: vhci_device speed not set
[  524.309715] usb 1-1: new full-speed USB device number 3 using vhci_hcd
[  524.389743] vhci_hcd: vhci_device speed not set
[  524.459736] usb 1-1: SetAddress Request (3) to port 0
[  524.494771] usb 1-1: New USB device found, idVendor=8087, idProduct=0026, bcdDevice= 0.02
[  524.494773] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[  524.499154] Bluetooth: hci0: Firmware revision 0.0 build 110 week 24 2021
[  526.635368] vhci_hcd: unlink->seqnum 522
[  526.635371] vhci_hcd: the urb (seqnum 522) was already given back
[  526.636509] vhci_hcd: unlink->seqnum 523
[  526.636512] vhci_hcd: the urb (seqnum 523) was already given back
```

I tried lots of things (like `service dbus restart`, `service bluetooth restart`) but they did not help. `systemctl` related commands for Ubuntu do not work in WSL2 because apparently that is not a thing there:
> systemctl status bluetooth.service
```
System has not been booted with systemd as init system (PID 1). Can't operate.
Failed to connect to bus: Host is down
```

dorssel, 2022-02-20:
I get this for service bluetooth:
```
Feb 20 01:54:28 Oakdale bluetoothd[719]: Failed to access management interface
Feb 20 01:54:28 Oakdale bluetoothd[719]: Adapter handling initialization failed
```

And this thread: https://github.com/moby/moby/issues/16208

Since WSL is essentially running distro instances in a container, it looks like bluetooth is not going to work on WSL. Hyper-V is fine, though, so it's not the device support. It is the service support that is lacking...

hansmbakker, 2022-02-20:
Mm, that's too bad..

I thought that, in general to share Bluetooth with docker, one could specify some networking setup (to share the host network with the container)?

In any case, thank you very much for trying this along with me!

akuropka, 2022-02-26:
> Since WSL is essentially running distro instances in a container, it looks like bluetooth is not going to work on WSL. Hyper-V is fine, though, so it's not the device support. It is the service support that is lacking...

While dbus is implemented for some time, systemd isn’t yet. There are workarounds and quite some articles about that available. It will come sooner or later officially (Ubuntu is working on that).

JPHutchins, 2022-03-14:
Good news @hansmbakker !  

```
[bluetooth]# scan on
Discovery started
[CHG] Controller 5C:F3:70:A1:5C:F0 Discovering: yes
[NEW] Device 70:3B:0E:F7:C5:EB 70-3B-0E-F7-C5-EB
[NEW] Device 5C:38:D1:38:FE:51 5C-38-D1-38-FE-51
````

This is on Windows 10, WSL2 Ubuntu 20.04, with a "Pluggable" BCM2070 adapter.

I followed yours and @dorssel 's directions to compile the kernel again with Bluetooth enabled.  The directions are here: 
https://github.com/dorssel/usbipd-win/wiki/WSL-support#building-your-own-usbip-enabled-wsl-2-kernel

And I found it helpful to refer to this page: https://ubuntu.com/core/docs/bluez/reference/device-enablement/linux-kernel-configuration-options

I also enabled the 2070 driver since I knew I would be using it.  But it is my first time compiling using a Kconfig for Linux so YMMV.  But at least I can use `usbipd wsl attach --distribution wsl2-usbip --busid 7-4` to see my BLE adapter in WSL2 with `lsusb`.

Then I would install dbus and systemd with apt and see the same problem.  However, I already recognized this from earlier in the day on a W11 attempt in which we got systemd working using this: https://github.com/arkane-systems/genie

After installing genie according to the guide, I run `genie -l` from my WSL2.  Here it has a known problem where it hangs on first run, needs investigation, but Ctrl-C and trying again will "work".

From there I can use `bluetoothctl` and use systemctl to see status of the bluetooth service.

I will be attempting to use this as a documented development environment for people that don't want to use a traditional VM/Baremetal environment for this project: https://github.com/JPHutchins/gatter.  So, I will be going through the process again on my colleague's W11 environment and I will try to document better.

Thanks for all this amazing work!

JPHutchins, 2022-03-14:
Update: forget about the genie stuff and systemd.

1. Start dbus: `sudo service dbus start`
2. Bluetooth is disabled but `/etc/init.d/bluetooth` sources `/etc/default/bluetooth`.  We can write the environment variable to that file so that it is enabled:
    `echo 'export BLUETOOTH_ENABLED=1' | sudo tee /etc/default/bluetooth`
3. Start bluetooth: `sudo service bluetooth start`.

At this point, assuming your USB device is attached to WSL2, you should be able to test with `bluetoothctl` and then `scan on`, for example.

sparklerfish, 2022-03-16:
I successfully got this working on a Windows 11 machine following these steps:
- Followed the [WSL2 kernel build guide](https://github.com/dorssel/usbipd-win/wiki/WSL-support#building-your-own-usbip-enabled-wsl-2-kernel)
- My WSL2 kernel is `5.10.60.1-microsoft-standard-WSL2`
- Cloned wsl2 and checked out `linux-msft-wsl-5.10.60.1`
- In the menuconfig, I enabled
  - Device Drivers -> USB Support -> Debug messages for USB
  - Networking Support -> followed [this list](https://ubuntu.com/core/docs/bluez/reference/device-enablement/linux-kernel-configuration-options) (except for things that weren't present)
  - Added Bluetooth Device drivers (not sure if this was needed?)
- Inserted modules:
  - `sudo modprobe bluetooth`
  - `sudo modprobe btusb`
  - `sudo modprobe bcm203x` (turns out was unnecessary, unloaded with `sudo modprobe -r bcm203x`)
- Started dbus and bluetooth services

After this I was able to use `bluetoothctl` successfully!