dorssel/usbipd-win #279: Blue screen issue when the device is attached to WSL

bhnaphade, 2022-02-21:
Blue screen issue 

I have a USB hub connected to my laptop and a USB-serial adapter connected to it.
The moment I attach the USB-serial adapter to WSL2 (Ubuntu 18) and then connect my Monitor on the type c plug on my Surface Laptop 3 I get Blue screen . This happens every time when the device is attached to WSL.. When no devices are attached I can connect my monitor or other USB devices and it works

dorssel, 2022-02-21:
What are the details on the blue screen? What module/driver does it say caused it, and what is the stop code?

bhnaphade, 2022-02-21:
[image]
this is the error every time another usb device is connected when the USB-serial adapter is attached to WSL

dorssel, 2022-02-21:
PNP_DETECTED_FATAL_ERROR is new to me. From what I read, it is a misbehaving driver. This could definitely be VBoxUsb.sys (the driver `usbipd-win` uses), but not necessarily. It could also be the hub driver, or a USB filter driver.

Just to rule those out: can you check the driver of the hub(s) that your device is connected to? Is it the stock Microsoft hub driver, or some 3rd party driver? You can find the hubs in Device Manager (View: By Connection).

 And to check the filter drivers (if any), what is the output of:
```
reg query "HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}" /v UpperFilters
```

bhnaphade, 2022-02-22:
Apparently this issue also happens if the usb-serial adapter is directly connected to the USB port on Laptop and when I connect type C device to Laptop

dorssel, 2022-02-22:
@bhnaphade 
Duplicate of #200 

For this reason, I've added a warning "incompatible USB filter detected". Is that not displayed when you do `usbipd wsl list`?

bhnaphade, 2022-02-22:
@dorssel  Thank you for the prompt help, removing the  EUsbHubFilter entry HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000} from regedit solved the problem

closed 2022-02-22 by bhnaphade